openscad/openscad #4655: openscad crashes on big minkowski

gratefulfrog, 2023-05-26:
**Describe the bug**
I try to do a big minkowksi, with $fn= a number more than 20.

Crash.

Is there any way of giving openscad more ram, and/or using multiple cores?

**Environment and Version info (please complete the following information):**
 - OS: [Ubuntu 18.04]
 - System: [e.g. Intel PC 64-bit,]
 - OpenSCAD Version 2021.04.17.nightly (git f07f3e2)  and where it was installed from repo

Any suggestions would be great.

Thanks for the best tool !!

kintel, 2023-06-28:
Could your provide an example .scad file?
Also, how much RAM do you have on your system?

gratefulfrog, 2023-06-30:
Attached the scad file and supporting dxf.

if I  increase $fn beyond 10, openscad crashes on the minkowski

I have 12GB RAM on my linux pc.

Thanks for your help - I love OpenScad !!!

B

kintel, 2023-06-30:
The attachment didn't make into github. If the file is small, you can just copy&paste the text.

gratefulfrog, 2023-06-30:
Here are the 2 attachments
[GuitarScad.zip](https://github.com/openscad/openscad/files/11917445/GuitarScad.zip)

kintel, 2023-06-30:
I'll run some tests. In the meantime I suggest two things:

1. Update to a more recent nightly build.
2. In Preferences->Features, enable `manifold` and try again.

gratefulfrog, 2023-06-30:
thanks for those tips, but where do I get a more recent build? I have
looked for this before and never found anything more recent?

My version of openscad does not offer the manifold feature...

I'm sorry to be such an idiot, believe me I am very sorry...

Thanks for your help!

kintel, 2023-06-30:
There are lots of options at https://openscad.org/downloads.html#snapshots, depending on your distro.

kintel, 2023-06-30:
I did some testing without enabling manifold: With $fn=30 your design rendered in 14 minutes using 3.3 GB memory at peak. I'm not convinced that the crash is related to memory, but if your system happens to have no swap space, it's possible. One way you can test is to run OpenSCAD on the cmd-line like this:

```
/usr/bin/time -v openscad GuitarScad/maker.scad -o out.stl
```

This should show you how long time it took, and how much memory was used (maximum resident set size).

gratefulfrog, 2023-07-01:
Hi,
Thanks for your help.

I have installed using snap openscad-nightly  OpenSCAD
<https://www.openscad.org/> version 2023.06.25.

I selected manifold in the features, and effectively it can compile with
higher values of $fn, for example 100, and very fast.

This is a huge improvement for me!

Thanks so much!

closed 2024-02-04 by t-paul