openscad/openscad #4658: CSG export unexpectedly changes directory

burunduk3, 2023-06-03:
**Describe the bug**
Before opening the output file **openscad** changes directory to the directory of the source file, and creates output file relatively to it, not to the directory it is invoked from.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a simple model in a directory, like `mechanical/sample.scad`
2. Run **openscad** from a parent directory, like `openscad --hardwarnings -o sample.csg mechanical/sample.scad`

**Expected behavior**
A `sample.csg` file in the directory I ran **openscad** from.

**Actual behavior**
The `sample.csg` file is stored to the `mechanical` directory, the one which contains the model. In my original setup I want to put resulting file in a directory called `.build`, which makes **openscad** fail, as there are no `mechanical/.build/` directory.

**Code reproducing the issue**
Any sample model will work, even just `cube();`.

**Environment and Version info (please complete the following information):**
 - OS: Gentoo Linux
 - System: Intel 64-bit
 - OpenSCAD Version: same behavior on 2021.01 from the repository and 2023.06.03 freshly built from sources

**Additional context**
* This behavior is inconsistent with generating **stl** files, where **openscad** uses filename relatively to the current directory.
* I didn't try looking into the source, but according to `strace`, **openscad** explicitly changes directory to the model directory before writing the result file.
* For me it's not a critical problem, as I can easily workaround with symlinks or something.

t-paul, 2023-06-03:
Duplicate of #4632.

burunduk3, 2023-06-03:
Thanks, will subscribe to the mentioned one.

closed 2023-06-03 by burunduk3