openscad/openscad #4661: Keep parameter text box focused after editing

shayded-exe, 2023-06-05:
If automatic preview is on, hitting tab after editing a parameter will cause the text box to become unfocused and require you to click into the box to edit again.

I would like to be able to just tab from one parameter to the next. I know you can do this with automatic preview off, but it's very annoying to have to turn that off just to be able to edit the parameters with my keyboard efficiently.

jordanbrown0, 2023-06-05:
I assume that the context here is the Customizer.

Tab should focus the **next** field in the form, so that you can enter, tab, enter, tab, et cetera.  To be clear, it should not leave the focus in the **same** field.

shayded-exe, 2023-06-06:
Correct, that's what I want. The issue is that when the preview is generated, the customizer completely loses focus and tab doesn't navigate to any fields until I click back into one.