openscad/openscad #4662: error while compiling with cgal

tiredtait, 2023-06-06:
Buiding on a rockpi / raspberry pi knockoff under linux, using the sources from github, the system libraries for everything but cgal (which is using the git repository) and GMP which was downloaded from the GMP website.  compiling gets the following error: 

/home/p/doc/local/git/openscad/src/geometry/manifold/manifoldutils.cc:85:3: erro
r: cannot bind non-const lvalue reference of type ‘CGAL::SM_Face_index&’ to an r
value of type ‘boost::iterators::detail::iterator_facade_base<CGAL::Surface_mesh
<CGAL::Point_3<CGAL::Epick> >::Index_iterator<CGAL::SM_Face_index>, CGAL::SM_Fac
e_index, std::random_access_iterator_tag, CGAL::SM_Face_index, long int, false, 
false>::reference’ {aka ‘CGAL::SM_Face_index’}
   85 |   for (auto& f : tm.faces()) {

./scripts/check-dependencies.sh gives:

depname           minimum           found             OKness            
qt                5.4               5.15.2            OK                
qscintilla2       2.9               2.11.6            OK                
cgal              5.4               5.6-I-900         NotOK             
gmp               5.0               6.2.1             OK                
mpfr              3.0               4.1.0             OK                
boost             1.56              1.74              OK                
opencsg           1.4.2             1.4.2             OK                
glew              1.5.4             1.7.0             OK                
eigen             3.0               3.3.9             OK                
glib2             2.0               2.66.8            OK                
fontconfig        2.10              2.13.             OK                
freetype2         2.4               2.10.4            OK                
harfbuzz          0.9.19            2.7.4             OK                
libzip            0.10.1            1.7.3             OK                
bison             2.4               3.7.5             OK                
flex              2.5.35            2.6.4             OK                
make              3                 4.3               OK                
double-conversion 2.0.1             2.0.1             OK                
Warning: you have a copy of CGAL under /usr/local/include
Warning: you have a copy of gmp.h under /usr/local/include
Warning: you have pkgconfig under /usr/local/lib
Please verify these local copies don't conflict with the system

bit more from the output of make:
In file included from /usr/local/include/CGAL/number_type_basic.h:51,
                 from /usr/local/include/CGAL/Gmpq.h:16,
                 from /home/p/doc/local/git/openscad/src/geometry/cgal/cgal.h:18
,
                 from /home/p/doc/local/git/openscad/src/geometry/cgal/cgalutils
.h:3,
                 from /home/p/doc/local/git/openscad/src/geometry/manifold/manif
oldutils.cc:7:
/usr/local/include/CGAL/double.h: In function ‘std::pair<double, double> CGAL::s
plit_numerator_denominator(double)’:
/usr/local/include/CGAL/double.h:196:37: note: parameter passing for argument of
 type ‘std::pair<double, double>’ when C++17 is enabled changed to match C++14 i
n GCC 10.1
  196 | split_numerator_denominator(double d)
      |                                     ^
In file included from /usr/local/include/CGAL/int.h:23,
                 from /usr/local/include/CGAL/number_type_basic.h:56,
                 from /usr/local/include/CGAL/Gmpq.h:16,
                 from /home/p/doc/local/git/openscad/src/geometry/cgal/cgal.h:18
,
                 from /home/p/doc/local/git/openscad/src/geometry/cgal/cgalutils
.h:3,
                 from /home/p/doc/local/git/openscad/src/geometry/manifold/manif
oldutils.cc:7:
/usr/local/include/CGAL/Modular_arithmetic/Residue_type.h: In function ‘CGAL::Re
sidue CGAL::operator+(const CGAL::Residue&)’:
/usr/local/include/CGAL/Modular_arithmetic/Residue_type.h:277:45: note: paramete
r passing for argument of type ‘CGAL::Residue’ when C++17 is enabled changed to 
match C++14 in GCC 10.1
  277 | inline Residue operator + (const Residue& p1)
      |                                             ^
In file included from /usr/local/include/CGAL/number_type_basic.h:56,
                 from /usr/local/include/CGAL/Gmpq.h:16,
[... 96 lines cut ...]
 required from here
/home/p/doc/local/git/openscad/src/geometry/manifold/manifoldutils.cc:85:3: erro
r: cannot bind non-const lvalue reference of type ‘CGAL::SM_Face_index&’ to an r
value of type ‘boost::iterators::detail::iterator_facade_base<CGAL::Surface_mesh
<CGAL::Point_3<CGAL::Epick> >::Index_iterator<CGAL::SM_Face_index>, CGAL::SM_Fac
e_index, std::random_access_iterator_tag, CGAL::SM_Face_index, long int, false, 
false>::reference’ {aka ‘CGAL::SM_Face_index’}
   85 |   for (auto& f : tm.faces()) {
      |   ^~~
/home/p/doc/local/git/openscad/src/geometry/manifold/manifoldutils.cc: In instan
tiation of ‘std::shared_ptr<ManifoldGeometry> ManifoldUtils::createMutableManifo
ldFromSurfaceMesh(const TriangleMesh&) [with TriangleMesh = CGAL::Surface_mesh<C
GAL::Point_3<CGAL::Cartesian<double> > >]’:
/home/p/doc/local/git/openscad/src/geometry/manifold/manifoldutils.cc:108:106:  
 required from here
/home/p/doc/local/git/openscad/src/geometry/manifold/manifoldutils.cc:85:3: erro
r: cannot bind non-const lvalue reference of type ‘CGAL::SM_Face_index&’ to an r
value of type ‘boost::iterators::detail::iterator_facade_base<CGAL::Surface_mesh
<CGAL::Point_3<CGAL::Cartesian<double> > >::Index_iterator<CGAL::SM_Face_index>,
 CGAL::SM_Face_index, std::random_access_iterator_tag, CGAL::SM_Face_index, long
 int, false, false>::reference’ {aka ‘CGAL::SM_Face_index’}
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:289: CMakeFiles/OpenSCAD.dir/sr
c/geometry/manifold/manifoldutils.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:592: CMakeFiles/OpenSCAD.dir/all] Error 2
make: *** [Makefile:182: all] Error 2

t-paul, 2023-06-14:
Maybe try building with a CGAL release version.

tiredtait, 2023-06-23:
going to an older version of CGAL worked, thanks

closed 2023-06-23 by tiredtait