openscad/openscad #4664: Length mark

sonichy, 2023-06-11:
[image]

closed 2023-06-11 by thehans

t-paul, 2023-06-11:
Image is broken and gives error `Cannot proxy the given URL`.

sonichy, 2023-06-13:
fixed

t-paul, 2023-06-13:
Still having problems. New link is not reachable.

[image]

sonichy, 2023-06-13:
New link is not any of github !

t-paul, 2023-06-13:
I know. As the screenshot above shows, the server `ttcad.com` is not reachable from here and the host checker https://downforeveryoneorjustme.com/ agrees too.

Do you have access to sites like imgur.com or imgpile.com or imgbb.com ?

sonichy, 2023-06-14:
OK, uploaded to imgse.com.

jordanbrown0, 2023-06-14:
Er... and the URL there is?