openscad/openscad #4668: Openscad crashes on linear_extrude with list for scale

coolian1337, 2023-06-12:
**Describe the bug**
Openscad crashes when executing this code
```
linear_extrude(height = 200, scale = [2.0, 1.0], center = true)
square(size = [26, 20], center = true);
```

**To Reproduce**
Steps to reproduce the behavior:
1. open a new openscad project
2. copy the code into the editor
3. save and let openscad preview the code

**Expected behavior**
It should give me a square extruded with correct scaling

**Code reproducing the issue**
```
linear_extrude(height = 200, scale = [2.0, 1.0], center = true)
square(size = [26, 20], center = true);
```

**Environment and Version info (please complete the following information):**
openscad -v output: `OpenSCAD version 2023.05.30` 
Artix linux with kerinel 6.3.4-artix1-1 64 bit
using x11
openscad installed from the Arch user repository with the package `openscad-git`

**Library & Graphics card information**
output of openscad --info:
```
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2023.05.30
System information: Linux 6.3.4-artix1-1 #1 SMP PREEMPT_DYNAMIC Sun, 28 May 2023 21:36:47 +0000 x86_64 Artix Linux 2 CPUs 7.67 GB RAM
User Agent: OpenSCAD/2023.05.30 (Linux x86_64; Artix Linux)
Compiler: GCC "13.1.1 20230429" 64bit
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.9
QScintilla version: 2.14.0
InputDrivers: 
GLib version: 2.76.2
lodepng version: 20210627
libzip version: 1.9.2
fontconfig version: 2.14.2
freetype version: 2.13.0
harfbuzz version: 7.3.0
cairo version: 1.17.8
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /usr/bin
Documents Path: /home/julianw/.local/share
User Documents Path: /home/julianw/Documents
Resource Path: /usr/share/openscad
User Library Path: /home/julianw/.local/share/OpenSCAD/libraries
User Config Path: /home/julianw/.config/OpenSCAD
Backup Path: /home/julianw/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/julianw/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/julianw/.local/share/fonts
  /usr/share/fonts
  /usr/local/share/fonts
  /home/julianw/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/julianw/Library/Fonts
  /usr/share/fonts/OTF
  /usr/share/fonts/TTF
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/awesome-terminal-fonts
  /usr/share/fonts/cantarell
  /usr/share/fonts/droid
  /usr/share/fonts/encodings
  /usr/share/fonts/gsfonts
  /usr/share/fonts/liberation
  /usr/share/fonts/misc
  /usr/share/fonts/noto
  /usr/share/fonts/util
  /usr/local/share/fonts/s
  /usr/share/fonts/encodings/large

GLEW version: 2.2.0
GL context creator: GLX (old)
GLX version: 1.4
PNG generator: lodepng

OpenGL Version: 4.6 (Compatibility Profile) Mesa 23.1.2
GL Renderer: Mesa Intel(R) HD Graphics 510 (SKL GT1)
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

jordanbrown0, 2023-06-12:
Works OK on 2023.06.06 on Windows/64, with no features enabled.

kintel, 2023-06-28:
I notice that the Mint build being reported wasn't built as a development snapshot; it misses the git hash in the version, which makes it hard to identify exactly what the source of the build is. I guess this is a Mint packaging bug though, but if someone knows anything about that, it could be good to get that sorted.

kintel, 2024-04-01:
Is this still and issue with recent snapshots?

JulianWogersien, 2024-04-01:
I dont really use openscad actively at the moment so i have no idea and i dont use the same system anymore.

kintel, 2024-04-02:
Thanks for the feedback. Let's close this and reopen if this resurfaces.

closed 2024-04-02 by kintel