openscad/openscad #4669: Preview artifact when using difference with two linear_extrude'ed squares

Equbuxu, 2023-06-12:
**Describe the bug**
Kinda hard to describe, but there is a preview issue when cutting away parts from separate objects produced by a single linear_extrude. This only affects preview, F6 produces what looks like a correct result.

[image]

**Expected behavior**
[image]

**To Reproduce**
Steps to reproduce the behavior:
Run this code:
```openscad
difference() {
    linear_extrude(10) {
        translate([20, 0]) square([10, 10]);
        square([10, 10]);
    }
    cube([10, 10, 10], center = true);
}
```

Changing the code to the following fixes the issue:

```openscad
difference() {
    union() {
        linear_extrude(10) {
            translate([20, 0]) square([10, 10]);
        }
        linear_extrude(10) {
            square([10, 10]);
        }
    }
    cube([10, 10, 10], center = true);
}
```

Interestingly, this slightly different piece of code produces exactly the same buggy preview (render() is important here):
```openscad
difference() {
    render() {
        translate([20, 0]) 
            cube([10, 10, 10]);
        cube([10, 10, 10]);
    }
    cube([10, 10, 10], center = true);
}
```
Not exactly sure if it's caused by the same issue

**Environment and Version info:**
Likely irrelevant, I can reproduce this on two completely different PCs, with the only commonality between them being a NVIDIA GPUs and Windows 11
 - OS: Windows 11
 - System:  1: a Ryzen 2700 / NVIDIA a4000 desktop and an i5-8300h / NVIDIA 1050ti laptop
 - Can reproduce in 2021.01-x86-64 and 2023.06.06-x86-64

thehans, 2023-06-12:
You need to set convexity
```
difference() {
    linear_extrude(10, convexity=2) {
        translate([20, 0]) square([10, 10]);
        square([10, 10]);
    }
    cube([10, 10, 10], center = true);
}
```

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?

Equbuxu, 2023-06-12:
Thank you, that was it!