openscad/openscad #4671: Export color information

tpimh, 2023-06-16:
Since #1593 is now closed, I submit this new issue to track the progress. Right now the color information in lost when rendering (F6).

Currently, the only format that OpenSCAD can export color information to is CSG. Other export formats supported by OpenSCAD discard the color information.

Here is a quick overview of supported 3D export formats:

- [STL](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+STL%22) doesn't support color
- OBJ supports color
- OFF supports color
- WRL supports color, one possible use for this is KiCAD import (#1304)
- [AMF](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+AMF%22) supports color, but [is dead](https://github.com/openscad/openscad/issues/1593#issuecomment-1593507155) and superseded by 3MF, color export possible with a [workaround](https://github.com/jschobben/colorscad)
- [3MF](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+3MF%22) supports color, widely used for multi-material 3D printing, color export possible with a [workaround](https://github.com/jschobben/colorscad)
- [STP](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+STEP%22) supports color, but is not yet supported by OpenSCAD

Here is a quick overview of supported 2D export formats:

- [DXF](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+DXF%22) color support [seems complicated](https://ezdxf.readthedocs.io/en/stable/concepts/aci.html)
- [SVG](https://github.com/openscad/openscad/issues?q=label%3A%22Tag%3A+SVG%22) supports color
- PDF supports color

One way to implement this is to add a new rendering process "Render with color" which would render each color separately and show the combined colored preview, then add support for different export formats.

gsohler, 2023-06-16:
> Largely a duplicate of #1608
>
> Note that "render each color separately" is not as easy as it might sound;
> you need to make sure that the right thing (however you define it) happens
> when you union or intersect parts of two colors.

I agree with jordan,
if you just see how the outside colored faces of a difference() look like,
it's hard to guess,  what's the inner color of the difference() object in
each location ....

jordanbrown0, 2023-06-16:
If you ignore compatibility with current OpenSCAD preview behavior, difference() actually isn't a problem.  The negative object(s) are, well, negative; the obvious behavior is that the resulting object is the same color as the positive object.

Current OpenSCAD preview behavior is that the resulting object takes the negative objects' colors along the cut faces.  I suspect that that behavior won't survive a transition to true color support.

nophead, 2023-06-16:
You can always apply colour to the top level and then the resulting object is that colour regardless of unions and differences. 

In my previews everything is coloured at the top level. If it is a vitamin it is the natural colour. If it is a printed object that becomes an STL then it is the colour of the plastic if specified else RepRap green of red ro blue to make objects that meet distinguishable. Things that become DXFs for milling are the colour of the sheet material specified.

owebeeone, 2023-09-03:
Maybe there needs to be another primitive that allows the preservation of multiple dijoint objects. Just a thought.

e.g. If you declare (by a new attribute) that a node is disjoint. It will remain a separate object. An attempt to union with that object actually results in a difference and the resulting model remains "disjoint". Difference should remain the same except semantics should be that the result is multiple disjoint objects.

I haven't thought this through completely but the idea is that there is no longer a single model but multiple models whose intersection is empty and maintained that way across union and difference ops.

Come to think of it a bit more, I could probably implement this idea in AnchorSCAD where it will generate multiple disjoint .scad files.

bobc, 2023-12-21:
When I looked at this before, I came to the conclusion that it could not be done with CGAL, because CGAL combines objects losing any color information in the process. Therefore a different back end renderer is needed. AFAIK nothing has changed in that respect, although I haven't kept up with developments in CGAL.

Recently I have been using FreeCAD to convert OpenSCAD models to STEP, via the "CSG" export in OpenSCAD. I noticed that the FreeCAD CSG importer does preserve color information and render it correctly in FreeCAD. However, I haven't tried printing via that route.

kintel, 2024-07-20:
As this is likely significant work for each possible file format and separate topics for 2D and 3D, let's instead track tickets for specific formats as they're requested.
We now have color export to OFF, so the 3D part of this feature is now supported. 2D color support is currently tracked in #1364

closed 2024-07-20 by kintel