openscad/openscad #4672: assertion violation! Expr: e_below != SHalfedge_handle()  upon rendering

gvcastellano, 2023-06-19:
Previewing the below design works ok, although the render appears to have an incomplete mesh.
After preview, attempting to render yields:

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 418 
Geometries in cache: 7
Geometry cache size in bytes: 139928
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 247888
Total rendering time: 0:00:00.144
Rendering finished.

Repeating the render does not reproduce the problem nor does it appear to actually render anything.  Restart openscad to reproduce the problem.

---BeginCode---
$fa=1;

mink_radius = 2;

  intersection () 
    { 
    minkowski ()
      {
      difference () {
        cylinder( d=44.5-2*mink_radius, h=35 );
          translate ([0, 20, 22])
            rotate ([0, 90, 0])
              cylinder ( h=50, r=17, center = true );
      }
      sphere ( r=mink_radius );
    }
    cylinder( d=49.5, h=41 );
  }

---EndCode---

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 22.04.2 LTS
 - System: Dell XPS 64-bit Intel i7-6700HQ
 - OpenSCAD Version  2021.01via std Ubuntu repositories

**Library & Graphics card information**
OpenSCAD Version: 2021.01
System information: Linux 5.15.0-71-generic #78-Ubuntu SMP Tue Apr 18 09:00:29 UTC 2023 x86_64 Ubuntu 22.04.2 LTS 8 CPUs 30.97 GB RAM
User Agent: OpenSCAD/2021.01 (Linux x86_64; Ubuntu 22.04.2 LTS)
Compiler: GCC "11.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.3
QScintilla version: 2.11.6
InputDrivers: 
GLib version: 2.71.3
lodepng version: 20180910
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.11.1
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/gvc/.local/share
User Documents Path: /home/gvc/Documents
Resource Path: /usr/share/openscad
User Library Path: /home/gvc/.local/share/OpenSCAD/libraries
User Config Path: /home/gvc/.config/OpenSCAD
[... 385 lines cut ...]
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

jordanbrown0, 2023-06-19:
[ I don't have an answer, but I do have a few pieces of the puzzle. ]

First, not related to your complaint... use minkowski(convexity=2) or more to get rid of preview artifacts.

Second, only tangentially related... the reason that you don't see the error on a subsequent render is that OpenSCAD caches render results, including failures.  The subsequent render re-uses the failed result, without recalculating it and encountering the error again.  (There is a good argument that this is a bug.)

This problem is unusual in that it appears only when you preview (F5) and then render (F6).  (Most errors of this general sort require only F6.)  Reproduce with Design/Flush Caches, F5, F6.

Here is a somewhat simpler and faster demonstration.

```
$fn = 30;

minkowski()
{
    difference () {
        cylinder ( d=40.5, h=35 );
        translate ([0, 20, 22])
            rotate ([0, 90, 0])
            cylinder ( h=50, r=17, center = true );
    }
    cube(2,center=true);
}

translate([100,0,0]) cube();
```

Likely cause:  something about exactly how the two cylinders fit together causes minkowski to produce bad geometry, that then causes CGAL to die while attempting to import it.  Note that the cube at the end is nowhere near the model; it's only there to cause a CGAL operation.

Workaround: there's a good chance that tweaking the parameters of the model slightly will avoid the problem.

gvcastellano, 2023-06-19:
Thanks for the tips, I meant to add that when mink_radius = 0.9 in the original example there's no trouble with rendering so you presume correctly.

Duxah, 2023-11-01:
Good afternoon. 

[image]

I'm having the same problem. I mean that when compiling OpenSCAD it produces the same error text. I'll tell you the story from the very beginning - I created a complex object in OpenSCAD, then generated an STL and when generating OpenSCAD did not issue a single warning, everything worked out perfectly. Then I import this STL, add a simple Cube, click the STL button and the program gives this error described above. What to do? It turns out that either the import does not work, or the export does not work. One of two things is not working. I am attaching a screenshot. I can send files if necessary..

Best regards, Eugenii

jordanbrown0, 2023-11-01:
This is probably all known behavior.

OpenSCAD (CGAL, actually) is very picky about what inputs it will accept - and, in particular, some OpenSCAD outputs are not acceptable.  (Yes, that's unfortunate.)

On import, you don't see a problem until you add that cube because CGAL doesn't get involved (and reject the polyhedron) until there's something for it to do.

There are those - who will, I hope, chime in - who know how to examine the STL to figure out exactly what's wrong with it.  There's usually some microscopic defect.

Duxah, 2023-11-01:
(Google translate)
Here is a guess and a suggestion:

1) check the conversion of a string to a double and vice versa, as well as the absence of any alignments to grids, since CGAL is so demanding, as an option - increase the number of decimal places in floating-point numbers when writing STL.

2) it is possible, in theory, as an additional checks after STL compilation, do an automatic test for the closedness of the resulting object, and if everything is ok, then the problem may really be in losses when writing/reading from the file. You can try to compile the STL and immediately add the cube and check the compilation again, as if we had just imported this STL. Thus, we will exclude or confirm the version of losses during writing/reading. But of course, this can only be done by the developer in debug mode, not by the user.

Best regards

jordanbrown0, 2023-11-01:
I haven't looked into the problems in detail; I just know that they are known.  There *is* a grid at about 10^-6 units, and my understanding is that indeed it plays a role in this kind of thing.  I am told that getting rid of the grid introduces other problems, where vertices that are supposed to match up don't.

Binary STL is unfortunately defined using single precision.  ASCII STL could in theory use more, though whether it was useful would depend on whether the consumers used the additional precision.

Duxah, 2023-11-03:
(Google translate)

There are at least 3 algorithms for snapping to the grid - rounding to the bottom, to the top, and from the middle. What method is used when reading STL? To improve, I would like to be able to change the grid snap type in the OpenSCAD settings, because I think this could help solve this problem when it occurs. Nowadays OpenSCAD either works or it doesn't, and then the gnashing of teeth occurs. And with the adjustment it will be like this: if it doesn’t work, then we go into the settings, tweak the parameter and there is a high probability that it will start working. It is very useful to have some settings to solve such problems. Most often, an error occurs during compilation with the text “given mesh is not closed”, and nothing can be done about it - neither MeshLab nor MeshMixer helps, you just waste time on some kind of shamanism with STL. I downloaded many different STL models from the Internet, and even generated STL from STEP format, and all of them did not compile in OpenSCAD. Please add some settings.

OpenSCAD is a fantastic program for solid modeling, and if you finally solve this issue it will be just a fairy tale, but with such errors you start looking for an alternative program

There is one more proposal, I don’t know where to write it, I’ll write it here, in general the gist is this: when we import an STL file, the object always appears God knows where, because every designer designs his model in the coordinates where he wants. I would like to have a function that will set the body to specified coordinates (for example 0,0,0) along the lower left rear edge (as an option). Or it would be nice to have a function GetMinX GetMinY GetMinZ GetMaxX GetMaxY GetMaxZ, then using translate() to move the body where we need it

best wishes

jordanbrown0, 2023-11-03:
With the note again that I have not looked into the problem in detail... it is not that STL coordinates are snapped (though they might be), it is that coordinates are snapped during the creation of the original model and the resulting STL can be defective.

Unfortunately, the current situation is that because of microscopic defects, importing an STL fails very frequently.  It might be possible to have OpenSCAD repair the defects, but so far nobody with the requisite expertise has stepped up to add that feature.

With respect to your last point... I believe it would be relatively easy to have the STL importer accept some kind of alignment option.  It already has a `center` option, that STL import ignores.  It would *not* be easy to export information about the imported STL to the running program, because we don't actually look at the STL until after the program runs.  (We could have a separate function that reads the STL and returns metadata, and #4478's `render()` function can yield information about a geometric object.)  A request for an alignment option would be best made in a separate Github issue, tagged "enhancement request".  But note:  if you intend to modify an object from an STL you have to know a lot about it already; I'd think that knowing its origin would be the least of your problems.

kintel, 2024-04-01:
Status as of 2024-03-24: This seems to be fixed now, both when using the CGAL and Manifold backend.

closed 2024-04-01 by kintel