openscad/openscad #4676: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron => union()

vinsdonz, 2023-06-23:
Hi,

I'm trying to make a simple product with some cubes and one polyhedron but I got the error message "The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron" when I try to render it (F6) whereas the preview is OK (F5).

I've read so much things about this kind of error but there is often a stl file involved or some problems with the point orders. Here, I can render my polyhedron if it's alone. I can render my cubes without polyhedron. The problem occurs only when I try to union them. I've tried to delete the faces of the polyhedron which are in contact with the cubes faces but it doesn't change anything.
When I try the F12 thing (sorry I'm french and my OpenScad menu too so I don't have the english version of the option) it doesn't change faces into pink / violet. 

I use only the original libraries and use the polyhedron for the first time (it may be the explanation :) ). I'm running it on linux and my version of OpenScad is 2021.01

Here is my code :

`//Hauteur de la planche
Height = 20;
//Epaisseur de la figure
Thickness = 2;
//Profondeur de la figure
Depth = 30;
//Profondeur - Epaisseur
DepthT = Depth - Thickness;
//Largeur de la pièce
Width = 50;

myPoints = [
    [  0,  0, 50 ], //0 =>top
    [  0, 20,  0 ], //1 => right
    [  0,-20,  0 ], //2 => left
    [ 25,  0,  0 ]  //3 => front
];

myFaces = [
    [  1,  0,  2 ], //back
    [  3,  2,  1 ], //bellow
    [  1,  3,  0 ], //front
    [  2,  0,  3 ]  //front
];

union(){
    cube(size=[DepthT,Width,Thickness]);
    translate([DepthT,0,0])
        cube(size=[Thickness,Width,Height]);
    translate([Thickness,0,Height])
        cube(size=[DepthT,Width,Thickness]);
    translate([0,0,Height])
        cube(size=[Thickness,Width,80]);
    translate([2,Width / 2,Height+2])
        polyhedron(myPoints, myFaces,10);
}`

Here is a screenshot of the render with F5 and the with F6 (polyhedron has disappeared): 

[image]
[image]

For me, regarding what I've read, it's a bug, but maybe it's just I don't understand a thing (or several things :p )

Thank you for reading me

jordanbrown0, 2023-06-23:
The face on the bottom of your polyhedron is wound backwards.

Isolate the polyhedron.  (Putting `!` in front is an easy way.)  Set View/Thrown Together, and F5.  Look for purple.  Any purple is wrong.  Purple is the inside of the polyhedron; if you see any then either a face is backwards or you have a hole.

[image]

Swap any two vertexes in your "bellow" list:
```
[ 2, 3, 1 ], //bellow
```

vinsdonz, 2023-06-23:
Thanks a lot for the quick answer. I've made the back before the bottom so I turned the piece to do the clockwise order and forgot to turn it again before doing the bottom. 
Sorry for the bug report, I close it.

closed 2023-06-23 by vinsdonz