openscad/openscad #4677: Difference in minkowski in manifold

johnsonm, 2023-06-23:
In at least one case, a `minkowski` on a `union` of disconnected shapes renders differently in the stable release and 2023.06.14 with manifold feature enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable manifold
2. Download Chamfered_text_2_V6.scad from https://www.thingiverse.com/thing:3094095 (github won't let me attach it, even as a .txt file)
3. F6 to render
4. In stable, or development with neither fast-csg or manifold enabled, or development with only fast-csg and not manifold enabled, (eventually) shows both "A" and "B"; in manifold, shows only "B"

**Expected behavior**

Showing both "A" and "B" with manifold feature enabled.

Alternatively, an error message in the console describing that there is a problem with the model.

**Environment and Version info (please complete the following information):**
 - OS: Fedora 37
 - System: Intel PC 64-bit
 - OpenSCAD Version 2023.06.14 AppImage downloaded from openscad.org

**Library & Graphics card information**

```text
OpenSCAD Version: 2023.06.14.ai15466 (git a144370a6)
System information: Linux 6.3.6-100.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Jun 5 15:44:30 UTC 2023 x86_64 Fedora Linux 37 (Workstation Edition) 8 CPUs 23.12 GB RAM
User Agent: OpenSCAD/2023.06.14.ai15466 (git a144370a6) (Linux x86_64; Fedora Linux 37 (Workstation Edition))
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.4
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20210627
libzip version: 1.5.1
fontconfig version: 2.13.1 (runtime: 2.14.1)
freetype version: 2.10.1 (runtime: 2.12.1)
harfbuzz version: 2.6.4 (runtime: 5.2.0)
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /tmp/.mount_OpenSC4O3NKC/usr/bin
Documents Path: /home/johnsonm/.local/share
User Documents Path: /home/johnsonm/Documents
Resource Path: /tmp/.mount_OpenSC4O3NKC/usr/share/openscad
User Library Path: /home/johnsonm/.local/share/OpenSCAD/libraries
User Config Path: /home/johnsonm/.config/OpenSCAD
Backup Path: /home/johnsonm/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/johnsonm/.local/share/OpenSCAD/libraries
/tmp/.mount_OpenSC4O3NKC/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/X11/fonts/Type1
/usr/share/X11/fonts/TTF
/home/johnsonm/.local/share/fonts
/home/johnsonm/.local/share/flatpak/exports/share/fonts
/var/lib/flatpak/exports/share/fonts
/usr/local/share/fonts
/usr/share/fonts
/home/johnsonm/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
[... 387 lines cut ...]
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

pca006132, 2023-11-19:
It turns out there is nothing to do with minkowski. The extruded height is too small comparing with other dimensions, so it may get removed by manifold by considering it as something degenerate. Changing the height to 0.001 fixes the problem.
In general there is no way around this issue if we use floating point arithmetic.

kintel, 2023-11-19:
Yeah, this wasn't designed in the most elegant way.
Two takeaways from this:
1. minkowski of 2D and 3D objects should be pretty well-defined and worth exploring
2. We should probably collect some notes/FAQ on CGAL vs. Manifold, especially once we make Manifold the default

closed 2023-11-19 by kintel

jordanbrown0, 2023-11-19:
0.001 in a typical model with dimensions less than 1000 still has about ten orders of magnitude left before you run out of 64-bit floating point accuracy.

johnsonm, 2023-11-19:
Just to be clear: I didn't create the design; I merely noted that it acted differently depending on whether Manifold was in use. Sorry my characterization was misleading relative to the cause.

pca006132, 2023-11-19:
> 0.001 in a typical model with dimensions less than 1000 still has about ten orders of magnitude left before you run out of 64-bit floating point accuracy.

We use f32. We planned for supporting 64 bits but it will take some time.

pca006132, 2023-11-19:
And this is not about running into floating point issues, this is due to our simplification pass removing the object. We have a bigger epsilon to work with comparing to the error in f32.

If there is no simplification pass, the boolean results will quickly become something mostly containing degenerate triangles.

johnsonm, 2023-11-20:
I left a comment on the original design noting the change that users will need for this to work with Manifold OpenSCAD.

As the OP, my goal wasn't to get the design to work, it was only to surface a difference between modes for evaluation. It definitely makes sense to me to document this as a difference. If there is a reasonable rule of thumb for epsilon values likely to be robust for Manifold, including that in such documentation would of course be helpful.

Thanks for taking the time to isolate the cause of the variance, as well as the far more extensive work integrating it into OpenSCAD and helping maintain Manifold! :heart:

pca006132, 2023-11-20:
I am not entirely sure about the exact number, but iirc it is something like 1e-5 multiplied by the max abs coordinate, so smaller features may be optimized away by manifold.

Generally 32 bit floating point can give you 1e-7 epsilon, but we need some wiggle room for our processing as well. We can probably fine tune it a bit but it may not worth the effort. We planned to support 64 bit floating points for more complicated models, but there will be some performance penalty to using that and we are still thinking what is the best way to enable it without users facing large performance regression.