openscad/openscad #4680: Text disappears from editor after selection

bjohas, 2023-06-25:
**Describe the bug**

This bug concerns the editor.

Sometimes when I select text with the mouse (either double click or left-button-held+swipe) the text disappears. This is intermittent, but I would guess happens about 10% of the time.

**Environment and Version info (please complete the following information):**
- OpenSCAD version 2021.01 
- Ubuntu

bjohas, 2023-07-04:
(If you accept bounties, I'd be happy to make a bounty available.)

t-paul, 2023-07-04:
I've seen that, or very similar issue. I never could really figure out what combination makes it happen but it sometimes looked like connected to having a selection when switching windows and then pressing a modifier key like Shift that then wrongly cleared the selection as if a "normal" key had been typed.

I did not see it lately, so it might have been fixed in the editor component. Although I don't see an entry in the ChangeLog that sounds related.