openscad/openscad #4682: Column editor

KungPhoo, 2023-06-27:
When selecting a column and typing text, only one character is added.
How to reproduce the effect:
New document.
Press 3xenter. Hold Alt+Shift. Press cursor up 3x. Release Shift+Alt. Type 'x'
I expected to see:
x
x
x

t-paul, 2023-06-27:
We use [QScintilla](https://www.riverbankcomputing.com/software/qscintilla/download) as editor component. A feature like that would need to be implemented there.

KungPhoo, 2023-06-27:
Scintilla is able of that. Notepad++ uses it, too and and the feature is included there.

t-paul, 2023-06-27:
Hmm, interesting. So we need to check if that is available in QScintilla too.

Scopeuk, 2023-06-27:
Further to this it appears our current behaviour is very strange.
Selecting multiple rows of text (hold alt and draw vertical cursor with mouse) and then typing results in text typing backwards on the bottom selected row.