openscad/openscad #4685: Manifold: Hit F6 twice causes assertion failure

kintel, 2023-06-28:
**Describe the bug**
With Manifold enabled, hit F6 twice -> 
```
Assertion failed: (false && "not implemented"), function visit, file RenderStatistic.cc, line 295.
```

**To Reproduce**
Steps to reproduce the behavior:

1. Enable Preferences->Features->Manifold
2. Use the following design (but pretty much any design should do):
```
union() {
  cube();
  sphere();
}
```
3. Hit F6. Then hit F6 again.
4. OpenSCAD exits with the following console message:
```
Assertion failed: (false && "not implemented"), function visit, file RenderStatistic.cc, line 295.
```

**Environment and Version info (please complete the following information):**
 - OS: macOS
 - OpenSCAD 2023.06.27 commit `462f97daaa`

t-paul, 2023-08-29:
I can't reproduce with 2023.08.25 on Debian/Testing.

There was an update to Manifold in https://github.com/openscad/openscad/pull/4701 on Jul 9th.

thehans, 2023-08-29:
Not sure the version of manifold used is relevant in this case.

Looks like our code has an unconditional assertion claiming "not implemented", even though there is some code above it:
https://github.com/openscad/openscad/blame/05b962168cce9bc30b967a19514c490b495bd732/src/RenderStatistic.cc#L295

Should the assertion simply be removed? ping @ochafik

@t-paul I guess the the nightlies are built with `-DNDEBUG` compile flag (disabled assertions)?  That would explain no reproducability.

Still seems strange that it would require two F6 to trigger it.  I'm guessing the first round of F6 somehow doesn't return that specific `ManifoldGeometry` type, but the 2nd time does?

```
Parsing design (AST generation)...
Saved backup file: /home/hans/.local/share/OpenSCAD/backups/unsaved-backup-YaYgPcrb.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using Manifold...
Geometries in cache: 3
Geometry cache size in bytes: 2560
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.022
Top level object is a 3D object:
   Facets:         44
Rendering finished.
 
Parsing design (AST generation)...
Saved backup file: /home/hans/.local/share/OpenSCAD/backups/unsaved-backup-YaYgPcrb.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using Manifold...
Geometries in cache: 3
Geometry cache size in bytes: 2560
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.016
   Top level object is a 3D object (manifold):
   Status:     NoError
   Genus:      0
   Vertices:       24
   Facets:         44
   BBox.min:   -0.809016, -0.951056, -0.866025
   BBox.max:   1.000000, 1.000000, 1.000000
Rendering finished.
```

amdavidson, 2023-08-30:
I can reproduce on Debian Bookworm with OpenSCAD freshly built from commit 05b962168cce9bc30b967a19514c490b495bd732

I couldnt get `cmake` to take `-DNDEBUG` as a flag... So I wasn't able to check that as a fix.

This is definitely a frustrating bug, it occurs on several of my personal models.

amdavidson, 2023-08-30:
Thanks! I rebuilt with `-DCMAKE_BUILD_TYPE=Release` and that does seem to have fixed it in some quick testing.

thehans, 2023-08-30:
You're welcome.

Regarding this issue in general:
I would suggest we just remove the assert line, which seems like something accidentally leftover after some implementation (of Render Statistics on `ManifoldGeometry` type) was added.

Then edit this issue (or maybe better, close this and create a separate issue) to say that 1st and 2nd render take different code paths (presumably due to returning different geometry types).

chotchki, 2023-09-23:
Pull request #4751 was merged and should close this issue.

t-paul, 2023-09-23:
Ah, thanks for the heads-up. I missed that the PR did not automatically link here, it needs something like "closes #4685" for that.

Thanks again for solving the issue.

closed 2023-09-23 by t-paul