openscad/openscad #4691: 3mf layer change feature

mofosyne, 2023-07-01:
**Is your feature request related to a problem? Please describe.**
I sometimes make designs where you manually change filament at different layer points.

**Describe the solution you'd like**
Would be handy if we can encode that to a 3mf export, rather than having to manually set it in prusaslicer

**Describe alternatives you've considered**
Manually entering the layer change.

**Additional context**
3mf stores these information in a metadata section

UBaer21, 2023-07-01:
You can store all sort of things in 3mf as it is a zip container. 
But encoding a filament change per layer would mean that openSCAD know about your layerheight, unit system and more important the slicer to encode this.  There is no standard for the meta data, so you probably need a full profile to be accepted by prusa - doesn't work for cura.
However what you can do is making a two part model with each half printed with a different extruder. Having two objects in a 3mf can be done with lazy-union active but is more of a byproduct and experimental.

jordanbrown0, 2023-07-02:
We should look into how practical it is to have the OpenSCAD program supply arbitrary data to be added to the export.  Might need to be export-type-specific, but that would still be valuable.

mofosyne, 2023-07-03:
I wonder how one would represent that inline in OpenSCAD.

In Jekyll the metadata is represented like

```
---
layout: post
title: Blogging Like a Hacker
---

Blog Content here...
```

Would it be sane to add such metadata like such? (Plus might be useful to start encoding stuff like Title, Author, etc... Or recommended printing settings etc...)

t-paul, 2023-07-03:
My ancient idea collection: https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

gsohler, 2023-07-03:
I like it!

OpenSCAD does  have (kind of) Annotations already. Characters '%' !' and
'#' are which control the display of modules.
class "ModuleInstantiation"  would be the perfect place to store such
information internally.
Depending on the output format, special keywords could map to  certain
places.
I'd vote for a keyword to occupy the 16 vacant bits of each STL triangle.

UBaer21, 2023-07-03:
Windows 3mf [3Dprinting API ](https://learn.microsoft.com/en-us/windows/uwp/devices-sensors/3d-generate-3mf) uses 
```
model.Metadata.Add("Title", "Cube");
model.Metadata.Add("Designer", "John Smith");
model.Metadata.Add("CreationDate", "1/1/2016");
```
Having something like that to be included in the 3mf would be nice .. also design version infos or even the thumbnail.

mofosyne, 2023-07-03:
According to https://3mf.io/3mf-specification/ this is all the current default metadata according tot he 3MF spec:

| Name | Comment |
|---|---|
| Title | A title for the 3MF document |
| Designer | A name for a designer of this document |
| Description | A description of the document |
| Copyright | A copyright associated with this document |
| LicenseTerms | License information associated with this document |
| Rating | An industry rating associated with this document |
| CreationDate | The date this documented was created by a source app |
| ModificationDate | The date this document was last modified |
| Application | The name of the source application that originally created this document |

Looks like layer change metadata is specific to prusaslicer. It is interesting that they didn't adopt this style of metadata  `<metadata name="vendor1:CustomMetadata1" preserve="1">CE8A91FB-C44E-4F00-B634-BAA411465F6A</metadata>` that was suggested in the main 3mf spec for vendor specific metadata.

E.g.

```xml
<metadata name="prusaslicer:custom_gcodes_per_print_z" preserve="1"><code print_z="6.5" type="0" extruder="1" color="#8BFA3D" extra="" gcode="M600
G1 E0.5 F1500 ; prime after color change"/>
<mode value="SingleExtruder"/></metadata>
```

# Content of a prusa generated test file...

## testfile.3mf/3D/3dmodel.model

```xml
<?xml version="1.0" encoding="UTF-8"?>
<model unit="millimeter" xml:lang="en-US" xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" xmlns:slic3rpe="http://schemas.slic3r.org/3mf/2017/06">
 <metadata name="slic3rpe:Version3mf">1</metadata>
 <metadata name="Title">para_box</metadata>
 <metadata name="Designer"></metadata>
 <metadata name="Description">para_box</metadata>
 <metadata name="Copyright"></metadata>
 <metadata name="LicenseTerms"></metadata>
 <metadata name="Rating"></metadata>
 <metadata name="CreationDate">2023-07-03</metadata>
 <metadata name="ModificationDate">2023-07-03</metadata>
 <metadata name="Application">PrusaSlicer-2.6.0</metadata>
 <resources>
  <object id="1" type="model">
   <mesh>
    <vertices>
     <vertex x="9.09291458" y="-8.74345016" z="5"/>
     <vertex x="9.16143608" y="-8.43690681" z="-5"/>
...
```

## testfile.3mf/Metadata/Prusa_Slicer_custom_gcode_per_print_z.xml

```xml
<?xml version="1.0" encoding="utf-8"?>
<custom_gcodes_per_print_z>
<code print_z="6.5" type="0" extruder="1" color="#8BFA3D" extra="" gcode="M600
G1 E0.5 F1500 ; prime after color change"/>
<mode value="SingleExtruder"/>
</custom_gcodes_per_print_z>
```

jordanbrown0, 2023-07-03:
I don't like the special-markup style, e.g. with "@"; to me it reeks of the feature being an afterthought.  Yes, it's common in other languages... where it was an afterthought.

It seems to me that there are two distinct metadata cases:
* metadata associated with a particular subassembly
* metadata associated with the model as a whole

These present different problems.

Subassembly metadata (e.g. color, material, maybe local printing options) requires that subassemblies maintain their identity into the final output, whereas we currently union everything together into one object before export.  (Lazy union doesn't really help there, because subassemblies may go deeper than that; a subassembly might be fed into something other than a union before it gets to the top.)

Whole-model metadata (e.g. title, default output file) is global.  We'd need to understand what the behavior is if you try to set it more than once, or if you try to set it in a module instead of at the top level.  Maybe some of those (e.g. title, author) can be set by a module, and are overridden by higher-level settings, but what happens if two modules set the same thing and the top level doesn't set it?

mofosyne, 2023-09-08:
Just noticed this issue discussing about metadata https://github.com/openscad/openscad/issues/1604 . So just mentioning it here in case this idea of a front matter openscad metadata format for 3mf layer change feature gets baked, so we can close this issue ticket as well.