openscad/openscad #4692: Obstrucated view in preview when applying `difference()` or `intersection()` to STL model

Jorengarenar, 2023-07-02:
**Describe the bug**
I've noticed that removing some part of imported STL model causes some obstrucation in the preview.
What's interesting, there are no error messages; the render is correct and printing goes well.
Just the preview has problems, making working a tad harder.

**To Reproduce**

1. Download some STL model (e.g. https://www.thingiverse.com/thing:1660164/files)
2. Import it
3. Create a cube
4. Apply `difference()` or `intersection()` between the two
5. Rotate the preview around the object

**Expected behavior**
Viewing the uncutted part normally 

**Code reproducing the issue**
```
difference() {
    import("Master_System_to_MegaDive_Adaptor_Case_v1.0.stl", center=true);
    translate([0,30,23]) color([0,0,0.4,0.3]) cube([150,100,20], center=true);
} /* <-- cut model in half */
```

**Screenshots**
Upper half, view from bottom:
[image]
Bottom half, view from bottom:
[image]
Bottom half, view from top to the side:
[image]
Correctly rendered:
[image]

**Environment and version info:**
 - **OS:** Arch Linux
 - **System:** x86_64, 16 GB of RAM, Nvidia RTX 2070
 - **OpenSCAD versions:** 2021.01 (repository), 2023.06.30.ai15629 (AppImage from homepage)

**Library & Graphics card information**
```
OpenSCAD Version: 2023.06.30.ai15629 (git bea27559b)
System information: Linux 6.3.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 21 Jun 2023 20:46:20 +0000 x86_64 Arch Linux 12 CPUs 15.56 GB RAM
User Agent: OpenSCAD/2023.06.30.ai15629 (git bea27559b) (Linux x86_64; Arch Linux)
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.4
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20210627
libzip version: 1.5.1
fontconfig version: 2.13.1 (runtime: 2.14.2)
freetype version: 2.10.1 (runtime: 2.13.0)
harfbuzz version: 2.6.4 (runtime: 7.3.0)
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /tmp/.mount_OpenSCZyCsfa/usr/bin
Documents Path: /home/joren/.local/share
User Documents Path: /home/joren
Resource Path: /tmp/.mount_OpenSCZyCsfa/usr/share/openscad
User Library Path: /home/joren/.local/share/OpenSCAD/libraries
User Config Path: /home/joren/.local/config/OpenSCAD
Backup Path: /home/joren/.local/share/OpenSCAD/backups
[... 413 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

kintel, 2023-07-02:
When importing a complex STL in preview mode, you need to use the `convexity=` parameter to indicate how concave your imported design is. See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry#Convexity

In your case `import("Master_System_to_MegaDive_Adaptor_Case_v1.0.stl", center=true, convexity=5);` should do the trick.

closed 2023-07-02 by kintel

Jorengarenar, 2023-07-02:
Thanks!
I've must have missed this bit of information.
Now it works just fine :)