openscad/openscad #4693: dark mode not working win 10...

jmerc77, 2023-07-02:
**Describe the bug**
recently after #4130, I realized i did not have dark mode properly enabled. so i enabled it but openSCAD fails to use it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit (i7-4790 CPU @ 3.60GHz, 16gb ram, 64-bit operating system, x64-based processor)
 - OpenSCAD Version 2021.01 from official web side (openscad.org). will test latest and report later. i try to keep up to date.

**Library & Graphics card information**

`OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 8 CPUs 15.94 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: P:/Program Files/OpenSCAD
Documents Path: C:\Users\Owner\Documents
User Documents Path: C:\Users\Owner\Documents
Resource Path: P:/Program Files/OpenSCAD
User Library Path: C:/Users/Owner/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Owner\AppData\Local/OpenSCAD
Backup Path: C:/Users/Owner/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/Owner/Documents/OpenSCAD/libraries
P:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/Owner/.local/share/fonts
C:/Users/Owner/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.6.0 NVIDIA 536.23
GL Renderer: NVIDIA GeForce GTX 1660 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
[... 383 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control`

**Additional context**
unsure on win 11 yet. but i am able to test that on my laptop (my other pc). my desktop w/ win 10 is 7-ish yrs old but in good shape. however, i am unable to run win 11 by upgrade due to old cpu and no tpm on desktop.

jmerc77, 2023-07-02:
ok. 2023... version isnt in stable release yet? will try that when it is.

jmerc77, 2023-07-02:
well tis is weird. I'm having the same issue with Prusaslicer as well. at least it has a manual option that half way works (top bar briefly stays light grey). makes me wonder if its a system bug. what's interesting is that, both are open source c++ based apps.

jmerc77, 2023-07-02:
incase its helpful:
win build version 19045.3155

UBaer21, 2023-07-02:
openSCAD uses his own color schemeso
You have two different color scheme options in »preferences«

- for the 3D view 
- for the Editor ⇒ change "color highlight syntax"  you can select between multiple dark/night options

you also can alter the json or create your own schemes ( folder color-schemes )

jmerc77, 2023-07-02:
knew those but not the issue i was refering to. i wa referring to the rest of the window. for example, the top bar would be dark as well as the status and menus. the win explorer on the left is a good example. chrome also works like that. i see the edit as i type. and agree it could be a missing feature maybe.

t-paul, 2023-08-30:
See https://www.qt.io/blog/dark-mode-on-windows-11-with-qt-6.5 which has some notes for qt5 too.

natescherer, 2023-09-08:
Confirmed that even on the latest dev snapshot colors behave the same way on Win 11 as the latest stable version. (Re: your question in the other linked issue thread, @t-paul)

natescherer, 2023-09-08:
Running the executable as
```
openscad.exe -platform windows:darkmode=2
```
does somewhat fix this, although the theme is more high-contrast than dark mode.

[image]

Nikorasu, 2023-12-19:
This is also a HUGE issue for me in Linux. No matter what I do, I cannot get the window-background to use Dark Mode.
The last version that had a functional dark mode for me was 2019.05, worked perfectly but it's so old now, and even the newer development versions I can't seem to find a way.
This is what it looks like now:
[image]
I'm starting to wish it didn't rely on the OS, and just gave us an option to manually configure it somewhere. :/

jmerc77, 2023-12-23:
> Running the executable as
>
> ```
> openscad.exe -platform windows:darkmode=2
> ```
>
> does somewhat fix this, although the theme is more high-contrast than dark mode. [...]

not working on my windows 11 laptop. i was having the problem on both windows 11 and windows 10. my laptop has an i7 and same ram amount but the cpu is a newer gen.

FormulaCarbon, 2024-01-03:
> Running the executable as
> 
> `openscad.exe -platform windows:darkmode=2`
> does somewhat fix this, although the theme is more high-contrast than dark mode.

does not work on my win11 laptop either, on latest stable build (2021.01)

system info:
Intel Core i7-12650H CPU
NVIDIA RTX 4060 Laptop GPU
32gb 5600mhz RAM
Windows 11 Version 22H2

when i run the command in Command Prompt in `C:/Program Files/OpenScad` this happens
`C:\Program Files\OpenSCAD>Unknown option "darkmode=2"
QString::arg: Argument missing: , C:/Program Files/OpenSCAD/windows:darkmode=2`

and then launches normally without dark mode.

BertrandSahel, 2024-02-19:
For me on Linux Ubuntu 23.10, the last Appimage working fine for the dark mode in editor is the 2023-09-25 version.

BertrandSahel, 2024-03-04:
Today in the Linux OpenSCAD-2024.03.03.ai18729-x86_64.AppImage , the remaining editor darkmode is fixed !!!!

TheElbinho, 2024-10-22:
Still broken in current nightly (2024.10.17) on Windows 11

Launching with a system-wide dark setting, OpenSCAD isn't dark. Launching with "-platform windows:darkmode=1" gives only a dark title bar, launching with darkmode=2 remains a possible workaround with a high-contrast look

pozix604, 2025-06-04:
`OpenSCAD-2025.06.03.ai25586-x86_64.AppImage` dark mode does not work on Debian 12 Gnome.