openscad/openscad #4694: OpenSCAD does not show installed size in the Windows apps control panel

meyersjd, 2023-07-03:
**Describe the bug**
OpenSCAD does not show its size in the apps control panel of windows

**To Reproduce**
Steps to reproduce the behavior:
1. Install OpenSCAD
2. Go to the App control panel and find OpenSCAD
3. No app size is listed.

**Expected behavior**
App size should be shown

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 and 11
 - System: Intel PC
 - 2021.01 OPENSCAD.org

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

fkkjohn, 2023-07-04:
This could help... It's not really a bug as much as an optional parameter:

https://devblogs.microsoft.com/oldnewthing/20040709-00/?p=38493

If I can find where in the code the install file is setting these parameters, I'll fix it.

kintel, 2023-07-04:
When opening a Pull Request, we will automatically build an installer on CircleCI. That might be the easiest way of testing this.

fkkjohn, 2023-07-04:
I also tried adding the registry key manually, but I don't think it made a difference on my end. Looks like it was already estimating the install size, because explorer reports 68750kb, but my Add/Remove window shows OpenSCAD with a generic icon and (remove only) as being 67.1Mb.

t-paul, 2024-04-24:
Dev snapshot shows "Humanity" as publisher

[image]

thehans, 2024-04-24:
It looks like we were previously writing "Publisher" and other info directly in the registry from installer.nsi here:
https://github.com/openscad/openscad/blob/185c1f59270b7e91764f06b1557b003b66fef38f/scripts/installer.nsi#L25-L32

However, since our windows installers are now created via [CPack NSIS generator](https://cmake.org/cmake/help/latest/cpack_gen/nsis.html) this `.nsi` file and others are no longer in use.  
CPack seems to automatically copy many of these values from various other package settings, though I don't know how many others remain unconfigured or misconfigured.

"Humanity" is apparently the default for [CPACK_PACKAGE_VENDOR](https://cmake.org/cmake/help/latest/module/CPack.html#variable:CPACK_PACKAGE_VENDOR), which I guess gets placed into the "Publisher" registry key.

Here are the NSIS docs for the various [Optional values](https://nsis.sourceforge.io/Add_uninstall_information_to_Add/Remove_Programs#Optional_values) including `EstimatedSize` and `Publisher` and a couple dozen others.

I'm not seeing anything in cpack docs about `EstimatedSize` or any size setting, but NSIS docs have a whole section dedicated to [Computing EstimatedSize](https://nsis.sourceforge.io/Add_uninstall_information_to_Add/Remove_Programs#Computing_EstimatedSize)

In order to handle any additional commands and registry keys I think we would need to append `WriteRegStr`  lines to `CPACK_NSIS_EXTRA_INSTALL_COMMANDS` here:
https://github.com/openscad/openscad/blob/185c1f59270b7e91764f06b1557b003b66fef38f/CMakeLists.txt#L1294-L1297

And then I guess corresponding `DeleteRegKey` lines should go in `CPACK_NSIS_EXTRA_UNINSTALL_COMMANDS`

Example `WriteRegStr` usage from NSIS docs
```
WriteRegStr HKLM "Software\Microsoft\Windows\CurrentVersion\Uninstall\Product" \
                 "DisplayName" "Application Name"
```
(backslashes will need extra escaping to go inside cmake strings)

As I understand it, the appropriate key path (in place of `"Software\Microsoft\Windows\CurrentVersion\Uninstall\Product"` above) could be read from this cpack variable?: [CPACK_PACKAGE_INSTALL_REGISTRY_KEY](https://cmake.org/cmake/help/latest/module/CPack.html#variable:CPACK_PACKAGE_INSTALL_REGISTRY_KEY) 
(so that it corresponds with the specific installed version/directory/instance)

Lastly, not sure if relevant, but there is also this build file https://github.com/openscad/openscad/blob/master/cmake/Modules/openscad_cpack.cmake.in
which is processed at *cpack-time* (giving it a different context of variables to work with which I don't remember the details of, off the top of my head).

t-paul, 2024-04-25:
Ok, I think #5102 now covers everything (see screenshots). Anyone wants to give this a try?

Download at: https://app.circleci.com/pipelines/github/openscad/openscad/4892/workflows/c17bebe1-9681-4f9f-87a0-79529f3218da/jobs/19178/artifacts

PR #5102 (2024-04-26): Fix win installer properties
Changes:
* Move NSIS files to `cmake/nsis`
* Use external scripts for extra commands
* Add vendor/publisher/... to .exe properties
* Add installed size
* Start menu entry created
* Generate version resource file
* Folder/Registry names "OpenSCAD" or "OpenSCAD (Nightly)"

[image]
<br>
[image]

Fixes #4694.
closed 2024-04-26 by t-paul