openscad/openscad #4698: Re-indent / auto-indent all code

bjohas, 2023-07-04:
**Is your feature request related to a problem? Please describe.**

Code indentation helps me read and understand the code. However, it seems that there's no function that re-indents all existing code in the editor.

**Describe the solution you'd like**

I'd like a menu function / keyboard shortcut to reindent all code in the editor (or all selected code).

**Describe alternatives you've considered**

Manual indentation (tedious...)

(Happy to provide a bounty...)

jordanbrown0, 2023-07-04:
Note #3565 .

I indent as I type (and existing autoindent helps there), which means that normally the only bulk indentation work I need to do is to move a block left or right, and select+tab already does that.

t-paul, 2023-07-04:
Ah, thanks, closing as duplicate. @bjohas you can subscribe to the other issue for updates.

We do have bountysource linked, and there's a small amount of success stories. I'm not sure this topic is a good candidate. Bounties seem to mostly work for smaller GUI changes, not so much for the more tricky ones.

closed 2023-07-04 by t-paul

bjohas, 2023-07-04:
Thanks for the messages.

Yeah, I indent as I type as well, but at least on linux the auto indentation isn't great, so I end up correcting indentation. In many IDEs there's a default formatting function that fixes indentation. That's what I was looking for. Will look out for progress on other issue

gsohler, 2023-07-05:
I would also prefer to auto-indent available!
Of course auto-indent help a little bit. But when I develop a model, i
usually start with something like a cube and then I insert  CSG ops in each
level. This is the time, where auto-indent does not help anymore.
A shortcut to re-ident would definitely help