openscad/openscad #4700: Add "copy" argument to `mirror` and other transforms

piegamesde, 2023-07-09:
**Is your feature request related to a problem? Please describe.**
I've made the following helper modules some time ago, and notice myself copying them over to almost every OpenSCAD project I do eventually:

```openscad
module copy_mirror(axis) {
    children();
    mirror(axis)
    children();
}

module copy_translate(axis) {
    children();
    translate(axis)
    children();
}
```

Basically, it is really a common thing to expect a "mirror" modifier to create a copy of the original, instead of simply flipping it. 

**Describe the solution you'd like**
Add a `copy` parameter to `mirror`, that if set to true creates a copy of the original instead of just mirroring.

**Describe alternatives you've considered**
This could be extended to other transforms as well. I've personally used `copy_translate` a couple of times, although one could argue that then most transforming operations should have it for consistency.

**Additional context**
Inkscape, Blender and other tools sometimes offer a similar option too.

Also, please excuse if this has been discussed elsewhere already, searching for duplicates is pretty hard on this topic.

jordanbrown0, 2023-07-09:
As you've noted, it's easy to write a module to do this.  (Mine is called "mirrordup".)

Mostly we don't add built-in features to support things that are easy to write modules for.

piegamesde, 2023-07-09:
How is `center` different here? `center` is present in basically all primitives, even though it would be easy to implement as a module. I feel towards `copy` for transformations in a similar way.

jordanbrown0, 2023-07-09:
You're not wrong.  I suspect that "center" came in early and that there's been a bit of a shift in attitude in the interim, realizing that there's an overwhelming number of minor improvements that could be made and discouraging new additions to the core program when they can be done using modules and libraries.

I'm not one of the approvers - I'm just a user and occasional developer - so all I have is an opinion.

You need (as with all changes):
* Somebody who wants to spend the time to do the actual development.
* Approval from one of the people who is authorized to approve changes.

pipatron, 2023-07-10:
> How is `center` different here? `center` is present in basically all primitives, even though it would be easy to implement as a module. I feel towards `copy` for transformations in a similar way.

How could you possibly do that without knowing the type and size of the child? You would need duplicate modules for each primitive.

I've just started standardizing on the absolutely awesome [BOSL2](https://github.com/BelfrySCAD/BOSL2/wiki/TOC) OpenSCAD library, which gives all those features with a simple one-line include. I wouldn't hold my breath for new features to be added to the language.

t-paul, 2023-08-29:
Closing the issue as there is no plan for adding this directly into the language.

closed 2023-08-29 by t-paul