openscad/openscad #4704: Allow to get syntax error details via command line

WolfgangFahl, 2023-07-21:
**Is your feature request related to a problem? Please describe.**
I am trying to wrap OpenScad similar to what https://github.com/nickc92/ViewSCAD did with jupyter but this time using
https://nicegui.io/ and it's https://threejs.org/ scene wrapper. See https://github.com/WolfgangFahl/nicescad

When calling OpenScad via the command line the details of syntax error are now shown and therfore i can't e.g. higlight the input line that is the culprit.

**Describe the solution you'd like**
Allow to make syntax error details available via command line - preferrably in computer readble format such as json.

**Describe alternatives you've considered**
I can't think of any.

**Additional context**
[image]

jordanbrown0, 2023-11-29:
First, I assume that you mean "the details of the syntax error are *not* shown", right?

What details are you looking for?  The only thing that the GUI seems to have that the CLI doesn't have is the column.

Or is it just that you want the information in a more readily parsed form?

jordanbrown0, 2023-11-29:
Stuff I found:  In addition to the error message (which includes a line number), the GUI has the number of characters from the start of the file.  (Maintained by the lexer in `parser_error_pos`.)  `ScintillaEditor::highlightError()` converts that position into line and column.

WolfgangFahl, 2023-12-11:
ERROR: Parser error: syntax error in file ../../../../../tmp/tmpcgipmhrt/tmp.scad, line 4
Can't parse file '/tmp/tmpcgipmhrt/tmp.scad'!

doesn't have any hint on what the problem is e.g. "unknown keyword", "missing semicolon" or any other hint.