openscad/openscad #4706: Usability: Viewport-Control pane width min of 620 pixels is overly large

haikusw, 2023-07-26:
**Describe the bug**
Viewport-Control pane width cannot be resized to less than 620 pixels which is quite a bit larger than it needs to be.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a document
2. Make sure "Hide Viewport-Control" in the Window menu is not selected (no check mark)
3. Use the splitter to attempt to shrink the width of the Viewport-Control pane
4. Notice that it will not get smaller than 620 pixels, despite there being lots of extra space in all the editable text fields etc.

**Expected behavior**
The Viewport-Control pane should be shrinkable to some reasonable minimum based on the controls it contains.  The Customizer, for example, can be made down to something like 284 pixels wide.  I think the Viewport-Control might need slightly more - roughly 350 pixels - as a minimum width, but I can't test that.

**Screenshots**

min width screenshot showing all the unnecessary extra space

[image]

**Environment and Version info (please complete the following information):**
 - OS: macOS 13.4.1 (c) 
 - System: 13" M1 MacBook Pro, 49" external monitor
 - OpenSCAD 2023.04.17 (git 27227d9d7), downloaded dev snapshot.

**Library & Graphics card information**
OpenSCAD Version: 2023.04.17 (git 27227d9d7)
System information: macOS 13.4.1 arm64 MacBookPro17,1 8 CPUs 16.00 GB RAM 
User Agent: OpenSCAD/2023.04.17 (git 27227d9d7) (macOS 13.4.1 arm64 MacBookPro17,1)
Compiler: Clang "13.0.0 (clang-1300.0.29.30)"
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.7
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.71.0
lodepng version: 20210627
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: /Applications/OpenSCAD (Dev Snapshot).app/Contents/MacOS
Documents Path: /Users/tyler/Documents
User Documents Path: /Users/tyler/Documents
Resource Path: /Applications/OpenSCAD (Dev Snapshot).app/Contents/Resources
User Library Path: /Users/tyler/Documents/OpenSCAD/libraries
User Config Path: 
Backup Path: /Users/tyler/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/Users/tyler/Documents/OpenSCAD/libraries
/Applications/OpenSCAD (Dev Snapshot).app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/Users/tyler/.local/share/fonts
/usr/local/share/fonts
/usr/share/fonts
/Users/tyler/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/Users/tyler/Library/Fonts
/System/Library/Fonts/Supplemental
/Users/tyler/Library/Fonts/AnonymousPro
/Users/tyler/Library/Fonts/JetBrainsMono-2.242
/Users/tyler/Library/Fonts/bootleggedfonts
/Users/tyler/Library/Fonts/JetBrainsMono-2.242/fonts
/Users/tyler/Library/Fonts/bootleggedfonts/Calibri
/Users/tyler/Library/Fonts/JetBrainsMono-2.242/fonts/ttf
GLEW version: 2.2.0
OpenGL Version: 2.1 Metal - 83.1
GL Renderer: Apple M1
GL Vendor: Apple
RGBA(8888), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(888-1), depth(0), stencil(0)

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

**Additional context**
(none)

Thanks.

t-paul, 2024-04-26:
It's not possible to shrink the fields due to their min/max values being very large. It seems the only option is having separate horizontal and vertical layouts that optimize space usage:

[Screencast from 2024-04-26 03-02-27.webm](https://github.com/openscad/openscad/assets/1330241/54ff6248-92d3-42d7-a48e-3ba42eba830e)

haikusw, 2024-04-27:
That works much better!  Thank you.

haikusw, 2024-04-27:
Hmm.  What happens at 0:08?  Seems like the right edge slides off as if there's a horizontal scroll bar in the window (but there isn't).  That shouldn't happen, seems like.

[image]

t-paul, 2024-04-27:
I don't see a way to prevent that from happening. Blocking that interferes with the direction switching logic.

haikusw, 2024-04-27:
The behavior you have looks good in the video except for the horizontal scrolling.  I'd _love_ to have the ability to have it change to vertical layout so the Viewport-Control widget can get narrower (!).

I don't know Qt (sorry), but it doesn't seem like horizontally resizable should generally require horizontal scrollability? (it doesn't seem to in the code prior to your change).

Before your change, at least on the version I was reporting this issue against, the Viewport-Control pane is horizontally resizable down to the minimum size.  The right edge remains anchored to the right edge of the window during resizing and no horizontal scrolling occurs.

I'm unclear why it was necessary to add the `<widget class="QScrollArea" name="scrollArea">` widget to get the switch between layouts when the width gets too small to use the horizontal layout like you have now.

Thanks for working on this!

commit 04fd4e3b (2024-04-28): Merge pull request #5107 from openscad/viewport-control-layout

Support horizontal and vertical layouts (fixes #4706).
commit f7cd078f (2024-04-28): Support horizontal and vertical layouts (fixes #4706).
closed 2024-04-28 by t-paul