openscad/openscad #4707: Incorrect shadowing of function arguments when using function literals

jmickelin, 2023-07-26:
**Describe the bug**
It seems that parameters in function literals do not properly shadow parameters of the same name used earlier in the call stack for the expression that is being evaluated. This causes some expressions to be misidentified as infinite recursions.

**To Reproduce**
Run the following code:

```
parametric_add = function(f1, f2) function(t) t <= 1/2 ? f1(2*t) : f2(2*(t - 1/2));

f = function (t) 3*t;
g = function (t) t*t;
h = function (t) -t;

echo(parametric_add(f, g)(0)); // ECHO: 0
echo(parametric_add(h, parametric_add(f, g))(1)); // ERROR: Recursion detected calling function 'f2'
```

**Expected behavior**
The expression should evaluate as follows (assuming a normal-order evaluation; the final result should be the same regardless of evaluation order, however, since these are pure functions):
```
parametric_add(h, parametric_add(f, g))(1)
= (function (f1, f2) (function (t) t <= 1/2 ? f1(2*t) : f2(2*(t - 1/2))))(h, parametric_add(f, g))(1)
= (function (t) t <= 1/2 ? h(2*t) : parametric_add(f, g)(2*(t - 1/2)))(1)
= 1 <= 1/2 ? h(2*1) : parametric_add(f, g)(2*(1 - 1/2))
= parametric_add(f, g)(2*(1 - 1/2))
= parametric_add(f, g)(1)
= (function (t) t <= 1/2 ? f(2*t) : g(2*(t - 1/2)))(1)
= 1 <= 1/2 ? f(2*1) : g(2*(1 - 1/2))
= g(2*(1 - 1/2))
= g(1)
= 1 * 1
= 1
```

**Environment and Version info (please complete the following information):**
 - OS: Fedora release 37 
 - System: Intel 64-bit PC
 - OpenSCAD Version: `openscad-2021.01-11.fc37.src.rpm` installed from the `Fedora 37 - x86_64` repository (`https://mirrors.fedoraproject.org/metalink?repo=fedora-37&arch=x86_64`)

**Additional context**
There are several standard methods for naming parameter references within closures in functional languages to avoid name clashes after function application (also called α-conversion). I believe [de Brujin indices](https://en.wikipedia.org/wiki/De_Bruijn_index) is the most common, but the provided Wikipedia article discusses some alternatives.

jordanbrown0, 2023-07-31:
The problem is unrelated to parameter shadowing - it still happens if you give each function a unique name for its argument - or if you get rid of arguments entirely.

```
function wrap(f) = function() f();

g = function () 3;

echo(wrap(g)());
echo(wrap(wrap(g))());
```
The problem appears tied to tail recursion elimination.  If the wrapper function does anything with the value returned (and so defeats tail recursion elimination), the problem goes away.
```
function wrap(f) = function() f()+0;

g = function () 3;

echo(wrap(g)());
echo(wrap(wrap(g))());
```

henriquealrs, 2023-08-01:
On PopOS repo install v21.01 I get this issue. But I can't reproduce it compiling master from source

henriquealrs, 2023-08-01:
Yes, just build 2021.01 tag and the issue happened. Probably fixed in newer versions

jordanbrown0, 2023-08-02:
Agree.  Problem does not happen on 2023.06.06.  (Not that that particular build is special - it's just a recent build that I happen to have.)