openscad/openscad #4710: OpenSCAD 2021.01 won't open in Mac OS X 10.9

nathanpc, 2023-07-30:
**Describe the bug**
The application never opens and crashes instantly when trying to run under Mac OS X 10.9.5, even though the download page says that the requirements are macOS 10.9 or newer.

**To Reproduce**
Steps to reproduce the behavior:
1. Use an old machine running Mac OS X 10.9.5
2. Copy the application to the Applications folder
3. Open the application
4. Crash report window opens

**Expected behavior**
The application should've opened normally.

**Environment and Version info:**
 - OS: Mac OS X 10.9.5
 - System: MacBook Pro (Mid 2010) [MacBookPro6,1]
 - OpenSCAD Version 2021.01 from the DMG straight from the Download page of the website

**Library & Graphics card information**
```
QObject::startTimer: Timers can only be used with threads started with QThread
Fontconfig warning: "/Applications/OpenSCAD.app/Contents/Resources/fonts/fonts.conf", line 84: unknown element "blank"
OpenSCAD Version: 2021.01
System information: Mac OS X Version 10.9.5 (Build 13F1911) MacBookPro6,1 2 CPUs 8.00 GB RAM
User Agent: OpenSCAD/2021.01 (Mac OS X Version 10.9.5 (Build 13F1911) MacBookPro6,1)
Compiler: Clang "12.0.0 (clang-1200.0.32.28)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.3.7
CGAL version, kernels: 4.14.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.9
QScintilla version: 2.11.6
InputDrivers:
GLib version: 2.56.3
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.3.1
cairo version: 1.16.0
lib3mf version: 1.8.1
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/nathan/Documents
User Documents Path: /Users/nathan/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/nathan/Documents/OpenSCAD/libraries
User Config Path: /Users/nathan/Library/Application Support/OpenSCAD
Backup Path: /Users/nathan/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/nathan/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /Users/nathan/.local/share/fonts
  /Users/nathan/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/nathan/Library/Fonts
  /Library/Fonts/Microsoft

GLEW version: 2.1.0
OpenGL Version: 2.1 NVIDIA-8.24.17 310.90.9.05f01
GL Renderer: NVIDIA GeForce GT 330M OpenGL Engine
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: Cocoa / CGL
PNG generator: Core Foundation
OS info: Mac OS X 10.9.5 (x86_64 kernel)
Machine: 64-bit
```

**Additional context**
Mac OS X crash report:

```
Process:         OpenSCAD [560]
Path:            /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:      org.openscad.OpenSCAD
Version:         2021.01 (2021.01.31)
Code Type:       X86-64 (Native)
Parent Process:  launchd [178]
Responsible:     OpenSCAD [560]
User ID:         501

Date/Time:       2023-07-30 11:03:40.238 +0100
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  884F1FD8-9248-A2A5-30E2-00ADBB1A5FA1

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[QNSPanel setTitlebarAppearsTransparent:]: unrecognized selector sent to instance 0x7f8822e22100'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff92d8925c __exceptionPreprocess + 172
1   libobjc.A.dylib                     0x00007fff888ece75 objc_exception_throw + 43
2   CoreFoundation                      0x00007fff92d8c12d -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
3   CoreFoundation                      0x00007fff92ce7272 ___forwarding___ + 1010
4   CoreFoundation                      0x00007fff92ce6df8 _CF_forwarding_prep_0 + 120
5   libqcocoa.dylib                     0x0000000106884a5d qt_plugin_instance + 64749
6   libqcocoa.dylib                     0x000000010687def2 qt_plugin_instance + 37250
7   libqcocoa.dylib                     0x000000010687d424 qt_plugin_instance + 34484
8   libqcocoa.dylib                     0x0000000106877c64 qt_plugin_instance + 12020
9   QtGui                               0x0000000103edf66e _ZN14QWindowPrivate6createEby + 254
10  QtWidgets                           0x000000010392a978 _ZN14QWidgetPrivate10create_sysEybb + 1256
11  QtWidgets                           0x0000000103929618 _ZN7QWidget6createEybb + 328
12  QtWidgets                           0x000000010393c4b8 _ZN7QWidget10setVisibleEb + 456
13  QtWidgets                           0x0000000103afcb9d _ZN7QDialog10setVisibleEb + 125
14  QtWidgets                           0x0000000103afc343 _ZN7QDialog4execEv + 243
15  OpenSCAD                            0x00000001021c09af _Z3guiRNSt3__16vectorINS_12basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEENS4_IS6_EEEERKN5boost10filesystem4pathEiPPc + 1231
16  OpenSCAD                            0x00000001021c56b5 main + 15317
17  libdyld.dylib                       0x00007fff905ab5fd start + 1
18  ???                                 0x0000000000000001 0x0 + 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8ef58866 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff88db835c pthread_kill + 92
2   libsystem_c.dylib             	0x00007fff890c1b26 abort + 125
3   libc++abi.dylib               	0x00007fff8ef60f31 abort_message + 257
4   libc++abi.dylib               	0x00007fff8ef86952 default_terminate_handler() + 264
5   libobjc.A.dylib               	0x00007fff888ed30d _objc_terminate() + 103
6   libc++abi.dylib               	0x00007fff8ef841d1 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff8ef83c5b __cxa_throw + 124
8   libobjc.A.dylib               	0x00007fff888ecfa1 objc_exception_throw + 343
9   com.apple.CoreFoundation      	0x00007fff92d8c12d -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
10  com.apple.CoreFoundation      	0x00007fff92ce7272 ___forwarding___ + 1010
11  com.apple.CoreFoundation      	0x00007fff92ce6df8 _CF_forwarding_prep_0 + 120
12  libqcocoa.dylib               	0x0000000106884a5d 0x10686d000 + 96861
13  libqcocoa.dylib               	0x000000010687def2 0x10686d000 + 69362
14  libqcocoa.dylib               	0x000000010687d424 0x10686d000 + 66596
15  libqcocoa.dylib               	0x0000000106877c64 0x10686d000 + 44132
16  org.qt-project.QtGui          	0x0000000103edf66e QWindowPrivate::create(bool, unsigned long long) + 254
17  org.qt-project.QtWidgets      	0x000000010392a978 QWidgetPrivate::create_sys(unsigned long long, bool, bool) + 1256
18  org.qt-project.QtWidgets      	0x0000000103929618 QWidget::create(unsigned long long, bool, bool) + 328
19  org.qt-project.QtWidgets      	0x000000010393c4b8 QWidget::setVisible(bool) + 456
20  org.qt-project.QtWidgets      	0x0000000103afcb9d QDialog::setVisible(bool) + 125
21  org.qt-project.QtWidgets      	0x0000000103afc343 QDialog::exec() + 243
22  org.openscad.OpenSCAD         	0x00000001021c09af gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, boost::filesystem::path const&, int, char**) + 1231
23  org.openscad.OpenSCAD         	0x00000001021c56b5 main + 15317
24  libdyld.dylib                 	0x00007fff905ab5fd start + 1
```

kintel, 2023-08-16:
I'm pretty sure this is supposed to work on macOS 10.9, but I guess it wasn't tested properly on such an old system, so we may have missed something. This is the last release of OpenSCAD capable of running on macOS 10.9, and the release is > 2 years old, so we have somewhat limited motivation to fix it.

Are you still actively using macOS 10.9? I might find time to look at it sometime this week or next..

jmealo, 2024-02-26:
OpenSCAD opens for me, but the windows are blank. If I launch it from the terminal, it says that it cannot find the `cocoa` plugin for QT. I'm on MacOS 14.2.1 (23C71)

kintel, 2024-02-26:
@jmealo Could you open a separate issue for your macOS 14 problems? This issue is about backward compatibility with macOS 10.9

kintel, 2024-11-28:
Closing this as not planned: We don't have capacity to troubleshoot old versions of OpenSCAD on old OS'es.