openscad/openscad #4712: Use center to align to an extent or center of a primitive

kwikius, 2023-08-01:
**Is your feature request related to a problem? Please describe.**
To easily align to an extent of a primitive in a consistent way

**Describe the solution you'd like**

A feature that I have found useful for primitives. 

As well as a bool, center argument of primitive can be a vector (2d for 2d primitives, 3d for 3d primitives) x, y, z values of the argument can be  0,1,-1

-  0 means centre the primitive in that axis 
-  a value of 1 means align one extent of object with axis so object is in positive side of axis
-  a value of -1 means align one extent of object with axis so object is in negative side of axis.

```
// Following creates an arrow without needing to use translate for example
cylinder(h = 20,d1 = 10, d2 = 0, center = [0,0,1]);
cylinder(h = 50,d1 = 5,d2=3, center = [0,0,-1]);
```

```
// a dumbell centered around the yz plane
height = 20; // distance between inner extents of spheres
fuse = 1;  // extra length to fuse the connecting cylinder to the spheres at each end
rotate([0,90,0]){
  cylinder(d = 5, h = height + fuse, center = [0,0,0]);
}
translate([-height/2,0]){
  sphere(d = 20, center = [-1,0,0]);
}
translate([height/2,0,0,]){
  sphere(d = 20, center = [1,0,0]);
}
```

Current programs arent affected, since can be detected if arg is a bool or a vector or default

gsohler, 2023-08-01:
Hi kwikius,

you might have a look into libraries like

https://github.com/davidson16807/relativity.scad

I believe this is exactly what your proposed new openscad feature is aiming
for.

Cheers

nophead, 2023-08-01:
Get's my vote.

jordanbrown0, 2023-08-01:
See #265.

kwikius, 2023-08-01:
Implementation here
https://github.com/kwikius/openscad/tree/module_ops_v1/src/core/primitives