openscad/openscad #4713: Curly brace function call syntax

kwikius, 2023-08-01:
**Is your feature request related to a problem? Please describe.**
Curly brace function call syntax. Now have simple variable declarations in functions. 
Much more legible functions

**Describe the solution you'd like**
```
// existing
function getApparentWindAngle()
=
let( windVector = getApparentWindVector())
(norm(windVector) !=0)
? atan2(windVector.y, windVector.x)
: 0;

//proposed
function getApparentWindAngle()
{
  windVector = getApparentWindVector();
  if (norm(windVector) !=0){
    = atan2(windVector.y, windVector.x)
  }else{
    = 0;
  }
}

**Additional context**
Above syntax appears to be parsable by very trivial mods to existing grammar

t-paul, 2023-08-01:
The feature request itself gets my vote. The proposed syntax is unusual and surprising, I think that needs to change to something more "normal".

nophead, 2023-08-01:
I can't see the point as it doesn't add any new capailities, just a different syntax. If it is going to be supported I suggest a return statement instead of a strange use of =.

I would write the original function as:
```
function getApparentWindAngle() = 
       let( windVector = getApparentWindVector() )
            norm(windVector) ? atan2(windVector.y, windVector.x) : 0;
```

Seems more concise and neater to me.

kwikius, 2023-08-01:
Much of OpenSCAD syntax follows C, Java, C++ etc. This curly brace function definition syntax will be familiar to users of those languages.
The current function call syntax is unnecessarily cryptic. Allowing variables in the body of the function was obviously an afterthought.
```return``` would be nice but would presumably be rejected as being a new keyword.  The ```=``` expr syntax is available, works ok with no new keywords required and ```=``` is of course used as current OpenSCAD return syntax. The proposed syntax only brings the "return" operator right in front of the expression being returned instead of having the optional sequence of ``` let(x = expr) ``` expressions in the way , and that makes it easier to read (in my opinion). 

```
//proposed
function getApparentWindVector()
{
   windVector = -[sin(windDirection), cos(windDirection)] * windSpeed;
   yachtVector = -[sin(yachtDirection), cos(yachtDirection)] * yachtSpeed;
  // maybe many more vars ...
   
   = windVector + yachtVector;
}

function getApparentWindAngle() 
{
   windVector = getApparentWindVector();

   = ( norm(windVector)!= 0)  ? atan2(windVector.y, windVector.x) : 0;
}
```

You can use the ternary op, but ternary op is more cryptic, and the current openSCAD syntax doesnt allow to choose if _opt_else_, syntax whereas proposed syntax does:
```
function switch(v)
{
    if (v == 0)  = "red";
    if (v == 10) = "large";
    if (v == -5) = "odd";
    
    // warn and return undef if here
}
```

jordanbrown0, 2023-08-02:
I'd favor a new keyword over new syntax.  `return (foo);` is just another new keyword, versus a new syntax.  And `return(foo)` looks like other languages.

One issue is that this is temptingly close to module syntax, which would be a good thing except that module *semantics* are quite strange.  All assignments are processed first, and then all module invocations.
```
x=undef;
assert(x != undef);
y=x+1;
```
yields a complaint about `undefined + number` and *then* the assertion failure.

```
function x(a) {
    if (a == undef) return (0);   // or whatever return syntax
    b = a + 1;
    return (b);
}
```
would naturally, by module evaluation rules, yield an error on the `b=a+1` and *then* process the `if`.

Perhaps the answer there is to make module evaluation rules more ... ordinary ... but one of the advantages of the current scheme is that (in theory) all module invocations could be processed in parallel.

kwikius, 2023-08-02:
Good point! 

'if' as a module rather messes up the logic of child scopes.

The following might return either 0 or undef, dependent on how and when the return expression is evaluated
```
function x(a){
  if ( a == undef) return 0;
  return a;
}
```

I know what I would expect to happen, but a different semantic for functions compared to modules and global scope, would be confusing, which nullifies the whole point, which was to make functions easier to comprehend

nophead, 2023-08-02:
Not sure how having two completely different syntaxes to acheive exactly the same thing makes OpenSCAD code easier to comprehend. And the new one is more long winded so I prefer the old one as I hate typing.

```
function x(a) = is_undef(a) ? 0 : a;
```