openscad/openscad #4714: Global Constants Unavailable in Functions

njsokalski, 2023-08-02:
I have multiple functions that all use the same constant that is declared outside the function (for example, a variable such as padding I would want to always have the same value in all instances of the function). However, OpenSCAD does not let me do this and gives me a warning message. Is there any way to declare a value (constant) outside the functions so that I do not need to include it as a parameter? Including it as a parameter would be somewhat inefficient, because it would force me to have a lot of parameters whose values always came from the same place.

jordanbrown0, 2023-08-02:
Give an example please, including the warning message.

```
function bigger(a) = a + howMuchBigger;

howMuchBigger = 2;

echo(bigger(2));
echo(bigger(7));
```

works on both 2021.01 and 2023.06.06.

njsokalski, 2023-08-02:
I will note that the values are being used in a let() in the function, not directly in the expression being returned

njsokalski, 2023-08-02:
Here are the declarations of the constants & the function that uses them:
```
gamerackpadding=0.4;
gamerackpinholemargin=2;//Space between a GameRackPinhole & a slot or the edge of a GameRackCorner
gamerackpinmargin=2;
gamerackpinradius=4;
gamerackpinholeradius=gamerackpinradius+gamerackpadding/2;//Radius of hole into which the pins will be inserted
gamerackpinheadradius=gamerackpinradius+2;
function GetSuggestedGameRackCornerValues(combinedsize=[100,100],top=8)=
	let(toothoverlap=1.5*top,
		slotdepth=top/2,
		sidesize=[(combinedsize.x-toothoverlap)/2,(combinedsize.y-toothoverlap)/2],
		maxbottomvalue=sidesize.x-2*gamerackpinradius-2*gamerackpinmargin-2*slotdepth-gamerackpadding,
		maxleftvalue=sidesize.y-2*gamerackpinholeradius-2*gamerackpinholemargin-slotdepth,
		bottomcount=floor((maxbottomvalue-gamerackpinholemargin)/(2*gamerackpinheadradius+gamerackpadding+gamerackpinholemargin)),
		leftcount=floor((maxleftvalue-gamerackpinholemargin)/(2*gamerackpinheadradius+gamerackpadding+gamerackpinholemargin)),
		bottomspace=maxbottomvalue/bottomcount,
		leftspace=maxleftvalue/leftcount,
		bottomholes=[for(i=[0:bottomcount])i*bottomspace],
		leftholes=[for(i=[0:leftcount])i*leftspace]
		)[sidesize,[for(i=[1:2:len(bottomholes)])bottomholes[i]],[for(i=[0:2:len(bottomholes)])bottomholes[i]],[for(i=[1:2:len(leftholes)])leftholes[i]],[for(i=[0:2:len(leftholes)])leftholes[i]]];
```
The first line that gives a warning is the one where I assign a value to maxbottomvalue (because it is the first one to use one of the external constants).

nophead, 2023-08-02:
I don't get any warnings but I get undefined values in the results because you are indexing off the end of the arrays.

```
echo(GetSuggestedGameRackCornerValues());
ECHO: [[44, 44], [23.6], [[0, 0], [2, undef]], [27.6], [0, undef]]
```

A for loop includes the last value, so you are indexing arrays with length 2 with an index of 0 and then 2 when the start is 0.

jordanbrown0, 2023-08-02:
I get the same results as nophead - no warnings or errors, results include undef.

A simple example using a global variable in a "let" works:
```
function foo() =
    let(a=b+1)
    a;

b = 27;

echo(foo());
```

Exact error message, please.  And it would be good to try to trim down your example to the minimum required to demonstrate the problem.  That both makes it easier on us and, if it's really your problem, help you to find the problem.

njsokalski, 2023-08-02:
Sorry about the complexity, here is a shorter example (note that there is no for loop involved, so that is obviously unrelated). The code is as follows:
```
gamerackpadding=0.4;
gamerackpinmargin=2;
gamerackpinradius=4;
function GetSuggestedGameRackCornerValues(combinedsize=[100,100],top=8)=
	let(toothoverlap=1.5*top,
		slotdepth=top/2,
		sidesize=[(combinedsize.x-toothoverlap)/2,(combinedsize.y-toothoverlap)/2],
		maxbottomvalue=sidesize.x-2*gamerackpinradius-2*gamerackpinmargin-2*slotdepth-gamerackpadding)maxbottomvalue;
suggested=GetSuggestedGameRackCornerValues(combinedsize=[182,162],top=8);
echo(suggested);
```
This gives the following warning & echo in the OpenSCAD console:
```
WARNING: Ignoring unknown variable 'gamerackpadding' in file Untitled.scad, line 322
ECHO: undef
```

nophead, 2023-08-02:
It works for me. No warnings and it echos 64.6;

[image]

What verison of OpenSCAD are you running. I have never had a problem accessing global variables in functions.

njsokalski, 2023-08-02:
I am using version 2021.01

nophead, 2023-08-02:
It works for in the that version, which is the last release. I can't explain why it doesn't work for you as it has always worked for me. Here is an example from 12 years ago: https://github.com/nophead/Mendel90/blob/master/scad/conf/colors.scad#L26

jordanbrown0, 2023-08-02:
Concur.  Works for me in 2021.01 and 2023.06.06.

How did you edit this file?  Did you perhaps use Notepad or some other external editor?  I notice that the problematic variable is at the top of the file, which makes me wonder whether an invisible UTF-8 Byte Order Mark is changing the name of that first variable.  Try adding "xxx=0;" above it, at the very top.  If on Linux, try "od -c <filename>" to see if there are invisible weird characters at the top.

jordanbrown0, 2023-08-02:
Sigh, github ate some markup.  `od -c <filename>`.

njsokalski, 2023-08-03:
I use Visual Studio Code as my editor. I will l look into the UTF-8 byte order thing to see if it could be related, as well as see what happens if I edit it using OpenSCAD directly. I'll be sure to post any info I find here.

njsokalski, 2023-08-04:
How/where do I need to go to check/fix the UTF-8 data? Even though I know what it is, I don't normally mess around with encoding, and am not quite sure where to look at or modify it.

jordanbrown0, 2023-08-04:
First, let's establish that that's what it is.  Have you been able to look at the file with some tool that will let you see the details?

nophead, 2023-08-04:
You could try just cutting and pasting the text from this github page into the OpenSCAD text editor. That is what I did to test it. I doubt the web browser will insert UTF-8 escape chararcters.

Also you could test the theory it is just the first constant that doesn't work by adding a dummy constant before it.

njsokalski, 2023-08-04:
Can you recommend some tool or utility that will let me see this data?

nophead, 2023-08-04:
HxD will show you a hex dump of a file.

jordanbrown0, 2023-08-31:
Did this ever get resolved?

kintel, 2024-12-22:
Closing as no further reports we made, assuming the issue was related to invisible characters

closed 2024-12-22 by kintel