openscad/openscad #4715: OpenSCAD without Qt is unable to find resources folder

coderforlife, 2023-08-05:
Due to an issue on this line:

https://github.com/openscad/openscad/blob/87bf17e53aac352c5f45dbdb05a4e05376512acb/src/openscad.cc#L932C3-L932C16

When OpenSCAD finds the application directory when there is no Qt, the actual directory cannot be found UNLESS the full path is specified to the `openscad` binary or `openscad` is in the current directory. This is because `argv[0]` will just be `openscad` and thus `fs::absolute(...)` has nothing to go off. See https://stackoverflow.com/a/933996/582298 how to do this correctly.

kintel, 2023-08-16:
Are you sure? `boost::filesystem::absolute()` defaults to resolving the absolute path based on the current directory.

Can you reproduce this issue yourself?

coderforlife, 2023-08-16:
> The current directory is NOT the directory that has openscad in it (unless
> you run openscad as ./openscad). The current directory is the one that is
> browsed to in the terminal. Openscad itself will likely be somewhere on the
> PATH.
>
> Jeff

And yes, I discovered this issue because it couldn’t find the resources
directory unless I ran the openscad program with its full path like
/usr/local/bin/openscad

kintel, 2023-08-16:
duh, you're right - I just never put openscad in PATH, but that's of course the normal thing to do :)

coderforlife, 2023-08-18:
According to https://stackoverflow.com/questions/5694190/boost-filesystem-how-to-find-out-in-which-directory-your-executable-is the way to do this is with [`boost::dll::program_location().parent_path()`](https://www.boost.org/doc/libs/1_67_0/doc/html/boost/dll/program_location.html). On Windows, MacOS, Solaris, and FreeBSD this works perfectly however on Linux apparently this requires compiling with `-rdynamic` which makes the compiled program larger and may make it slightly slower. So maybe something along these lines (possibly incomplete code, mostly follows https://stackoverflow.com/a/933996/582298):

```c++
filesystem::path get_program_dir(const char* name /*pass in argv[0]*/) {
try {
     // covers Windows, MacOS, and a few others
     return boost::dll::program_location().parent_path();
} catch (boost::system::system_error) {
    const char *proc_tests[] = {
        "/proc/self/exe", // Linux
        "/proc/self/exefile", // QNX
        "/proc/curproc/file", // DragonFly BSD
        "/proc/curproc/exe", // NetBSD
    };
    system::error_code ec;
    for (int i = 0; i < sizeof(proc_tests)/sizeof(proc_tests[0]); i++) {
        filesystem::path path = filesystem::read_symlink(filesystem::path(proc_test[i]), ec);
        if (!path.empty()) { return path.parent_path(); }
    }
    // At this point we are hoping for anything...
    filesystem::path path(name)
    if (strchr(name, '/')) { return filesystem::canonical(path).parent_path(); }
    const char* path_env = getenv("PATH");
    if (path_env) {
        std::string input(path_env);
        std::vector<std::string> parts;
        boost::split(parts, input, boost::is_any_of(":"));
        for(std::string part : parts)  {
            if (!part.empty() && filesystem::exists(filesystem::path(part) / path)) {
                 filesystem::path full(filesystem::canonical(filesystem::path(part) / path));
                 if (filesystem::is_regular_file(full) && filesystem::is_executable(full)) { // is_executable doesn't seem to exist...
                     return full.parent_path();
                 }
            }
        }
    }
    // Current directory is better than nothing...?
    return filesystem::current_path()
}
```