openscad/openscad #4717: OpenCSG cannot be automatically generated when using standard `make`

coderforlife, 2023-08-05:
Because of the removal of examples at
https://github.com/openscad/openscad/blob/87bf17e53aac352c5f45dbdb05a4e05376512acb/scripts/uni-build-dependencies.sh#L555C15-L555C15
this causes OpenCSG to not be able to compile when using standard `make`. Just remove that `sed` for examples. The examples are really small and take a fraction of a second to compile. Just leave them be and it will compile better.

kintel, 2023-08-16:
I believe the examples need GLUT, which isn't a dependency of OpenSCAD, so including those will require us to manage GLUT as an extra dependency.

coderforlife, 2023-08-18:
Ah, didn't notice that. I guess maybe a more clever solution will be needed instead of simply removing all "example" from the make file.

t-paul, 2023-08-26:
What does "compile better" mean? Is the script failing in some environment? The script is not meant to work on each and every system, the recommended way is using development packages shipped via distributions as much as possible.

coderforlife, 2023-08-26:
The program cannot be compiled at all on any system when Qt is disabled in the build and using the automatically generated OpenCSG (instead of a system OpenCSG). This is because the `sed` command I highlighted blindly removes all references to `examples` which makes the make file try to make at `/` (instead of `examples/`)

thehans, 2023-08-27:
In case its of any use: there is a CMakeLists.txt for OpenCSG, which I had to make a while ago to allow for VCPKG compatbility:
https://github.com/microsoft/vcpkg/tree/master/ports/opencsg

Perhaps it could be added to upstream OpenCSG repo if they are open to the idea.  
I think it currently omits examples (and maybe other features, I don't remember), so it might be good to add actual options for those in the CMakeLists for completeness.
I don't have much time or energy lately to do the work, but if anyone else feels up to the task then feel free.

kintel, 2023-08-27:
We can build OpenCSG in-line now: https://github.com/openscad/openscad/pull/4596

coderforlife, 2024-05-08:
Just rebuilt all the binaries and the in-line OpenCSG builds wonderfully!

closed 2024-05-08 by coderforlife