openscad/openscad #4722: Improvement for subtracting and adding parts to an object

fah, 2023-08-15:
**Improvement for subtracting and adding parts to an object**

Currently the difference() function has the weakness that only the first argument is additive.
All other parts are subtractive.

Lets introduce a "differenceYaml()" function in addition to the new keywords "add" and "del":

This function would be translated by a precompiler.

Example:
```
// blockxPlus() represent a module call to create objects that should stay.
// drillxMinus() represent a module call to create objects that should be removed.

differenceYaml() {
	add( block1Plus() ); // Start part 1.0
	del( drill1Minus() ); // remove from part 1.0
		add( block2Plus() ); // Sub part 1.1 start. 
		del( drill2Minus() ); // Remove from Sub part 1.1 only, not from 1.0 . 
		// Now: Add part 1.1 to 1.0 (below this is the union 1)
	add( block3Plus() ); // Start part 2.0 . modify the the above created object part 1.0 + 1.1
	del( drill3Minus() ); // remove something from the above created object part 1.0 + 1.1
}
```

This would be translated internally as:
```
difference() {
	union()  { // union 2 start
		union() { // union 1 start
			difference() {
				block1Plus();
				drill1Minus();
			}	  
			difference() {
				block2Plus();
				drill2Minus(); 
			}
		} // union 1 end
		block3Plus();
	} // union 2 end
	drill3Minus(); 
}
```

This is a proposal and I just want to share my idea with you.

The add() and del() keywards have some advantages:
1. Easier translation(?)
2. Syntax and logic check is possible (Then easier translation)
3. Better readability
4. del() can be used on its own on the above created parts, if needed, without adding something.
...

Indentation has the advantage that parts can be easily (re-)combined. 

Readability is **much** better.

jordanbrown0, 2023-08-15:
Two notes:
* It appears that you're intending for indentation to have some meaning in how the operations relate to one another.  Although indentation is recommended in OpenSCAD for readability, like in most languages (Python excepted) changing indentation never changes the meaning of an OpenSCAD program.
* In the "translated internally as" example, union 1 is unnecessary.  Union 2 can be the two differences plus block3Plus():
```
difference() {
    union()  { // union start
        difference() {
            block1Plus();
            drill1Minus();
        }	  
        difference() {
            block2Plus();
            drill2Minus(); 
        }
        block3Plus();
    } // union end
    drill3Minus(); 
}
```

---
Indeed, complex add/subtract combinations can be hard to read.  One way to make them easier to follow is to define the subassemblies as modules, e.g.:
```
module block1() {
    difference() {
        block1Plus();
        drill1Minus();
    }	  
}

module block2() {
    difference() {
        block2Plus();
        drill2Minus(); 
    }
}

difference() {
    union()  {
        block1();
        block2();
        block3Plus();
    } 
    drill3Minus(); 
}
```
and perhaps even as:
```
module block1() {
    difference() {
        block1Plus();
        drill1Minus();
    }	  
}

module block2() {
    difference() {
        block2Plus();
        drill2Minus(); 
    }
}

module block123Plus() {
    block1();
    block2();
    block3Plus();
}

difference() {
    block123Plus();
    drill3Minus(); 
}
```

gsohler, 2023-08-16:
The Order of the additive and substractive Terms could very important and i
am Not sure If simple yaml add and del Keywords could do IT.

There is a PR for Python Support in the Queue. With IT you would write:

Obj += add1
Obj -= sub1
Obj += add2
Obj -= sub2

Yet another idea ...

pca006132, 2023-08-16:
what would you do in case the user only wrote del and no add?
and honestly I don't feel this is more readable than just union and difference. I would vote for python style operator overloading instead.

fah, 2023-08-16:
@pca006132 
> what would you do in case the user only wrote del and no add?
This could be handled in two ways:
1. forbid it: At least on add() has to be at the first line. -> Causes an error message (Prefered by me, since it makes no sense.)
2. ignore it: Issue a warning.
3. work with the existing object in the space.

@gsohler 
Hope to see this soon. :-)

@jordanbrown0 
My idea has many advantages that you do not see as advantages in your consideration.
1. it is compact (better than calling submodules) and easy to understand and maintain. (Like the Python PR. It seems to have some other advantages).
1.2. you need 23 lines, I need 8 for the same result. 
2. it can be implemented into openscad relatively quickly with a precompiler.
3. i don't quite agree that indentation is not helpful.
3.1. If I have parts that repeat in different places, copy and paste helps enormously with creation speed. 
3.2. > The need for indentation: 
Stacking objects can also be done with curly braces. Indentation is not the only way. But indentation is a more natural way to immediately see and understand the impact on the model.
3.3. > Object 1.1. does not need to be indented: 
Objects 1.0. and 1.1. are created separately, but perhaps in the same area. They do not influence each other. **They build on each other (purely additive).** 
Object 2.0. can then influence both with a hole or so.  Without the indentation of object 1.1. , object 1.0. would be changed by object 1.1. So the need for indentation (or similar) makes sense.

jordanbrown0, 2023-08-17:
Languages can be built that use indentation for structure.  Python is one.  But OpenSCAD is not... and changing fundamental aspects of the language design is exceedingly unlikely.

fah, 2023-08-19:
@jordanbrown 
OK, I agree. OpenSCAD will not change to a new paradigm like indentation.

Nevertheless, I think that new keywords like "add" and "del" or "remove" within the old difference() function would be very helpful to shorten the source code and make it more readable.

If you want to close this idea, no problem. I found a substitute with SolidPython to get the structuring the way I wanted it.

jordanbrown0, 2023-10-13:
I've left this on my to-be-thought-about list for a while, thinking about it occasionally.

One way to think about the `add()` and `del()` concept is to think of union as being sort of like addition, and difference as being sort of like subtraction, and allow for something like negative values.  In that model:

- `del() x();` would yield a node that is tagged as negative.
- `union() { a(); del() b(); c() }` would be treated like `a+(-b)+c`, which can be regrouped as `(a-b)+c` and then expressed in OpenSCAD language like so:
```
union() {
    difference() {
        a();
        b();
    }
    c();
}
```
Note that this arithmetic would not quite parallel conventional arithmetic:
- The result of a union could never be negative.  In conventional arithmetic, `a-b` can be negative if `b` is larger than `a`.  I'm pretty sure that undesirable results would occur if that behavior was carried through into `union()`, making it so that the "left over" negative parts of an expression were carried over into higher-level unions.
- union would probably not be considered commutative.  In arithmetic, `a + (-b) + c` is equivalent to `a + c + (-b)` and, for that matter, to `(-b) + a + c`.  Here, partially tied to the "no negative result" conclusion above, once you regroup to do one operation at a time the result would be different if you rearranged the components.  Also, executing strictly in order allows you to subtract `b` from `a` without also subtracting it from `c` - or to subtract it from both, if you like, by putting the subtraction last.

Note that an explicit `add()` operation isn't required, since that's what `union()` normally does.

Note that this "arithmetic" probably would apply to any implicit-union operation, not just `union()` itself.  This would be straightforward for the transform modules (`translate()`, `rotate()`, et cetera).  `intersection()`, since it does not implicitly union its children, should probably reject negative children.  The behavior of `difference()` would probably be particularly brain-damaging, because `difference() { a(); b(); del() c(); }` would probably be interpreted as "subtract `c` from `b`, then subtract that difference from `a`".  Behavior of `minkowski()` is left as an exercise for the reader, though allowing minkowski subtraction might not be unreasonable.

I'm not sure whether this is a good idea or not.  It would make some union-difference combinations less verbose and arguably easier to follow, at the expense of making `union()` and implicit-union semantics more complex.

---

Also, a technical note:  the original proposal uses a conventional-language-like syntax of `del(moduleInvocation());`.  That's not how geometry is passed around in OpenSCAD; an invocation that generates geometry cannot be in the middle of an expression.  It would need to be "child" syntax, `del() moduleInvocation();`.

jordanbrown0, 2023-10-13:
But:  the no-negative-result rule would yield a few weird results:
```
module mydel() {
    del() children();
}

module nop() {
    children();
}
```
You might think that `mydel() foo();` would be equivalent to `del() foo();`, but since the result of the implied union in the body of mydel() is not allowed to be negative, it wouldn't.  Similarly, you might think that `nop() del() foo();` would yield a negative object, but again the implied unions (both of the children of `nop()` and of the body of `nop()`) would not be allowed to be negative.

Perhaps those special cases could be allowed, or perhaps the rules would be:
- The union of a single positive object is positive.
- The union of a single negative object is negative.
- The union of two positive objects is positive.
- The union of a positive object and a negative object is positive (though perhaps empty).
- The union of two negative objects is negative.
- Unions of three or more objects are decomposed left-to-right into unions of two objects.

What about the union of a negative object and a positive object?  Perhaps it would be a positive object, as a slight nod to commutativity, or perhaps it would be negative, with the higher-level rule being that the type of the first object in the union controls the type of the  union's result.

pca006132, 2023-10-14:
Wow this sounds interesting, not sure if users will want this but sounds fun to discuss.

One way to make the union result make sense is to do symmetric difference for operands with opposite polarity. To understand this you can think of each operand as a pair of geometries $(a, b)$ where $a \cap b = \emptyset$, and union is defined as $(a, b) \cup (a', b') = (a \cup a', b \cup b')$ with an additional normalization defined as $(a, b) \to (a \setminus a \cap b, b \setminus a \cap b)$. Normal geometries are $(a, \emptyset)$, negation is defined as $-(a, b) = (b, a)$. To extract positive/negative geometries, you just do projection.

This preserves things like commutativity and associativity. The main problem is that the semantic may be surprising to users.

JamesC1, 2024-04-14:
I approve of del() as an operator.  I vaguely think that there should be neg() and undel() to provide the full set of logical options for a binary property.

I'm pre-contaminated with tinkercad.com, where being a negative shape is a first-class property of shapes, like colour, and negative shapes turn into negative space when they are grouped with positive shapes.

If it went this way, then difference() would become a synonym for union() or group().

There would be two plausible interpretations of union(), which might end up being separate operators.  One is commutative, and applies all of the positive shapes first.  The other is more powerful, and applies shapes in the order in which they are presented.  The difference shows if there is a large shape, with a medium sized hole, with a small shape extending across the hole.  In the first interpretation, the hole dominates; in the second, one act of grouping can put the small object into the empty space.  If openSCAD went with the second interpretation, then control over grouping would become important.

Whether shapes have been grouped is visible in the minkowski() operator.  if there is code of this general form:
```openscad
size=10;

minkowski(){
     cylinder(size, size, size);
     
     OPERATOR()   // for any of several different operators
     {
         translate([0,0,size])
         sphere(size);
         
         cylinder(size, size, size);
     }
 }
``` 
then the result will be flat on one end (because the bottom of the sphere is hidden) if the operator groups its children.  If not, there will (mostly) be a minkowski sum of three objects, because braces by themselves deliver ungrouped shapes to their surrounding context.

Please find attached a test of which operators group their children.  The answer is approximately all of them, including for(), which is documented, and color(), which seems completely unnecessary.

So there is a version of this enhancement that I like, but getting there would take a lot of work, and there might be reasons for grouping early that I don't know about, in which case it might not be able to be reliable at all.

JamesC1, 2024-04-14:
Test case for various grouping.
[gh-4722-grouping-test-01.txt](https://github.com/openscad/openscad/files/14970074/gh-4722-grouping-test-01.txt)

gsohler, 2024-04-14:
In pythonscad  you can use any number of positive terms you like:

union([pos1, pos2, pos3], neg1, neg2)

or
union(pos1, neg1, neg2, neg3)

or

union([pos1, pos2], [neg1, neg2],[neg3, neg4])

and it should be rather easy to also add negative space features ....

jordanbrown0, 2024-04-14:
All module invocations yield exactly one child.  There are no module invocations that yield multiple children, or zero children.  Though there is some, misguided in my opinion, special handling of empty geometries in difference() and, I believe, intersection(), so that echo() and failing if() are effectively dropped.

jordanbrown0, 2024-04-14:
Negative space is an interesting concept, but it is very unclear what its semantics would be.

If you union multiple negative objects, is the result a negative object?

If you union a negative object with a positive object that is a subset of it, is the result a negative object?

These boolean operations are not conventional arithmetic.  They are much closer to set arithmetic, and set arithmetic does not have negative values.

Set arithmetic *does* have the complement operator, and that might be interesting, but it would generally result in infinite-size objects.  There are a lot of cases where those can be useful.

JamesC1, 2024-04-15:
I am presuming three-valued arithmetic, so a volume in the space could be:
- irrelevant: negates to irrelevant
- negative: negates to positive
- positive: negates to negative

That enables us to negate a finite-volumed collection of positive and negative space, and get a finite-volumed collection of positive and negative space, treating the irrelevant space as irrelevant, and not needing to be represented explicitly.

In a good design, there must be a thing that I can do, so that all negative space is either applied (the overlapping positive space disappears, and the subset becomes irrelevant) or wasted (which depends on fine semantic detail later in this comment).  I want this primitive so that I can take several module results and combine them without any of them making unexpected, otherwise invisible holes in another.

I don't know whether that operator should be called union(), group(), or something else.  I'm going to call it flatten() for now, for no very good reason.  I'm assuming that being output from a module should not imply being flattened.  I'm assuming, inconsistent with the current design of openSCAD, that each child in a geometry contains space with a single value, plus the unmarked irrelevant space around it, so reducing to one child implies flattening.  If the eventual design is inconsistent with my mental model, should future modules be able to walk the set of positive and negative spaces?  eg how does one child with a mixture of volumes of different value interact with mechanisms analogous to $children / children(n)?

I've already mentioned that there are at least two semantics for flattening: a subtle one which applies each volume in order, and a commutative one which applies all positive volumes first, so any negative volume trumps all positive volume in the same operation.  The semantics of whichever primitive are the semantics of negative space, with a proviso that we could offer both operators.

If we offer both operators, and as we choose between them, minkowski is one situations where choices among them matter a lot.  I do not have a story about how the arithmetic should multiply, which is effectively what minkowski does.

Thinking about what minkowski plus flattening might do, if we pick the commutative version and there are filters to take only the negative volumes, or only the positive volumes out of a module result, then it should be possible to define positive/negative minkowski in terms of filtering, negating, and multiple positive/positive minkowskis.

For the avoidance of doubt, I would not vote for -ve*-ve to create positive volumes of space; I think that this would confuse normally socialised people unacceptably.  I am open to the view that -ve*-ve should become irrelevant.

If we pick only the subtle primitive, then defining +/- minkowski in terms of +/+ minkowski would probably require:
- enumerating in rising priority order the +/- spaces in each input geometry
- deciding a combined priority order between all of those spaces; probably a concatenation of the existing lists, which I expect would be the normal behaviour of grouping with braces
- probably executing (or the moral equivalent) +/+ minkowski on every pair of spaces

I note that the current (+/+) minkowski operator produces uncoloured output.  If all of the input geometries are convex, include the origin, and each input geometry has higher priority than the one before it, then I intuit that the following claim is true:
- every point on the surface of the output will correspond to exactly one point on the last input

When the inputs are not convex, I am fairly sure that it will be possible to arrange ambiguity, where two points far apart on the last object both account for a point on the result.

All of this will be worse when the whole volume of the object must be accounted for.

What other operators are going to generate similar complexity?  Hull?

JamesC1, 2024-04-16:
In formulating my position here, I have been proposing OEP2 over OEP4, from this list:
https://github.com/openscad/openscad/wiki/OpenSCAD-Enhancement-Proposals

For the avoidance of doubt, I now think that either semantics could be compatible with either proposal.

JamesC1, 2024-04-19:
Here's a set of success criteria for new features of openSCAD, which I think make sense to me:
https://github.com/openscad/openscad/issues/114#issuecomment-38931142

jordanbrown0, 2025-03-08:
BOSL2's `diff()` mechanism is much like this.

See: https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-diff