openscad/openscad #4723: Import of freecad 3mf fails silently (stl works)

fallingcats, 2023-08-19:
**Describe the bug**
When importing a 3mf file exported from a freecad 0.21 project the import does not throw errors but does not get rendered in any way. Importing a freecad exported stl file of the same object works though.  

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
```
// works
import("exports/activ energy XYZ562a socket v08.stl");

// does not work
import("exports/activ energy XYZ562a socket v08.3mf");
```

**Environment and Version info (please complete the following information):**
    Operating System: Arch Linux 
    KDE Plasma Version: 5.27.7
    KDE Frameworks Version: 5.108.0
    Qt Version: 5.15.10
    Kernel Version: 6.1.44-1-lts (64-bit)
    Graphics Platform: X11
    Processors: 8 × AMD Ryzen 7 4700U with Radeon Graphics
    Memory: 15.0 GiB of RAM
    Graphics Processor: AMD Radeon Graphics
    Manufacturer: HP
    Product Name: HP ENVY x360 Convertible 15-ee0xxx
 - OpenSCAD Version 2021.01

**Additional context**
Zip containing both files: [activ energy XYZ562a socket.zip](https://github.com/openscad/openscad/files/12384634/activ.energy.XYZ562a.socket.zip)

UBaer21, 2023-08-19:
Just tested with 2023.08.08 where it seems to work fine.

fallingcats, 2023-08-21:
> Just tested with 2023.08.08 where it seems to work fine.

Sorry, I can't confirm that, building something newer locally fails for me

t-paul, 2023-08-21:
I can't reproduce on Debian/testing with both 2021.01 release and latest snapshot version. But essentially all builds (except Flatpak I believe) still use lib3mf 1.8.x. As far as I know Arch already uses lib3mf 2.x.

In Flatpak, I'm seeing:
WARNING: Could not read file '/tmp/o/activ energy XYZ562a socket-Cut v08.3mf', import() at line 1

So it seems the file is not compatible with the official 3mf library. This probably means the issue can't be soved on OpenSCAD side.

UBaer21, 2023-08-21:
[image]

works fine on  win with 2023.08.13  maybe try a full path

fallingcats, 2023-08-21:
> @fallingcats can you provide the error msg from the console?

As I tried to say with "fails silently", there is no error msg in the integrated console

t-paul, 2023-08-21:
See my comment above, dev version shows a warning. It's possible that is missing in the release.

fallingcats, 2023-08-21:
Oh, it seems openscad only shows an error once, not repeatedly when pressing Preview/F5. Here is the complete (but probably useless log):

```
Parsing design (AST generation)...
Saved backup file: /home/bernhard/.local/share/OpenSCAD/backups/unsaved-backup-LwTJCtkv.scad
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
WARNING: Could not read file '/home/bernhard/Projects/3D Printer/designs/scooter battery mount/exports/activ energy XYZ562a socket-Cut v08.3mf', import() at line 1 
Geometries in cache: 2
Geometry cache size in bytes: 304
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.023
```

If you could provide a static binary, I could try that.

t-paul, 2023-08-21:
You could try with the AppImage, either release or dev snapshot from the [download page](https://openscad.org/downloads.html). Those still use the old lib3mf for reading files.

You can ignore the warning banner there, I'm about to remove that, downloads should be working fine again.

UBaer21, 2023-08-21:
Ok i now had i very weird experience .. Autocad (meshmixer) couldn't open it. Microsoft3D builder could - after i saved it there Scad couldn't open it anymore (nor Meshmixer) .. 
Then i opened the original again in SCAD rendered (in union with cube) and exported it - this could be opened with meshmixer ..  @fallingcats  can you give it a try (rename the zip to 3mf as github doesn't support 3mf)
[activeEnergySCAD.zip](https://github.com/openscad/openscad/files/12400456/activeEnergySCAD.zip)

fallingcats, 2023-08-21:
> You could try with the AppImage, either release or dev snapshot from the [download page](https://openscad.org/downloads.html). Those still use the old lib3mf for reading files.

Both dev and regular AppImage versions work for me. It seems the error may come from the way it's packaged in Arch. I will investigate further.

fallingcats, 2023-08-21:
> @fallingcats can you give it a try (rename the zip to 3mf as github doesn't support 3mf)
> [activeEnergySCAD.zip](https://github.com/openscad/openscad/files/12400456/activeEnergySCAD.zip)

Also does not work in my distribution provided version

fallingcats, 2023-08-22:
> But essentially all builds (except Flatpak I believe) still use lib3mf 1.8.x. As far as I know Arch already uses lib3mf 2.x.
The version of lib3mf-1 the openscad package of arch depends on is 1.8.1-6, see [archive](https://archive.archlinux.org/packages/l/lib3mf-1/)/[package](https://archlinux.org/packages/extra/x86_64/lib3mf-1/), [openscad package](https://archlinux.org/packages/extra/x86_64/openscad/).
I've tried 1.8.1-5 which was the current version during release of openscad 2021.01, but that didn't change anything.

jordanbrown0, 2023-08-22:
> Oh, it seems openscad only shows an error once, not repeatedly when pressing Preview/F5

We really need to fix it so that the result from an error isn't cached like that.  This is a common confusion.  (But unrelated to the question of why it can't read this file.)

t-paul, 2023-08-25:
Root cause is the `item` line 
```
 <build>
  <item objectid="1" transform="1 0 0 0 1 0 0 0 1 166.887435 102.433224 0" printable="1"/>
 </build>
```
raising the error "Lib3MF Error 5 (Wanted to convert string with non-numeric characters to double)".

This is essentially the same issue as #4204 which is an upstream bug https://github.com/3MFConsortium/lib3mf/issues/282 - I have `LC_NUMERIC=de_DE.UTF-8` set so the notation with '.' is invalid.

t-paul, 2023-08-25:
Upstream PR submitted https://github.com/3MFConsortium/lib3mf/pull/322 - but closing as duplicate.

You can register for notifications in #4204 if you want to follow updates.

closed 2023-08-25 by t-paul

fallingcats, 2023-08-26:
> This issue starts to break OpenSCAD installations as some Linux distributions have moved to lib3mf v2.2.0. At this point, I'm aware of **Arch** and Fedora **shipping this version**.

@t-paul Sorry but I've been trying to tell you that this is [wrong](https://github.com/openscad/openscad/issues/4723#issuecomment-1687242834). If you don't believe me look here:
- https://archlinux.org/packages/?sort=&q=lib3mf
- https://gitlab.archlinux.org/archlinux/packaging/packages/openscad/-/blob/main/PKGBUILD#L36

t-paul, 2023-08-26:
AUR ships v2.2.0, so people might use it.

fallingcats, 2023-08-26:
> AUR ships v2.2.0, so people might use it.

Please remember that the AUR is "completely unofficial" and not part of the distribution. It is not "shipped by arch" but random users.
I've tried that package, it does not even build (for me). In any case a 2.0 -> 2.2 regression shouldn't be the cause of my problem?