openscad/openscad #4724: Indirect module calls by name

PinkPandaKatie, 2023-08-20:
**Is your feature request related to a problem? Please describe.**
I have been wanting a way to call a specific module whose name is contained in a variable or expression. Almost all my projects have multiple parts to them, and I declare variable called "part" which specifies which part to render. For example:

```
part = "mainbox_lower"; // [mainbox_lower, mainbox_middle, mainbox_upper, pcb_model]
```

At the bottom, I have to have a block that looks like this:

```
if (part == "mainbox_lower") {
    mainbox_lower();
} else if (part == "mainbox_middle") {
    mainbox_middle();
} else if (part == "mainbox_upper") {
    mainbox_upper();
} else if (part == "pcb_model") {
    pcb_model();
}
```

**Describe the solution you'd like**
I would like to be able to do something like

```
indirect(part);
```

instead of the giant if/else block.

In particular,

```
mod = "abc";
indirect(mod, 5, 6, x=10, $fn=20) {
    sphere(d=1);
}
```

should be equivalent to

```
abc(5, 6, x=10, $fn=20) {
    sphere(d=1);
}
```

**Describe alternatives you've considered**
I have looked at the module literals WIP branch but that still doesn't do quite what I want. This enhancement could do most of what that branch could do in a much simpler way, however.

With that branch, this could be implemented as a function that returns a module object from the global table (e.g. `get_module`) so you could do something like

```
(get_module(part))();

mod = "abc";
(get_module(mod))(5, 6, x=5) { sphere(d=1); }
```

I'm going to fork and see if I can implement this. Any suggestions would be welcome.

t-paul, 2023-08-21:
This should be covered by the proposal in #4478. Please have a look and join the discussion there. It would be great to get this moving again.

jordanbrown0, 2023-08-21:
Yes, #4478 is the most likely avenue for this.

It does not include a way to take a string and look up a module (though, hmm, that does seem possible), but it does allow passing around module references (what C would call pointers to functions), having name-value lists where the value is a module reference, and so on.