openscad/openscad #4729: directly access points and faces from modules

jmerc77, 2023-08-29:
**Is your feature request related to a problem? Please describe.**
there's niche problem where i have a custom scad lib that requires a list of points in 2d but i want to draw it using the shapes and boolean ops in a module.

**Describe the solution you'd like**
directly accessing points and faces from a module instead of needing to export then import. not sure if anyone knows they want this, though. Also thought it would be cool if it worked for 2d modules and 3d modules.

**Describe alternatives you've considered**
i could export a model file, convert it to points and faces, reimport, then apply the lib. but its very messy to do that with lots of steps.

**Additional context**
[openscad sweep lib](https://www.thingiverse.com/thing:5354505) the lib mentioned. test with it maybe?

jmerc77, 2023-08-29:
ok. btw, there is one feature that i thought would be handy not mentioned in that. i would call it "test".

used like:
```
res = test(){
child_a
child_b
}
```
you could then use res in an if().

it would test if the two children intersect.

1. it first applies intersection to them
2. then counts the closed bodies and returns that count
3. it does not output any geometry. just the count.
4. count can be any number 0 or greater.

this could be used when it is unknown if a part of geometry is there or not and take action on it or other geometry.
it could also be used to verify that the model is in specification. for example, testing that the parts fit loosely enough.

i am curious if recursive geometry is possible, too (though may exist, i never tried).

t-paul, 2023-08-29:
I think your test example could be done in #4478 by inspecting a regular intersection. Closing as duplicate.

You can generate geometry recursively. Example: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Tips_and_Tricks#Stack_cylinders_on_top_of_each_other

jordanbrown0, 2023-08-29:
Yes, #4478 would let you test for whether the resulting geometry is empty.  It will also let you know the number of distinct objects, but that seems less useful.  (And yes, you need to generate the full mesh to do this test, but... you need to generate the full mesh to get the answer.)