openscad/openscad #4730: qt5-devel obsolete?

FatherGeoffHorton, 2023-08-30:
When building on WSL2 with Ubuntu (5.15.90.1-microsoft-standard-WSL2), scripts/uni-get-dependencies.sh tries to install qt5-devel, which doesn't exist. Installing qt5-assistant works.

thehans, 2023-08-30:
Which release of Ubuntu is it based on?  Can you run `lsb_release -a` and paste the results here?

FatherGeoffHorton, 2023-08-30:
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy

FatherGeoffHorton, 2023-08-30:
It happens on a native Linux system as well.

No LSB modules are available.
Distributor ID: Linuxmint
Description: Linux Mint 21.1
Release: 21.1
Codename: vera

thehans, 2023-08-30:
I think the main package it should grab is named `qtbase5-dev`.

If the script is trying to install `qt5-devel` that appears to be the name under RPM-based distros, which indicates that it is not correctly detecting what distro you are on.  

Looking at the script, it actually greps the file `/etc/issue` to help determine the distro
Can you paste the contents of that?

gsohler, 2023-08-30:
Especially Fedora and CentOS have /etc/redhat-release(99% sure)

FatherGeoffHorton, 2023-09-01:
On the true linux box:
Linux Mint 21.1 Vera

On WSL2:
Ubuntu 22.04.3 LTS