openscad/openscad #4732: Manifold - Minkowski fails on concave polyhedron with render mode (but works if you preview first)

inks007, 2023-09-03:
Minkowski fails when given certain shapes (e.g. a concave polyhedron + tall convex object) using manifold.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable manifold
2. Use the code below
3. Use a fresh instance of OpenSCAD, or 'Flush Caches'
4. Render (F6). The following error appears:
   > Rendering Polygon Mesh using Manifold...
   > WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: precondition violation!
   > Expr: ! running
   > File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Timer.h
   > Line: 83
5. Use a fresh instance of OpenSCAD, or 'Flush Caches'
6. Preview (F5)
7. Render (F6). No error this time.

No other experimental features were enabled apart from manifold.

**Expected behavior**
Render (F6) to pass at step 3.

**Code reproducing the issue**
```
$fn = 40;
cracked_square = [[-10,-10],[-10,10],[10,10],[10,-10],[-8,-10],[7,7]];
silo_shape = [for(a=[0:360/$fn:89.99]) 5*[cos(a),sin(a)],[0,5],[0,-10],[5,-10]];

minkowski() {
  rotate([10,30,0]) linear_extrude(20) polygon(cracked_square);
  rotate_extrude() polygon(silo_shape);
}
```

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 22H2 (19045.3324)
 - System: AMD Ryzen 7 7700X
 - OpenSCAD Version: 2023.08.25 (git 05b962168) [latest Development Snapshot]

P/S. Manifold is awesome, thank you all for integrating this library.

pca006132, 2023-09-03:
I think this is a problem in the minkowski code. Remove https://github.com/openscad/openscad/blob/master/src/geometry/cgal/cgalutils-applyops.cc#L239-L243 and it should be good.

Minkowski does not benefit much from parallel optimization anyway, as the majority of the time is spent on computing the convex decomposition. I am thinking about how to speed up 3D offsetting or even minkowski, but the progress is not good as I am working on something else right now.

pca006132, 2024-02-15:
I think it will only print warning now and will not fail.

pca006132, 2024-02-15:
OK found the root cause, will fix.

PR #4993 (2024-02-17): fix manifold stuff (without geometry cache changes)
Closes #4732. Basically #4991 but remove the geometry cache changes.
closed 2024-02-17 by kintel