openscad/openscad #4733: Fail to compute the difference of two simple polygons

54610a, 2023-09-04:
**Describe the bug**
Difference function from _CGAL 5.6 - 2D Regularized Boolean Set-Operations_ fails when dealing with some simple polygons.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the code below
2. An error occurs

>CGAL error: precondition violation!
>Expression : (m_traits.compare_y_at_x_2_object()(p, cv) == EQUAL) && compare_xy(cv.left(), p) == SMALLER && compare_xy(cv.right(), p) == LARGER
>File       : C:\dev\vcpkg\installed\x64-windows\include\CGAL/Arr_segment_traits_2.h
>Line       : 611

**Expected behavior**
It should be easy to produce the result.
Also, it's strange that if I change the pgn_1 to { Point_2(0,0), Point_2(1,0), Point_2(0,3) }, it works. It works in many other situations, but I can't find the problem. 

**Code reproducing the issue**
```
#include <CGAL/Simple_cartesian.h>
#include <list>
#include <iostream>
#include <CGAL/Exact_predicates_inexact_constructions_kernel.h>
#include <CGAL/Polygon_2.h>
#include <CGAL/Polygon_with_holes_2.h>
#include <CGAL/Polygon_set_2.h>
#include <CGAL/Boolean_set_operations_2.h>
#include <CGAL/draw_polygon_set_2.h>
#include <CGAL/draw_polygon_2.h>
#include <CGAL/draw_polygon_with_holes_2.h>
#define PI 3.1415926535
typedef CGAL::Simple_cartesian<double>                    K;
typedef CGAL::Polygon_2<K>                                Polygon_2;
typedef CGAL::Polygon_with_holes_2<K>                     Polygon_with_holes_2;
typedef CGAL::Polygon_set_2<K>                            Polygon_set_2;
typedef CGAL::Point_2<K>                                  Point_2;
typedef CGAL::Vector_2<K>                                 Vector_2;
using std::cout; 
using std::endl;
typedef std::list<Polygon_with_holes_2>                   Pwh_list_2;
int main()
{
	Pwh_list_2 symmR;
	Point_2 points_1[] = { Point_2(0,0), Point_2(1,0), Point_2(0,6) };
	Point_2 points_2[] = { Point_2(0.5,1), Point_2(1,1), Point_2(1,2), Point_2(0.5,2) };
	Polygon_2 pgn_1(points_1, points_1 + 3);
	Polygon_2 pgn_2(points_2, points_2 + 4);
	difference(pgn_1, pgn_2, std::back_inserter(symmR));
	return 0;
}
```

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit

**Additional context**
I'm a beginner in c++, there might be silly mistakes, but I am learning and I am willing to follow instructions.

t-paul, 2023-09-04:
Did you mean to post that there: https://github.com/CGAL/cgal/ ?

I don't see any OpenSCAD reference here and OpenSCAD uses Clipper for 2D operations.

54610a, 2023-09-04:
Oh yes, I'm really sorry. I am going to seek help there.

closed 2023-09-04 by 54610a