openscad/openscad #4739: minkowski render wrong surfaces in certain shapes

Dachtire, 2023-09-12:
**Describe the bug**
minkowski() render wrong surfaces when using difference() and translate() with particular number on z axis. The number would cause error like "1.6", "1.4" etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the code below
2. Render

**Expected behavior**
The left one is wrong, the right one is correct. Left and right cube should have similar shape, only the height differs.

**Code reproducing the issue**
```
module test(test_z) {
    difference() {
        translate([0, 0, test_z]) cube([20, 20, test_z], true);

        translate([0, 0, test_z]) minkowski() {
            cube([10, 10, test_z / 2], center = true);
            cylinder(h = test_z / 2, center = true);
        }
    }
}

translate([-15, 0, 0]) test(1.6);  // left side
translate([15, 0, 0]) test(1.5);  // right side
```

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 22.04
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2023.09.09.nightly (git dcaf25f)

t-paul, 2023-09-12:
Do you have the experimental feature "Manifold" enabled? If so, please check if any out of [this list of issues](https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+minkowski+label%3A%22Tag%3A+Manifold%22) matches your problem.

jordanbrown0, 2023-09-12:
I duplicate this on 2021.01.

Preview looks like classic Z-fighting.  Strangely, render looks like Z-fighting on the left side; I've never seen Z-fighting with render before.

Interestingly, removing the `translate([0,0,test_z])` or even factoring it outside the difference makes the problem go away.

General advice - though usually it only matters for preview - is not to try to difference with surfaces that are exactly aligned, as they are here.  Preview will try to both draw and not-draw the surface, yielding Z-fighting; it seems like render has a similar issue.  Making either the cylinder or the second cube a little taller makes the problem go away.  Adding 0.001 was enough for render; adding  0.01 was enough for preview.

Dachtire, 2023-09-13:
> Do you have the experimental feature "Manifold" enabled? If so, please check if any out of [this list of issues](https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+minkowski+label%3A%22Tag%3A+Manifold%22) matches your problem.

this issue occurs when "Manifold" is enabled or disabled

pca006132, 2023-09-13:
> I duplicate this on 2021.01.
>
> Preview looks like classic Z-fighting. Strangely, render looks like Z-fighting on the left side; I've never seen Z-fighting with render before.
>
> Interestingly, removing [...]

Is there self-intersection in the rendered output? That can cause Z-fighting.

kintel, 2024-01-14:
You're simply trying to subtract an object with the same thickness, yielding an undefined result. The problem is made worse by using a translation inside the difference, which is encoded using floating point coordinates, making the results depend on position in space.
As always, the solution is to make the negative object slightly thicker than the negative one.

closed 2024-01-14 by kintel