openscad/openscad #4741: libhogweed.so.6 failed to reference __gmpn_cnd_swap

ElektroBOXua, 2023-09-12:
During compilation libhogweed.so.6 failed to reference  «__gmpn_cnd_swap»
You can reproduce this bug by using version of libhogweed 3.9.1-2.2

I am running OpenSUSE and this is my current libhogweed version.
I tried to install version libhogweed6 3.6-1 and that fixed it for me.

t-paul, 2023-09-12:
What is that, and how is it related to OpenSCAD?

t-paul, 2023-09-12:
It's not a direct dependency as far as I can tell and it's not linked on Debian.

That error looks like a runtime error, not compilation. So that might be some local setup problem regarding library path or mismatch of libraries. Both would more point to an OpenSUSE question.

ElektroBOXua, 2023-09-12:
That is runtime error tho, but when i compile, it also requires this library for linkage:

```
furdog@localhost:~/Documents/openscad/build> make
[  1%] Automatic MOC and UIC for target OpenSCAD
[  2%] Built target OpenSCAD_autogen
[  7%] Built target mimalloc-static
[  7%] Built target graphlite
[  8%] Built target Clipper2
[  9%] Built target utilities
[  9%] Built target collider
[ 10%] Built target cross_section
[ 10%] Built target polygon
[ 15%] Built target manifold
[ 15%] Linking CXX executable openscad
Використання вбудованих специфікацій.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib64/gcc/x86_64-suse-linux/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Ціль: x86_64-suse-linux
Налаштування: ../configure CFLAGS=' -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -g' CXXFLAGS=' -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -g' XCFLAGS=' -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -g' TCFLAGS=' -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -g' GDCFLAGS=' -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,ada,go,d,jit,m2 --enable-offload-targets=nvptx-none,amdgcn-amdhsa, --enable-offload-defaulted --without-cuda-driver --enable-host-shared --enable-checking=release --disable-werror --with-gxx-include-dir=/usr/include/c++/13 --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --enable-ssp --disable-libssp --disable-libvtv --enable-cet=auto --disable-libcc1 --enable-plugin --with-bugurl=https://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --with-slibdir=/lib64 --with-system-zlib --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-libphobos --enable-version-specific-runtime-libs --with-gcc-major-version-only --enable-linker-build-id --enable-linux-futex --enable-gnu-indirect-function --program-suffix=-13 --without-system-libunwind --enable-multilib --with-arch-32=x86-64 --with-tune=generic --with-build-config=bootstrap-lto-lean --enable-link-serialization --build=x86_64-suse-linux --host=x86_64-suse-linux
Модель потоків: posix
Підтримувані алгоритми стискання LTO: zlib zstd
Версія gcc 13.2.1 20230803 [revision cc279d6c64562f05019e1d12d0d825f9391b5553] (SUSE Linux) 
COMPILER_PATH=/usr/lib64/gcc/x86_64-suse-linux/13/:/usr/lib64/gcc/x86_64-suse-linux/13/:/usr/lib64/gcc/x86_64-suse-linux/:/usr/lib64/gcc/x86_64-suse-linux/13/:/usr/lib64/gcc/x86_64-suse-linux/:/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/
LIBRARY_PATH=/usr/lib64/gcc/x86_64-suse-linux/13/:/usr/lib64/gcc/x86_64-suse-linux/13/../../../../lib64/:/lib/../lib64/:/usr/lib/../lib64/:/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/lib/:/usr/lib64/gcc/x86_64-suse-linux/13/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-frounding-math' '-v' '-o' 'openscad' '-shared-libgcc' '-mtune=generic' '-march=x86-64' '-dumpdir' 'openscad.'
 /usr/lib64/gcc/x86_64-suse-linux/13/collect2 -plugin /usr/lib64/gcc/x86_64-suse-linux/13/liblto_plugin.so -plugin-opt=/usr/lib64/gcc/x86_64-suse-linux/13/lto-wrapper -plugin-opt=-fresolution=/tmp/ccEzHg1Z.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --build-id --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 -o openscad /usr/lib64/gcc/x86_64-suse-linux/13/../../../../lib64/crt1.o /usr/lib64/gcc/x86_64-suse-linux/13/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/13/crtbegin.o -L/usr/lib64/gcc/x86_64-suse-linux/13 -L/usr/lib64/gcc/x86_64-suse-linux/13/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/13/../../.. CMakeFiles/OpenSCAD.dir/OpenSCAD_autogen/mocs_compilation.cpp.o CMakeFiles/OpenSCAD.dir/src/geometry/manifold/ManifoldGeometry.cc.o CMakeFiles/OpenSCAD.dir/src/geometry/manifold/manifoldutils.cc.o CMakeFiles/OpenSCAD.dir/src/geometry/manifold/manifold-applyops.cc.o CMakeFiles/OpenSCAD.dir/src/geometry/manifold/manifold-applyops-minkowski.cc.o CMakeFiles/OpenSCAD.dir/src/openscad.cc.o CMakeFiles/OpenSCAD.dir/src/Feature.cc.o CMakeFiles/OpenSCAD.dir/src/FontCache.cc.o CMakeFiles/OpenSCAD.dir/src/handle_dep.cc.o CMakeFiles/OpenSCAD.dir/src/LibraryInfo.cc.o CMakeFiles/OpenSCAD.dir/src/RenderStatistic.cc.o CMakeFiles/OpenSCAD.dir/src/version.cc.o CMakeFiles/OpenSCAD.dir/src/core/Arguments.cc.o CMakeFiles/OpenSCAD.dir/src/core/Builtins.cc.o CMakeFiles/OpenSCAD.dir/src/core/BuiltinContext.cc.o CMakeFiles/OpenSCAD.dir/src/core/CgalAdvNode.cc.o CMakeFiles/OpenSCAD.dir/src/core/Children.cc.o CMakeFiles/OpenSCAD.dir/src/core/ColorNode.cc.o CMakeFiles/OpenSCAD.dir/src/core/Context.cc.o CMakeFiles/OpenSCAD.dir/src/core/ContextFrame.cc.o CMakeFiles/OpenSCAD.dir/src/core/ContextMemoryManager.cc.o CMakeFiles/OpenSCAD.dir/src/core/control.cc.o CMakeFiles/OpenSCAD.dir/src/core/CSGNode.cc.o CMakeFiles/OpenSCAD.dir/src/core/CsgOpNode.cc.o CMakeFiles/OpenSCAD.dir/src/core/CSGTreeEvaluator.cc.o CMakeFiles/OpenSCAD.dir/src/core/customizer/Annotation.cc.o CMakeFiles/OpenSCAD.dir/src/core/customizer/CommentParser. [... cut ...]
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: попередження: libGLEW.so.2.2, потрібна для /home/furdog/openscad_deps/lib/libopencsg.so, може конфліктувати з libGLEW.so.1.9
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /usr/lib64/libhogweed.so.6: невизначене посилання на «__gmpn_cnd_swap»
collect2: помилка: ld повернув статус виходу 1
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:4143: openscad] Помилка 1
make[1]: *** [CMakeFiles/Makefile2:508: CMakeFiles/OpenSCAD.dir/all] Помилка 2
make: *** [Makefile:166: all] Помилка 2
```

I have explicitly copied this library to openscad_deps and it worked fine

Sorry for locale.

t-paul, 2023-09-12:
ldd: `libhogweed.so.6 => /lib64/glibc-hwcaps/x86-64-v3/libhogweed.so.6.8 (0x00007f66a0227000)`
compile: `/usr/lib64/libhogweed.so.6`

That does not look very good.

ElektroBOXua, 2023-09-12:
My knowledge of building/linking process ends at this point. But yeah, it is not really consistent for me.

ElektroBOXua, 2023-09-12:
Ok, i have discovered that the problem lies in "gmp 6.0.0", which is loaded from openscad_deps.
The system library libhogweed is incompatible with this version of gmp, so removing this dependency and using the system one also fixes that problem.

t-paul, 2023-09-12:
Recent distributions should have pretty much all the dependencies as development packages, so manually building dependences *should* not be needed in most cases.

I don't know how up-to-date the OpenSUSE part of `scripts/uni-get-dependencies.sh` is as it's not really possible to maintain that for everything, but you could give that a try and maybe help updating it if needed.

closed 2023-09-12 by ElektroBOXua