openscad/openscad #4742: Font sizes unlinked in editor panes

christianvdstap, 2023-09-15:
**Describe the bug**
New editor pane opens with default font size even when the current pane is already in a different font size. However, the font size increase / decrease is linked which means that the editor panes keep having different font sizes.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a file
2. Reduce font size
3. Open other file or create new file
4. Notice the default font size in new pane

**Expected behavior**
Font sizes are the same in all editor panes OR the font sizes can be set per editor pane

**Code reproducing the issue**
NA

**Screenshots**
NA

**Environment and Version info (please complete the following information):**
 - Linux Eruuru 6.2.0-32-generic  32-Ubuntu SMP PREEMPT_DYNAMIC Mon Aug 14 10:03:50 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Openscad-nightly, installed automatically when prompted, 2023.09.13

**Library & Graphics card information**
NA

**Additional context**
NA

jordanbrown0, 2024-09-22:
There are two distinct editor font size controls.  The one in Preferences applies across application restarts and (it appears) is used for new windows.  The menu / keyboard increase/decrease settings appear to be separate for each editor window and not preserved.

So the default font size is controllable across all windows and set in Preferences, and the menu/keyboard setting can be set for each window.

(Confusing?  Yes.  I'm only reporting what I observe.)

jordanbrown0, 2025-10-24:
Please note that that's marked as personal opinion; others may disagree.  Let's take the philosophical discussion off Github.

jordanbrown0, 2025-10-24:
> I like being able to see different tabs at different scaling.

OK, each to his own.

I did track down what's happening, though I don't immediately see how to fix it.

Those two actions (Ctrl+/-) are in theory per-tab, but in practice they are bound to every tab individually.  It's not that the one keystroke changes some global, but rather that it independently changes every tab.  Perhaps it needs to be bound once (not to each tab), and that to a function that does the zoom on the current tab.

Note that the Preferences item would probably be the base, with the per-tab Ctrl+/- being added to that base.

jordanbrown0, 2025-10-25:
The other variation that would be sensible would be to have there only be one zoom level for all tabs in a particular window (or all windows?).  A new window would start at the Preferences value, Ctrl +/- would temporarily zoom, new tab would share that same temporary zoom, Ctrl +/- would temporarily zoom all of the tabs.

So:
Option 1:  all tabs (in one window?  in all windows?) share the same temporary zoom level.  They all match, all the time.
Option 2:  Tabs each have their own temporary zoom level.  (Do new tabs start at the preferences level, or do they inherit from the current tab?  Current behavior is that they start at the preferences value.)

Option 2, starting at the preferences value, is very likely the simplest change.

closed 2025-10-26 by kintel