openscad/openscad #4743: offset_extrude()

your-friend-alice, 2023-09-15:
**Is your feature request related to a problem? Please describe.**

It would be nice to have an `offset_extrude(h)` function, with the same `delta`, `chamfer`, and `r` parameters as `offset()`. This transformation would loft its 2d children through an `offset()` version of itself, positioned at height `h`.

**Describe alternatives you've considered**

it is possible to recreate the desired effect using `hull()`, but that only works for convex shapes.

It is possible to approximate the desired effect using `roof()` to support concave shapes as well, but this is weirdly fragile as `roof()` is an experimental feature and often breaks unexpectedly, and it's also impossible to represent negative `delta` parameters correctly (it behaves as if it's a negative `r` parameter instead).

Here's a library implementing these 2 approximations, you can use it with `use`, or open it directly to see a comprehensive demo of its problems: https://gitlab.com/your_friend_alice/3d-models/-/blob/main/lib/offset_extrude.scad

[image]
[image]
Here's a closer look at an example of using `roof()` to approximate `offset_extrude(delta=-4)`, showing concave corners coming out rounded instead of sharp:
[image]

t-paul, 2023-09-15:
Yes, the original approach in #2079 stalled and there's mostly the conclusion that the logic `roof()` is based on would help solving the issues.

pca006132, 2023-09-24:
Hi, I wonder if difficulty implementing this is due to Clipper not providing point associations between the original polygon and offset polygon? If that association is provided, it seems simple to implement.
I feel like this is something Clipper2 dev will likely be interested in providing.

> Here's a closer look at an example of using roof() to approximate offset_extrude(delta=-4), showing concave corners coming out rounded instead of sharp:

I think this works as intended? In general, when you apply a negative offset to a concave corner with round joint type, the result is round. If you want something else, you may want to use mitered. It is not practical to get sharp corners for every case as that will tend to create spikes for narrow angles. See http://www.angusj.com/clipper2/Docs/Units/Clipper/Types/JoinType.htm

your-friend-alice, 2023-09-24:
@pca006132 one thing that complicates it is that the relationship between points in the original and offset polygon is many-to-many. if there's a radius then one point can map to many points, while since a positive offset can make small concave features disappear, and a negative offset can make small convex features disappear (replacing a larger number of points with a smaller number of points, with no obvious way to map them).

Thinking about this some more, the one-to-many relation due to radius isn't that problematic, it's the ambiguous more-to-fewer mapping from feature removal that's the problem. could we sidestep around it? the problem is that offset() detects wherever the geometry has turned itself "inside out" and trims that off. if we could keep those points, then we maintain the simple relationship between the original and offset points, and cut away the inside-out geometry in 3d space instead of 2d.

your-friend-alice, 2023-09-24:
> when you apply a negative offset to a concave corner with round joint type, the result is round

In this case the joint type is not round though, it's a square joint. And in 2d, `offset()` treats it as a non-round joint; here's an example of offset's behavior in 2d:

```
module shape() difference() {
    circle(r=10);
    square(10);
}
module pretty2d(h=1, line=0.1) {
    color([0, 0, 0]) render() linear_extrude(h) difference() {
        offset(delta=line/2) children();
        offset(delta=-line/2) children();
    }
    color([1, 1, 1]) render() linear_extrude(h) offset(delta=-line/2) children();

}
$fn=45;
pretty2d(1) square(999, center=true);
pretty2d(2) shape();
pretty2d(2) offset(delta=-3) shape();
```

[image]

I've expanded my library and the accompanying demo to move the errors around in the problematic cases, you can see in the latest version that you can set `flip_errors` to trade-off between convex and concave joints being muddled. Just to illustrate that a negative offset with a sharp concave joint is a sensible operation and what that would look like:

[image]

if someone is desperately looking at a solution, you could slice and dice the `flip_errors=true` and `flip_errors=false` outputs to assemble something correct, but the downside is that you'd have to provide a 2d shape to act as the "mask" for selecting the outputs:

```
module offset_extrude_flip_errors_mask(h, delta=0, r=0, chamfer=false, center=false, convex=false) {
    // children(0) = shape
    // children(1) = mask
    module o(flip_errors) offset_extrude(h, delta=delta, r=r, chamfer=chamfer, center=center, convex=convex, flip_errors=flip_errors) children();
    module l() linear_extrude(arbitrary_large) children();
    color([0.8, 0.6, 1]) render() difference() {
        o(false) children(0);
        l() children(1);
    }
    color([0.8, 1, 0.6]) render() intersection() {
        o(true) children(0);
        l() children(1);
    }
}
offset_extrude_flip_errors_mask(h=5, delta=-3) {
    difference() {
        circle(r=10);
        square(999);
    }
    offset(delta=1)rotate(180) square(3);
}
```

(`color` and `render` calls added for clarity)

[image]

pca006132, 2023-09-25:
> ambiguous more-to-fewer mapping from feature removal that's the problem. 

I think this is easy to handle. There are three cases: 1-1 map, 1-0, 1-many. For the case of 1-0, it is just 1-many from the pov of the offset polygon. So you can just handle it using a cone but in the opposite direction.

> looks like clipper2 allows you to attach a unique identifier to the input points with [USINGZ](http://www.angusj.com/clipper2/Docs/USINGZ.htm), one could possibly do more with ZCallback

Great!

> In this case the joint type is not round though, it's a square joint. And in 2d, offset() treats it as a non-round joint; here's an example of offset's behavior in 2d:

So this is just an issue with roof implementation? Should not be an issue if we use clipper offset.

gsohler, 2023-09-27:
> I was thinking about this, and it seems that polygon with holes may
> present some issue. After offsetting, the hole may disappear, and it is
> non-trivial to find out where to connect to.

of course: holes in polygons disappear when using positive offset on it.
the hole will turn into a cut-out pyramid which aligns with the bottom of
the result. as the hole disappears its only a 3d blind hole.
the top point of the pyramid just connects to faces facing down ...

nophead, 2023-09-27:
Yes but the end of the blind hole might be between two slices.

pca006132, 2023-09-27:
Yes I know conceptually where to put them, but I think computing them is not easier than computing 2D offset. Consider the following sketch:
[image]

You have to figure out where the straight line is, which corresponds to the collapsed small arc after offset. This doesn't seem obvious to me, perhaps you need something like voronoi diagram, which is implemented with the `roof` function. But the roof function doesn't do square/miter joints.

gsohler, 2023-09-27:
The center of both circles will stay the same during all time for negative
offset. The right circle will disappear first for sure. not sure if it
disjoints from big circle before it disappears

pca006132, 2023-09-27:
> The center of both circles will stay the same during all time for negative offset. The right circle will disappear first for sure. not sure if it disjoints from big circle before it disappears

Well, my point is that it is non-trivial to find out this information if what you have is just the 2D offset result from Clipper. And I currently don't have an algorithm in mind.

gsohler, 2023-09-27:
Maybe you should look, how roof function translates any 2D polygon into an
offset_extruded object.
the polygons can even have holes and small noses.
Depending on the shape of the polygon, the points of the polygons are
"combined" in a dedicated order at certain heights,
until on the very top only 3 faces make up a pyramid top.
watch it to see, how it works ...

pca006132, 2023-10-27:
wondering, if the roof function supports square joints? I thought about this problem more and I feel like voronoi diagram is the only way, and the only problem you faced seems to be the joint type.

thehans, 2023-10-28:
@pca006132 
I think the term/algorithm you are looking for is [Straight Skeleton](https://en.wikipedia.org/wiki/Straight_skeleton).

The `roof` module (Original PR here #3486 ) supports two methods of offsetting: "voronoi" (default, which gives a radius effect), and "straight" (short for straight skeleton).  You are correct that this information about where edges merge etc is not readily accessible through clipper's functions.  That is the reason `roof` uses a straight skeleton algorithm coming from CGAL library
https://doc.cgal.org/latest/Straight_skeleton_2/index.html

Sample scad code (requires dev snapshot with experimental feature "roof" enabled):
`roof(method="straight") square(); // creates a square pyramid`

pca006132, 2023-10-28:
thanks for the pointer! but if I understand it correctly, straight skeleton is not exactly the same as squared corners as squared corners are clipped for very acute angles (where the proper joint without squaring will be very long)

thehans, 2023-10-28:
OK, it was not clear before what you meant exactly about square corners.  What you describe is equivalent to `offset` with `delta` and `chamfer=true` in OpenSCAD (see the rightmost diagram in this [image from the manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#/media/File:OpenSCAD_offset_join-type_in.svg)).

The existing `roof` does not support this, but I believe a straight skeleton operation could still be used to do this.  It would just require a certain level of control over the starting conditions.  Specifically you would need to be able to define the edge direction, (or perpendicular offset movement direction) *independently* of the endpoint locations.  
This would be needed, because for any reflex vertex, you would need to add an edge which starts at zero length(which obviously has an undefined direction), so this direction would need to be set exactly halfway of the angle between its the neighboring segments.  
After that, the straight skeleton could proceed as usual and should give the desired result.

pca006132, 2023-10-28:
I am not familiar with straight skeleton. Wondering what is the advantage of using straight skeleton over voronoi diagram, which seems to be able to handle the rounded corner case, and should not be too hard to extend to handle chamfer? Is it faster than voronoi diagram?

pca006132, 2023-10-28:
One interesting thing I see with straight skeleton vs voronoi diagram is that the algorithm for voronoi diagram seems to be faster in terms of complexity (O(n log n) vs O(n^2 log n)). I am not entirely sure which one is more robust, but it seems that there are quite a few issues regarding straight skeleton in CGAL in recent years (so even if it is robust, we will need a pretty recent version to compile it)

amatulic, 2024-03-17:
Apparently CGAL 6.5.1 supports 3D offsets and even shrink wrapping. https://doc.cgal.org/latest/Alpha_wrap_3/index.html

This would solve the problem of chamfering, if OpenSCAD provided an interface to that operation.