openscad/openscad #4747: Discussion Python language support -

lkarthee, 2023-09-18:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
Any

**Issue**
Python language support is being added in multiple PRs. I tried to find documentation about this enhancement, i could not find much. Adding a new language support is great, but I am little concerned about the

lkarthee, 2023-09-18:
I not against Python and any other language integration.

> Personally I doubt, that any3D model download site will ever publish python based models and that's not even what i am looking after ...

Printables.com has around [3658 openscad models](https://www.printables.com/search/models?q=openscad). It allows uploading python files. Here is my sample upload -  https://www.printables.com/model/586245-openscad-python-cube/files

> Anybody which does not feel comfortable about that, just never activate the Feature and don't use it.
Anybody which want to express his models in different style and is aware of the security implications, feel free to use it instead.

1) Discussion is about whether there is a way to reduce the security risks and provide users a safe python sandbox  environment. Can [PocketPy](https://pocketpy.dev/)  or [something similar](https://wiki.python.org/moin/PythonImplementations) be used to implement safe python sandbox ?
2) If point 1 is possible what are the tradeoffs and limitations. 
3) Do we need trust and execute anything mode in OpenScad?

The problem is not python as such - it is the unrestricted access to python standard library modules like http.server, http.client, os, json modules. What is the relevance of these modules for a user who wants to create models using OpenScad.

I could do all of the commented things: 
```python
# import json

# delete a file 
# import os
# os.remove('data.json')

# create a json and write a file with list of all folders in a location.
# from pathlib import Path
# p = Path('.')
# data = [str(x) for x in p.iterdir() if x.is_dir()]
# with open('datas.json', 'w') as f:
#    json.dump(data, f)

# start a http server and serve all the contents of the current directory. OpenScad hangs as expected after serve_forever()
# import http.server
# server_class=http.server.HTTPServer
# handler_class=http.server.SimpleHTTPRequestHandler
# server_address = ('', 8000)
# httpd = server_class(server_address, handler_class)
# httpd.serve_forever()

output(cube(10, center = True));
```

Regards
Kartheek

gsohler, 2023-09-18:
I see your point.
For openscad to work with python it needs to have a python shared library which resembles the functionality which is offered from python side. one part of this shared library is the OS module which is different from windows, linux etc.
It might be possible to tweak these os dependant parts to e.g. disable internet functions or to grant only read access to files.
Best approach would probably be to compile this shared library from a patched source code. However this is just an idea and implementation might require additional effort.

t-paul, 2023-09-18:
We are not going to solve Python problems. If you enable it, consider it same level as any other python program.

If you don't enable python, nothing changes to the current state, or to be on the absolutely safe side, it's possible to disable at build time.

If you enable python and decide to run a HTTP server or mine bitcoin, you are free to do so.

jordanbrown0, 2023-09-18:
There are two key problems:
* It is precisely some of the os.* functions that many people want, and in particular the ability to read and write auxiliary files.
* I am not a Python internals expert, but my understanding is that it is *extremely* difficult, if not impossible, to build a sandboxed Python, at least not without lobotomizing it so much that it's not interesting.  As I understand it, the problem is that Python's introspection mechanisms are powerful enough that you can use them to escalate to full access to the underlying system.

A subset Python that *is* sandboxable would address the "want a more powerful programming language" desires, but would not address the desires for additional library and system-access functionality.

lkarthee, 2023-09-19:
[Python Language](https://docs.python.org/3/reference/index.html#reference-index) is different from [Python Standard Library](https://docs.python.org/3/library/index.html#library-index). Quote from python standard library docs.

> While [The Python Language Reference](https://docs.python.org/3/reference/index.html#reference-index) describes the exact syntax and semantics of the Python language, this library reference manual describes the standard library that is distributed with Python. It also describes some of the optional components that are commonly included in Python distributions.

There are multiple interpreters which implement python language and they can be configured and embedded.

Quoting @t-paul 
> We are not going to solve Python problems.

It boils down to what are we trying to implement - either 1 or 2:
1) OpenScad support for Python: Here we are extending python with OpenScad libraries and I can write a normal python program in any editor and run it using `python cube.py` or open in OpenScad to run it. 
```python
# cube.py
from openscad import *

c = cube()
o = output(c)
with open('cube.stl', 'w') as f:
   export_stl(o, f)
```
It is awesome. One can mix different libraries say libfivepy, etc and leverage strength of different libraries. I can use pyenv, virtualenv, set a workflow and use IDE like pycharm. This is a general use case where restrictions are unreasonable. No expectations of security - everything is untrusted so run it anyways. No annoying popups for python files in OpenScad.

2) Python scripting for OpenScad executable: It is a script which has python syntax and semantics. More like a python DSL alternative for scad language. There is no need for explicit import statements for OpenScad. Everything is self contained in OpenScad executable and will run within OpenScad executable context. Here restrictions to python modules are reasonable.
```python
# cube.py
# note: there are no import statements for resolving cube or output. 
c = cube()
o = output(c)
```
This is a more restrictive use case. OpenScad will benefit from Python, but Python ecosystem cannot leverage OpenScad capabilities as it won't have programmatic access through python bindings. I cannot run `python cube.py` as there are no python bindings.

What's the command line story - will there be trusted file concept? 

>  If you enable it, consider it same level as any other python program.

If OpenScad python scripts were to be assumed as general python programs:
- then drop the trusted file popups for every file. Let user see the dialog only once when they enable python support. 
- also provide python bindings, which will then be an implementation of  use case 1 not 2. Without python bindings, it will be unfair for python ecosystem as they won't have programmatic access to OpenScad.

I am little confused as to what we are trying to implement as python support. If you can share any docs/proposal, it would help my understanding. 

> If you enable python and decide to run a HTTP server or mine bitcoin, you are free to do so.

I am not expecting to run HTTP Server or mine bitcoins in OpenScad. If I were to do that, I would be using python for it.

Regards
Kartheek

pca006132, 2023-09-19:
I don't have much idea about the python integration to OpenSCAD, but I think users will likely want to use things like numpy for computation, which requires CPython.

Python provides audit hook https://docs.python.org/3/library/audit_events.html that allows you to intercept standard library calls, which can probably be used to build a sandbox. 

However, I still think that if a user is determined enough, they can break the sandbox. Even browsers that prioritize security and have thousands of security researchers poking everyday will have some serious security bugs every now and then. I doubt if OpenSCAD should provide such a claim that the user can execute whatever script they downloaded in OpenSCAD without worrying about security issues. Scripts that users download are usually pretty simple (in terms of constructs they use) anyway.

jordanbrown0, 2023-09-19:
> Scripts that users download are usually pretty simple (in terms of constructs they use) anyway.

The BOSL2 library is about 56K lines.  It seems not-unlikely that there would be similar-sized Python-based libraries.

lkarthee, 2023-09-20:
@pca006132 Sandbox or no sandbox, either way I feel trusted file dialogs are not needed. They are annoying and ruin developer experience.  

Will there be a way to leverage existing scad libraries like BOSL2 when writing OpenScad python scripts, if we are adding scripting support to OpenScad executable - see option 2 in this [comment](https://github.com/openscad/openscad/issues/4747#issuecomment-1724721565).

gsohler, 2023-09-20:
Ikarthee, you will NOT get the annoying trusted file dialog if you either

* specify --trust-python on the command line( e.g. put it into the symbol at your own risk)
* on your self edited python files
* on files which you trusted before once. trusted files are saved by hash

Yes you are free to mix openscad and python as long as scad language is on the top of the hierarchy.
You can "use" <pythonfile.py>

and then use functions and modules defined in python space

hope this helps

pca006132, 2023-09-20:
> > Scripts that users download are usually pretty simple (in terms of constructs they use) anyway.
> 
> The BOSL2 library is about 56K lines. It seems not-unlikely that there would be similar-sized Python-based libraries.

I think this kind of libraries are easy to audit by the user. Anyway, the user should pay attention to libraries they download from the web.

jordanbrown0, 2023-09-20:
56K lines is "easy" to audit?

> Anyway, the user should pay attention to libraries they download from the web.

They don't need to for OpenSCAD today.  That's kind of the point.  For OpenSCAD today, it is reasonably safe to download models and libraries and run them.  That's not true of a Python environment - there, you must (somehow) trust or audit the stuff you download.

In this respect OpenSCAD is more akin to MS Word or Inkscape or Acrobat Reader - the expectation is that you can download content and look at it with reasonable confidence that it cannot attack your system.

Am I sure that there are no bugs in OpenSCAD that would allow OpenSCAD programs to attack your system?  No, of course not.  But if there are, they are obscure and would probably require complex assembly-language attacks.  On the other hand, absent a sandbox, Python content can just straightforwardly operate on arbitrary files.

Note also that OpenSCAD users are not necessarily programmers.  They may download models and use the customizer, without ever looking at the actual program.  They may also be beginning programmers, people who are going to write some OpenSCAD language but are not experienced in programming in general.  Expecting these users to audit libraries is not reasonable.

pca006132, 2023-09-23:
Thinking about this, for normal users, they probably have no idea what potential consequences are if they enable python support and mark a file as trusted. Yes I know we have a prompt for them, but I think a lot of people will not read it and just click yes. I think we need to do a bit more.

I think we can do several things here:
1. Use the audit API. Some events are allowed, and some events will require user permission. For example, we can allow the script to read and write things in the current directory (maybe subdirectory as well) but requires user clicking yes when it wants to read/write things outside of allowed directory. The user can at least intervene when the script does something weird. I think this will make most attack approaches much harder, as they don't have ways to directly access the web, access arbitrary files and execute shell code.
2. Blacklist certain modules. So the script cannot import stuff from something like ctypes, which can do low level memory access. Or maybe a whitelist + blacklist approach? Only whitelisted modules + names in the current directory - blacklist.

These should be implemented in C:

> https://docs.python.org/3/library/sys.html#sys.addaudithook
> Note that audit hooks are primarily for collecting information about internal or otherwise unobservable actions, whether by Python or libraries written in Python. They are not suitable for implementing a “sandbox”. In particular, malicious code can trivially disable or bypass hooks added using this function. At a minimum, any security-sensitive hooks must be added using the C API [PySys_AddAuditHook()](https://docs.python.org/3/c-api/sys.html#c.PySys_AddAuditHook) before initialising the runtime, and any modules allowing arbitrary memory modification (such as [ctypes](https://docs.python.org/3/library/ctypes.html#module-ctypes)) should be completely removed or closely monitored.

We can allow the user to disable 2 for example, but only via command line arguments. If they don't know how to use command line arguments, maybe they should not be touching these... This is not really foolproof, but I think this will raise the barrier of attacks, so making it a bit more secure.

gsohler, 2023-09-23:
Hi Chun, this is definitely something I should look into and maybe can add
it to my PR.
When coding the python stuff, I didn't even know that such audit stuff
exists, but I am happy to learn.
In case you are experienced with this already, I am happy to receive PRs to
my branch.

Thanks again!

pca006132, 2023-09-23:
I didn't know it exists as well, I just googled it a few days ago when reading about this issue.
I'm probably unable to help implementing this, as there are still a lot of work going on for manifold (e.g. 3D offset) that I still want to implement. However, I am happy to help reviewing this when it is done.

gsohler, 2023-09-24:
This is an example of a working python audit hook.
its actually quite easy to enable.
Not it can be a challenge to develop sophisticated rules

static int PyCheckHook(const char *event, PyObject *args, void *userData)
{
    if(strcmp(event,"open") == 0){
        if(args == NULL) return 0;
        const char *mode_str =  PyToChar(PyTuple_GetItem(args,1));
if(mode_str == NULL) return 0;
if(strcmp(mode_str,"r") == 0) return 0; // read is allowed
if(strcmp(mode_str,"") == 0) return 0; // read is allowed
        const char *path_str =  PyToChar(PyTuple_GetItem(args,0));
char *rpath = realpath(path_str, NULL);
if(rpath == NULL) return 0; // path does not exist
char pwd[80];
getcwd(pwd,sizeof(pwd));
if(strncmp(rpath,pwd,strlen(pwd)) != 0)
{
printf("File write not in current work directory\n");
exit(1);
}
        return 0;
    }
}

lkarthee, 2023-09-25:
I have a draft PR #4758 which is related to new language runtimes. Please share your feedback. Its a WIP - PR contains more  details about it.

WillAdams, 2023-11-15:
I would just like to note that I've been using @gsohler 's version of OpenSCAD with Python in it for a while now, and it's been the "magic sauce" which I needed to get my library:

https://github.com/WillAdams/gcodepreview

to a state where it is easily used (or actually usable?), since it made possible writing out files w/o resorting to the command line, and persistently storing the XYZ coordinate for each G-code command so that it could be used by a later command (and accessed by simpler OpenSCAD modules which access the persistent Python variables behind the scenes). 

I really appreciate all the efforts which were made previously, as well as @GilesBathgate 's RapCAD which allowed me to work on things until this point, and I hope that this can be incorporated into the main program.

[image]

closed 2023-11-29 by lkarthee