openscad/openscad #4749: MacOS build failure: github/openscad/build/objects/lexer.cxx:955:2: error: ISO C++17 does not allow 'register' storage class specifier

paenian, 2023-09-20:
Attempting to build on MacOS Ventura - 13.5.2 - getting errors building lexer.cxx:

```
github/openscad/build/objects/lexer.cxx:955:2: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
        register yy_state_type yy_current_state;
```

Multiple errors in the same vein - also had to install libffi manually.

**To Reproduce**
cmake .. -DEXPERIMENTAL=1
make

**Expected behavior**
build builds :-) 

**Environment and Version info (please complete the following information):**
 - OS: MacOS Ventura - 13.5.2
 - System: intel core I7, 2.3ghz, intel iris plus graphics
 - OpenSCAD Version latest github pull

thehans, 2023-09-20:
That lexer source is generated code, which comes from the tool [flex](https://en.wikipedia.org/wiki/Flex_(lexical_analyser_generator)).

Can you check what version of flex is being used.  When you first configure cmake, it should print what version it detects:
https://github.com/openscad/openscad/blob/0b71427abf50412b4e6a491f028de4401c47efe9/CMakeLists.txt#L443

If possible try updating to a newer flex version maybe to see if that helps.

paenian, 2023-09-20:
Thanks for the fast response!  Indeed, cmake reports flex 2.5.35.
I ran `brew install flex` which installed version 2.6.4_2
rerun `cmake .. -DEXPERIMENTAL-1`
cmake still reports flex 2.5.35

brew doesn't put flex into usr/local, but usr/local/opt/flex - cuz macos provides its own flex, I suppose.

I changed the flex path in CMakeCache.txt to /usr/local/opt/flex/bin/flex, which gets cmake to report the new version, and make is now using it as well, which eliminates the error.

Thanks again!

closed 2023-09-20 by thehans

kintel, 2023-09-25:
Btw., the flex which comes with macOS/Xcode has been 2.6.4 for a while, so my guess is that you have multiple other flex versions installed. If you were to uninstall them all, you should be able to fall back to the one from Apple.