openscad/openscad #4754: A constrain()  function would be nice...

vortex64, 2023-09-24:
I apologize in advance if something like this already exists but I am unaware.

But when building a parametric model.  A built in constrain() function would be convenient.

Similar to this...
https://www.arduino.cc/reference/en/language/functions/math/constrain/

Sure, I could write my own.  But having it built in would be nice.

jordanbrown0, 2023-09-25:
It looks like `constrain(val, a, b)` is equivalent to `max(a, min(b, val))`.

vortex64, 2023-09-25:
Acknowledged.  But a constrain() wrapper would make openscad code a  little more readable.

I'm starting to be a bit more responsible in adding checks to parametric models I create and post online.

nophead, 2023-09-25:
```function constrain(val, a, b) = max(a, min(b, val));```

Just add that. Why do you need it to be build-in?

vortex64, 2023-09-25:
I guess I will do that.    I just thought it would be nice that it would be built in.
Parametric models need a checks like this almost all the time.  

OpenSCAD has been very comprehensive and I don't typically have a separate include file for separate macros and defines like this.   I guess I'll have to start doing that.   For better or worse...

pca006132, 2023-09-26:
If you want some helper functions, you should probably check BOSL2. I think the core language should be minimal, and only implement things that are really hard to do without putting it in the core.

thehans, 2023-09-26:
I don't think Arduino is the best point of reference for a well thought out API.  I would call this a [clamp function](https://en.wikipedia.org/wiki/Clamping_(graphics)).  

That is the name used by [GLSL for example](https://registry.khronos.org/OpenGL-Refpages/gl4/html/clamp.xhtml).  Another interesting thing to note is that, GLSL accepts vector inputs (e.g. vec2 vec3 vec4) and will clamp each component individually to lower/upper floating point bounds.

vortex64, 2023-09-26:
I am aware of BOSL2.  As said from the beginning, I  can write my own helper.   It just seemed like something that would help new OpenSCAD users to adopt good form.   In other words if they saw more usage of "constrain()" in examples and help files then they would naturally adopt it in their own projects.   I think we can all agree that many new OpenSCAD users are not software oriented individuals.

jordanbrown0, 2023-09-27:
But is it better practice to silently "fix" out-of-range values, or to have them trigger an assertion failure so that the human will fix them right?

vortex64, 2023-09-27:
Jordan,   I think it is up to the scad author.  It can be quite a burden having to explicitly add a string describing all possible errors for a multitude of parameters in each assertion.    To be honest I am not quite sure how Thingiverse's online customizer handles assertions.   Something worth investigating.  In any case, a "silent fix" (i.e., constrain()) is expedient and may prevent a model taking an eternity render or otherwise "blowing up".   

I guess finally in my argument   Is adding "constrain()" so difficult?   I have OpenSCAD source buildable under MSYS64 on my laptop for other reasons.   I think I'll take a look.

nophead, 2023-09-27:
It isn't technically difficult to add but its a language design principal to not add facilities to the core language that can easily and efficiently done in user land. Look at other languages like Python and C++. Nearly all functions are added by libraries, even more than OpenSCAD which has trig functions and min and max built in.

closed 2023-09-27 by vortex64

jordanbrown0, 2023-09-27:
Yeah, like nophead said.  The problem is not that it's difficult - it's easy.  The problem is that there are a million little "nice to have" features that could be added, and they each make the documentation and implementation just that much bigger and more complicated.  Plus, if you add enough "trivial" features, eventually you'll make a stupid mistake implementing one, and it'll have a bug that crashes the application when you tickle it just wrong - which you can't get if you implement in OpenSCAD-language.  (Well, OK, you can create infinite loops.)

Also, OpenSCAD-language implementations are often significantly more concise.  Note that the implementation of this function is one line of OpenSCAD-language.  It would be a dozen or two lines of C++.

We really should have a state-of-the-art standard library included with OpenSCAD.  MCAD is sort of an attempt at that, but has not had any love for many years.

vortex64, 2023-09-27:
"We really should have a state-of-the-art standard library included with OpenSCAD. MCAD is sort of an attempt at that, but has not had any love for many years."

Agree, but I don't think "language design principals"  and "core language should be minimal" philosophical views are the best way to look at this.

I think one should at at how OpenSCAD ultimately gets used.     If new users observe range checking in the provided examples they would easily identify the utility and be more likely to incorporate similar checks in their own projects.   
But they will unlikely seek out a library to provide this on their own. 

Just now I went to Thingiverse and tried to open an OpenSCAD model with their online customizer.  I wanted to see what would happen if unreasonable parameters were entered.   Unfortunately I was met with:
"503 Service Temporarily Unavailable".   

Hmmm...I wonder if unreasonable data entry was making it crash too often :)

Anyhow, and for the last time, I'll not argue the point any further.

jordanbrown0, 2023-09-28:
> If new users observe range checking in the provided examples they would easily identify the utility and be more likely to incorporate similar checks in their own projects.  But they will unlikely seek out a library to provide this on their own (or even consider range checks).

Yep!  Hence why we need a library *included* with OpenSCAD, so that the majority of examples would start out with `include <stdlib.scad>` or something like that, just like C programs canonically start with `#include <stdio.h>`.

> Anyhow, and for the last time, I'll not argue the point any further.

At this point I'm not trying to get you to understand why we don't want to add "easy" functions to the core.  It's a *very* good question, and we *do* need to figure out an appropriate balance.  I'm thinking out loud about ways to achieve that.

A standard library addresses several problems:
* Concise definition in OpenSCAD-language instead of C++
* Larger range of people can contribute (because it's not C++, and because you don't have to understand OpenSCAD internals and build OpenSCAD from source)
* Bug resistant (again, through not being in C++)
* You don't *have* to include the library if you don't want.
* Readily available to all users (because it's included)
* Incompatible upgrades are not horrible - `include <stdlib2.scad>`

But it does not address some:
* Have to be careful to avoid having the documentation explode.
* Features have to be designed carefully, so that you don't have to make incompatible changes or duplicate them when you realize they weren't quite right the first time.
* Namespace issues, avoiding having new feature names collide with existing user usage.  (OpenSCAD "use" semantics help there.  You can also segregate into multiple includes.)