openscad/openscad #4755: Build error WSL2/Debian

FatherGeoffHorton, 2023-09-25:
**Describe the bug**
make fails on WSL2 Debian. Here's the relevant output:
```
[ 64%] Building CXX object CMakeFiles/OpenSCAD.dir/src/glview/OffscreenContextGLX.cc.o
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc: In member function ‘bool OffscreenContextGLX::createGLXContext(size_t, size_t, bool)’:
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:116:7: error: ‘GLX_CONTEXT_MAJOR_VERSION_ARB’ was not declared in this scope
  116 |       GLX_CONTEXT_MAJOR_VERSION_ARB, static_cast<GLint>(majorGLVersion),
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:117:7: error: ‘GLX_CONTEXT_MINOR_VERSION_ARB’ was not declared in this scope
  117 |       GLX_CONTEXT_MINOR_VERSION_ARB, static_cast<GLint>(minorGLVersion),
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:118:7: error: ‘GLX_CONTEXT_PROFILE_MASK_ARB’ was not declared in this scope; did you mean ‘GL_CONTEXT_PROFILE_MASK’?
  118 |       GLX_CONTEXT_PROFILE_MASK_ARB, compatibilityProfile ? GLX_CONTEXT_COMPATIBILITY_PROFILE_BIT_ARB : GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       GL_CONTEXT_PROFILE_MASK
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:118:60: error: ‘GLX_CONTEXT_COMPATIBILITY_PROFILE_BIT_ARB’ was not declared in this scope; did you mean ‘GL_CONTEXT_COMPATIBILITY_PROFILE_BIT’?
  118 |       GLX_CONTEXT_PROFILE_MASK_ARB, compatibilityProfile ? GLX_CONTEXT_COMPATIBILITY_PROFILE_BIT_ARB : GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                            GL_CONTEXT_COMPATIBILITY_PROFILE_BIT
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:118:104: error: ‘GLX_CONTEXT_CORE_PROFILE_BIT_ARB’ was not declared in this scope; did you mean ‘GL_CONTEXT_CORE_PROFILE_BIT’?
  118 |    GLX_CONTEXT_PROFILE_MASK_ARB, compatibilityProfile ? GLX_CONTEXT_COMPATIBILITY_PROFILE_BIT_ARB : GLX_CONTEXT_CORE_PROFILE_BIT_ARB,
      |                                                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                     GL_CONTEXT_CORE_PROFILE_BIT
/home/geoff/openscad/src/glview/OffscreenContextGLX.cc:122:9: error: ‘glXCreateContextAttribsARB’ was not declared in this scope
  122 |     if (glXCreateContextAttribsARB) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:2217: CMakeFiles/OpenSCAD.dir/src/glview/OffscreenContextGLX.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:508: CMakeFiles/OpenSCAD.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

```

**To Reproduce**
I started with a clean install of Debian on WSL2, then:
```
sudo apt update
sudo apt upgrade
sudo apt install git
git clone https://github.com/openscad/openscad.git
cd openscad/openscad
git submodule update --init --recursive
sudo ./scripts/uni-get-dependencies.sh
./scripts/check-dependencies.sh
mkdir build
cd build
cmake .. -DEXPERIMENTAL=1
make
```

**Expected behavior**
Successful compilation

**Environment and Version info (please complete the following information):**
uname -r: 5.15.90.1-microsoft-standard-WSL2
/etc/issue: Debian GNU/Linux 12

kintel, 2023-09-25:
How does WSL deal with graphics? Do they provide an X11 server, or do you connect directly to Windows for graphics?
Do you know if OpenGL (accelerated or not) is supported in WSL?

gsohler, 2023-09-25:
Hi, same error also appears in 
Fedora release 37 (Thirty Seven)
All packages are updated just before

kintel, 2023-09-25:
@gsohler oh, that might be my fault. I'll have a look!

kintel, 2023-09-25:
I think I know what happened; I regenerated GLAD without a required GLX extension when I merged the EGL context work yesterday. Should be fixed in #4756

closed 2023-09-25 by kintel

FatherGeoffHorton, 2023-09-25:
> How does WSL deal with graphics? Do they provide an X11 server, or do you connect directly to Windows for graphics? Do you know if OpenGL (accelerated or not) is supported in WSL?

It runs X11 (and Wayland, if needed).

FatherGeoffHorton, 2023-09-25:
Can confirm it's fixed. Thanks.

gsohler, 2023-09-25:
in Fedora37 i still get same error. not sure if its an direct or indirect effect.

gsohler, 2023-09-25:
my bad ... of course it compiles also in fedora37. thank you for fixing!

PR #4756 (2023-09-25): Regenerated GLAD: Added GLX_ARB_create_context
Fixes #4755