openscad/openscad #4763: New Language?

pca006132, 2023-09-28:
I want to discuss about the possibility of re-designing OpenSCAD, which can call existing OpenSCAD code using some sort of FFI (similar to how the Python language integration calls OpenSCAD modules).

Disclaimer: I am not really a seasoned OpenSCAD user, I only have limited experience writing simple OpenSCAD code. This issue is mainly to encourage others to talk about the language design, and not to spam the mailing list.

# Background

I think the major issue in openscad is the semantics $variables.

1. #3781 (and this issue is blocking release as well)
2. #3881
3. #350 ($variable semantics limits the usefulness of no implicit union)

There are also limitation caused by the language design:

1. Implicit union (which #350 hopes to get rid of but it is changing the semantics)
2. Geometry being second-class in the language (#4478 wants to address but it will also cause semantics changes)
3. Lack of multi-file export (#1041)

... and probably more. I think there are fundamental issues with the language design here (the existence of $variables, and how we handle geometries), and the best way to fix these issue is to start from scratch.

# Ideas
## Variable Binding

Functional programming languages usually use variables (or let-bindings, as they are not really mutable) like this:

```
let a = 1 in // binding
  a + 1 // using the binding

let a = 1 in
let a = 2 in // shadows a = 1
  a + 1 // using a = 2
```

This seems better than the behavior in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#Variables_are_set_at_compile-time,_not_run-time (not sure if it is really obsolete or not, and whether the obsolete behavior is expected...)
```
// The value of 'a' reflects only the last set value
   a = 0;
   echo(a);  // 5
   a = 3;
   echo(a);  // 5
   a = 5;
```

If module level default value for variable is wanted, we can treat the module as something like a function, and *call* the module with the required parameters. For example (just random syntax)

```
// module
{ a = 1 }: // module parameter with default value
{ x = b: a + b } // x is a function from int to int
```

```
// using the module
let m = import modulename { a = 10 } in
  m.x(1) // 11

// alternatively
let open modulename { a = 10 } in
  x(1) // import the names inside the module
```

This will also allow you to import the module multiple times with different parameters.

## $variables

> https://www.haskell.org/hugs/pages/users_guide/implicit-parameters.html
> Dynamic binding of variables is a notion that goes back to Lisp, but was later discarded in more modern incarnations, such as Scheme, as dynamic binding can be very confusing in an untyped language.

The modern way of solving this problem is through implicit parameters, e.g. https://docs.scala-lang.org/tour/implicit-parameters.html

For OpenSCAD, I think we can just do it by having an implicit parameter `$ctx` by default for every function, with type `Context`. `$ctx[a -> 1]` returns a new context with field `a=1`, and can be accessed as `$ctx.a`. Here are some examples:

```
cube_module(){$ctx[fn -> 128]} // override fn to 128
cube_module() // using the default fn
other_user_defined_module(){Context[foo -> 128]} // pass some user defined data and forget everything else
```

## Misc

We should be able to return geometries as function results, first-class functions (so we can pass functions into parameters), and probably attach fields to geometries as if the geometry is a struct.

E.g. `cube() {north = pos[0, 1, 0], foo = 1}`, and such `north` field with `pos` type will be transformed when we perform linear transformation on the geometry for example, while other data remains the same.

jordanbrown0, 2023-09-28:
I'm not aware of any semantic changes in #4478.

OpenSCAD already has first-class functions.  They're even in 2021.01.

If we're going to do a language redesign, I'd move to a general purpose language, or at least as close to one as we can get and still get sandboxing and simple minimal cases.  Somebody said that there's a Python that isn't based on CPython; maybe it wouldn't have the same sandboxing problems.  The weirdest thing that I would want in a language is that if a function returns geometry, and that return value is not consumed - if it ends up as a statement - then it should get added to the current subassembly, not discarded.  This is sort of analogous to the behavior in most REPLs, where an expression at the top level gets evaluated and printed, not discarded.

jordanbrown0, 2023-09-28:
>OpenSCAD already has first-class functions. They're even in 2021.01.

But to be clear, it doesn't have first-class modules.  Unlike most (all?) modern languages, OpenSCAD has a strong dividing line between functions (which return values) and modules (which don't).  They are different constructs with different syntax and different namespaces.

lar3ry, 2023-09-28:
I'm fine with many of the suggestions, but only if I am not forced to use Python.

pca006132, 2023-09-28:
On Fri, Sep 29, 2023, 1:45 AM jordanbrown0 ***@***.***> wrote:

> I'm not aware of any semantic changes in #4478
> <https://github.com/openscad/openscad/pull/4478>.
>
I think there are comments near the end saying that early return are
problematic?

OpenSCAD already has first-class functions. They're even in 2021.01.
>
Ah, actually what I want is first-class function + geometry.

> If we're going to do a language redesign, I'd move to a general purpose
> language, or at least as close to one as we can get and still get
> sandboxing and simple minimal cases. Somebody said that there's a Python
> that isn't based on CPython; maybe it wouldn't have the same sandboxing
> problems.
>
Yes there are alternative Python implementations. I think the main problem
is that the mature implementations usually have a subset of the standard
library implemented...

I think the most commonly used language embedded in other programs is lua,
so I think it will probably be secure enough for OpenSCAD.

> The weirdest thing that I would want in a language is that if a function
> returns geometry, and that return value is not consumed - if it ends up as
> a statement - then it should get added to the [...]
The problem is that there will be no way of discarding the returned
geometries.
Comparing this to REPL doesn't make sense because if you have something
like (1; 2; 3), where ; separates different statements, only 3 will be
printed (e.g. in scala), only the last statement is the value.

>

nophead, 2023-09-28:
No problem with a new language but I have invested 10 years in OpenSCAD, so
I will continue to use the last release, as it does all I need to do.

jordanbrown0, 2023-09-28:
The comments that you're referring to in #4478 were a side conversation.  #4478 adds object syntax akin to JavaScript and sort of like Python dict syntax, where in an expression `{...}` defines an object.  This means that, for instance, you can have a function that returns an object:
```
function f(args) = { a: 1, b: 2 };
```
or as a function literal
```
f = function(args) {a: 1, b: 2};
````

There was a separate conversation about (somehow) trying to unify the syntax of functions and modules, since for a lot of purposes they are entirely separate sub-languages.  (For instance, there is no "if" in functions; you have to use `a?b:c`.)  The obvious way to do that, or at least the obvious syntactic direction to head, would be something like
```
function foo(args) = { ... something that looks more like module syntax ... };
```
Such a syntax would conflict with #4478's object literal syntax, so I pointed out that conflict.  The example that I used of this hypothetical module-like function syntax used `return` statements, and somebody pointed out that that would be an entirely new concept in OpenSCAD.  They're not wrong, but it was a hypothetical example of a new syntax that would likely conflict with #4478, not something that is in (or proposed for) #4478.

pca006132, 2023-09-28:
Oh OK thanks, I was not reading the entire conversation because it is too long.

I think it will be nice if we can integrate function and modules, because there is no reason why they should be separate IMO. You don't need return statement if you always return the last expression. For if expressions, return the last expression in the branch. And we can remove for statement and replace with a range + map function, which is more functional.

t-paul, 2023-09-28:
I think discussion about language design would be nice (an should accept the ideas from the existing OpenSCAD2 proposal).

I don't think discussing a "replacement" or "alternative" is useful. There are existing alternatives, if anyone wants to help out in one of those projects, that's great, creating more new projects, thinning out the developer power is just making things worse. Code CAD needs projects that grow, not more different projects because of pet language number 3517 because of feature xyz.

The main ideas of the existing Python PR and the reasony why I'm supporting it are:
* It is an optional feature extending possibilities while not changing the existing OpenSCAD
* It's a way of actually sharing lots of things, e.g. geometry and GUI potentially growing the dev power and general interest

jordanbrown0, 2023-09-28:
[ with respect to a hypothetical language where unused returned geometry would be added to the current subassembly, not discarded ]

> The problem is that there will be no way of discarding the returned
geometries.

Not quite true.  You could have a module - what C would call a void function - that takes the geometry value as an argument and doesn't use it.  You could assign it to a variable that you never look at again.  But since the geometry would be *the* return value from the expression, why would you want to discard it?

The comparison to REPL is a loose one.  In a REPL, each thing that you type is evaluated, and the resulting value is printed.  In a REPL, that applies only to the top level of what you type; if you type something that has multiple statements it does not apply to the individual statements.  Here, the idea was that if a statement ever yields a geometry value, that geometry value is "printed" into the current subassembly.  The idea is to make it so that simple constructs like `cube();` can work, without requiring any additional processing, and so that you can define a collection of objects simply, with something like `function foo() { cube(); sphere(); }` - roughly the same level of syntactic complexity that OpenSCAD has for its modules.

jordanbrown0, 2023-09-28:
Merging functions and modules into one concept is definitely one of the things I would do in a new language, so that `cube();` adds a cube to the current subassembly and `x = cube();` puts a cube into a variable.  (#4478 does that, somewhat, because you can embed a module invocation in a geometry expression, so you can say `x = {{ cube(); }};`, and you might well find yourself using functions for everything instead of modules.)

---

You are of course welcome to design a new language, but I'm not interested.  The whole reason that I want programmatic CAD in a general-purpose language is to get *out* of the language design business.  The world doesn't need yet another specialty language.  The only thing I'd be interested in along those lines would be some small extension to an existing language to get the top-level-geometry-expression behavior that I want.

And, advice:  don't do a functional language.  "Why doesn't x=x+1 work?" is probably one of the top three questions asked about OpenSCAD.

pca006132, 2023-09-28:
Well idk, I view this as something like python2 vs python3 style changes because the current semantics causes issues and it is hard to fix it without breaking existing code. But if there is a lack of interest I guess I should close this.

closed 2023-09-28 by pca006132

jordanbrown0, 2023-09-28:
*Others* might be interested in a totally new language, or something OpenSCAD-like with significant incompatibilities.  *I'm* not, which doesn't at all prevent it from happening.

jordanbrown0, 2023-09-28:
I've spent a fair amount of work on #4478 and thinking about things like OO-like features, and while one side of me says "I could do that, and it would be fun", another side says "and it would be reinventing a wheel that has been invented several times already, only mine would be less mature".

pca006132, 2023-09-29:
I just feel like it may not be appropriate to discuss here because the devs don't like the idea of a new language. And I don't really have a plan to make a new project.
My goal is to improve existing projects, but not to make the situation worse.