openscad/openscad #4766: [Manifold] different fn numbers cause artifacts

dlarue, 2023-10-03:
**Describe the bug**
enabled Manifold library in Features and some $fn values cause artifacts within other shapes the object with the different $fn number is union'ed with. 

**To Reproduce**
Steps to reproduce the behavior:
get the planets thing: https://www.thingiverse.com/thing:1430614
set the sides/fn value to 125 for the first planet, mercury :         make_planet(r_mer, theta_mer, 125);
notice the artifacts on the surface of the sun
set the sides/fn value to 124 for the first planet, mercury and see those artifacts go away. 

**Expected behavior**
Expecting rendering with much smoother surfaces and no change in the unioned object's surface

**Code reproducing the issue**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
[OpenSCAD](https://www.openscad.org/) version 2023.09.25.nightly (git 14d0ae2)
-OS Kubuntu 22.04 on old MacBook Air with 4GB of RAM
OpenSCAD Version: 2023.09.25.nightly (git 14d0ae2)
System information: Linux 5.15.0-84-generic #93-Ubuntu SMP Tue Sep 5 17:16:10 UTC 2023 x86_64 Ubuntu 22.04.3 LTS 4 CPUs 3.73 GB RAM
User Agent: OpenSCAD/2023.09.25.nightly (git 14d0ae2) (Linux x86_64; Ubuntu 22.04.3 LTS)
Compiler: GCC "11.2.0" 64bit
[image]
[image]

pca006132, 2023-10-05:
Hi, can you verify if the rendering artifact is due to manifold or OpenGL problem? On my machine I don't see any artifact:

[image]

And the result STL file looks fine (the top triangulation is sparser due to how we simplify the mesh):
[image]

Maybe you can look at the result STL with other programs such as f3d or meshlab and look at the triangulation? I'm wondering if the artifact is caused by OpenGL because the triangles are too fine.

Also, I noticed that the scad file is slightly different from the one in your screenshot. Maybe you can post the complete scad file so we can reproduce it?

dlarue, 2023-10-05:
I opened the STL in Meshlab and was shocked at all the other issues I saw which I didn't see in the Rendered image. I'm attaching 2 files, one produces a clean STL and a Testing version which generates an STL with many surface issues. zipping them up and including stl files since github won't let me upload scad files.
[planets_9.zip](https://github.com/openscad/openscad/files/12819580/planets_9.zip)

dlarue, 2023-10-05:
This issue seems to be unrelated to the Manifold rendering and only showed up when it was possible to increase the face numbers and get fast rendering(seconds vs many minutes). I tested using using an fn size for the Mercury planet at 24 faces vs 135 faces and surface mesh issues show up on the Sun surface with Mercury face count of 135 in both the CGAL rendered model and the manifold library rendered version but not when the Mercury face count is 24.

pca006132, 2023-10-05:
Are these the artifacts you see?
[image]

These are *probably* overlapping triangles, related to a triangulator bug that we are now trying to fix.

I don't see anything else in the stl.

dlarue, 2023-10-05:
What I noticed and have been looking at is down at Mercury. I'll get some screen shots.

pca006132, 2023-10-05:
[image]
[image]

the problem is that I saw nothing wrong with mercury... I wonder if this is some GPU problem.

btw the triangulator bug is in manifold: https://github.com/elalish/manifold/issues/530

dlarue, 2023-10-05:
I could be GPU related and I will try this on another system after I get back from a site this afternoon. It is triggered by different face numbers and some other factors I don't quite know about because previously it would render the aberrations using the 125(or was it 124) number of faces but today I did not see that when I changed only the Mercury face number. I randomly tired 135 and it appeared.

I have an i7 / 16GB RAM machine with an 840 NVidia GPU I will try on this afternoon.  And if the triangulation bug mentioned is only related to the manifold library then yes, this issue is not a manifold-only issue since it shows up on the nightly build with Manifold enabled and disabled.  AND it renders the aberrations and worst to the STL model.  planets9Testing.stl is garbage.

dlarue, 2023-10-05:
I just reproduced it on a booted Lubuntu 22.04 iso, using OpenSCAD-Nightly appImage dated 9/30 but I could only reproduce it so far if I enabled the Manifold option.  I reproduced by using the planets9.scad and changed only the Mercury planet from 24 faces to 124 faces and it shows up in the STL too. If I change the face count to 125 the aberration goes away.

dlarue, 2023-10-05:
Another thing I noticed was after fresh startup with Manifold option off, rendering with 124 or 125 produced no aberrations. But then I turn Manifold option on it'll manifest the aberration at 124 and it goes away at 125. Here's where another issue shows up, when I go in an disable the Manifold option, rendering is still really fast and the aberration continues to show up. So it looks like the Manifold option does turn on with the option selection but it does not switch off when the option is switched off until after a restart of the application.

pca006132, 2023-10-06:
can you post a screenshot with meshlab?

dlarue, 2023-10-06:
It's the same as what I see in OpenSCAD, the same as the first pics.

[image]

dlarue, 2023-10-06:
this screen grab of Meshlab showing the aberration is from the i7 16G NVidia GPU Lenovo machine running Lubuntu 22.04.3? from bootable ISO.

[image]

pca006132, 2023-10-14:
I wonder if anyone else can help me with this? I cannot see any artifact in the mesh provided and cannot reproduce the problematic rendering using meshlab. Not sure if this is something related to OpenGL rendering or issue with the resulting STL file.

t-paul, 2023-10-14:
I have seen similar effects with that https://github.com/t-paul/connection-box design after slicing. I'm not 100% sure it's the same thing, but I'm trying to trim down the model to a simpler reproduction of the issue.

[image]

dlarue, 2023-10-14:
Here's a bit of it trimmed which still triggers the issue on my machine.
Note, I rendered first with Manifold feature OFF then turned Manifold feature ON and rendered and it showed up.
I spent about 15 minutes trying to cause the issue and forgot to check of Manifold was turned ON and it was not.
I turned it ON and there was the feature.

`//!OpenSCAD
fullsun = false;

module make_planet(r_pl, th_pl, sides, xoffset=0) {
  translate([0, 0, z_off_sun]){
    rotate([th_pl, xoffset, 0]){
      translate([0, 0, r_sun]){
        {
          $fn=sides;    //set sides to sides
          sphere(r=r_pl);
        }
      }
    }
  }
}

function comp_dth(r_1, r_i) = 180 * (((3 + (r_1 + r_i)) / r_sun) / 3.14159);

d_mer_km = 4878;
d_ven_km = 12104;
d_earth_km = 12755;
d_jup_km = 142796;
d_sun_km = 1392600;
sf_r = 2.50; //2.25; // scale 2,250,000,000,000
r_mer = sf_r * (d_mer_km / d_earth_km);
r_ven = sf_r * (d_ven_km / d_earth_km);
r_earth = sf_r * (d_earth_km / d_earth_km);
r_jup = sf_r * (d_jup_km / d_earth_km);
r_sun = sf_r * (d_sun_km / d_earth_km);
z_off_sun = r_sun * -0.9365;
h_block = 3 * r_sun;
width = 2 * r_jup + 3 * r_earth;
theta_mer = 340.8;
th_ven = theta_mer + comp_dth(r_mer, r_ven);

difference() {
  union(){
    union(){
      color([0.6,0.6,0.6]) {
        make_planet(r_mer, theta_mer, 124);
      }
      color([0.89,0.77,0.55]) {
        make_planet(r_ven, th_ven, 125);
      }

    }
    // SUN
    color([1,1,0.2]) {
      difference() {
        translate([0, 0, z_off_sun]){
          {
            $fn=192;    //set sides to 192
            sphere(r=r_sun);
          }
        }

        if( !fullsun ) {
        translate([0, 0, (h_block / -2)]){
          cube([h_block, h_block, h_block], center=true);
        }
        translate([(h_block / 2 + width / 2), 0, 0]){
          cube([h_block, h_block, h_block], center=true);
        }
        translate([(-1 * (h_block / 2 + width / 2)), 0, 0]){
          cube([h_block, h_block, h_block], center=true);
        }
        } //fullsun
      }
    }
  }

  if( !fullsun ) {
  translate([0, 0, (h_block / -2)]){
    cube([h_block, h_block, h_block], center=true);
  }
  } //fullsun
}
`

dlarue, 2023-10-14:
setting the two spheres into a cube does it too.

`//!OpenSCAD
fullsun = false;

module make_planet(r_pl, th_pl, sides, xoffset=0) {
  translate([0, 0, z_off_sun]){
    rotate([th_pl, xoffset, 0]){
      translate([0, 0, r_sun]){
        {
          $fn=sides;    //set sides to sides
          sphere(r=r_pl);
        }
      }
    }
  }
}

function comp_dth(r_1, r_i) = 180 * (((3 + (r_1 + r_i)) / r_sun) / 3.14159);

d_mer_km = 4878;
d_ven_km = 12104;
d_earth_km = 12755;
d_jup_km = 142796;
d_sun_km = 1392600;
sf_r = 2.50; //2.25; // scale 2,250,000,000,000
r_mer = sf_r * (d_mer_km / d_earth_km);
r_ven = sf_r * (d_ven_km / d_earth_km);
r_earth = sf_r * (d_earth_km / d_earth_km);
r_jup = sf_r * (d_jup_km / d_earth_km);
r_sun = sf_r * (d_sun_km / d_earth_km);
z_off_sun = r_sun * -0.9365;
h_block = 3 * r_sun;
width = 2 * r_jup + 3 * r_earth;
theta_mer = 340.8;
th_ven = theta_mer + comp_dth(r_mer, r_ven);

difference() {
  union(){
    union(){
      color([0.6,0.6,0.6]) {
        translate([0,0,2])make_planet(r_mer, theta_mer, 224);
      }
      color([0.89,0.77,0.55]) {
        translate([0,0,-1])make_planet(r_ven, th_ven, 225);
      }

    }
    // SUN
    color([1,1,0.2]) {
        rotate([0,0,0]) translate([0, 85, -96]){
          {
            rotate([0,0,0])cube([20,20,200],center=true);
          }
        
      }
    }
  }

  if( !fullsun ) {
  translate([0, 0, (h_block / -2)]){
    cube([h_block, h_block, h_block], center=true);
  }
  } //fullsun
}
`

jordanbrown0, 2023-10-15:
Here's what looks like a pretty minimal demonstration.
```
difference() {
    union(){
        translate([5, 5, 2]){
            rotate([90, 0, 0]){
                    $fn=7;
                    sphere(r=1);
            }
        }
        cube([10,10,2]);
    }
    translate([0, 0, -20000]){
        cube(1);
    }
}
```
The exact dimensions and location of the sphere and first cube do not seem important (as long as they are the same relationship).  Strangely, the Z position of the negative cube is *very* important; the artifacts get much more prominent as it moves further away.  It was moving it so far away that let me get the $fn down to something reasonable.

pca006132, 2023-10-15:
Wow thanks! This is a very succinct repro.

I tested this on the latest version of manifold and there is no such issue, so I guess this will be resolved when we update manifold (have to wait a bit for the triangulator fix)

dlarue, 2023-10-15:
What is odd is that it does indeed render with the issue when I have started OpenSCAD with Manifold enabled but if I disable it, restart OpenSCAD, render then it isn't there AND when I turn Manifold on and render again it still does not show up.

t-paul, 2023-11-07:
Should be fixed via https://github.com/openscad/openscad/pull/4728.

closed 2023-11-07 by t-paul