openscad/openscad #4767: Automatic backup file has wrong name

jon-bondy, 2023-10-04:
**Describe the bug**
I opened a file, did an F5 or F6, then opened a second file, and then did an F6.  The backup file for the final F6 was given the name of the first file, not the second file

**To Reproduce**
See above

**Expected behavior**
See screen shot

**Code reproducing the issue**
Any code should work

**Screenshots**
See screen shot

**Environment and Version info (please complete the following information):**
 - Windows 10
 - AMD CPU
 - [OpenSCAD](https://www.openscad.org/) version 2023.09.10 (git dcaf25f0e)

[image]

jordanbrown0, 2026-10-22:
The first time it's called for a particular MainWindow, MainWindow::saveBackup() carefully determines the backup path, creating it if necessary, and the base name of the current file.  It then creates a temporary file name based on that path and that name.
The second time that it's called, it again determines the backup path and the base name of the current file... and then discards them and uses the temporary file name created the first time.

The intent was presumably to reuse the same temporary name for successive backups of the same file, but it is over-enthusiastic.

Probably the right answer is to store the backup name in the activeEditor, rather than in the MainWindow.

jordanbrown0, 2026-10-22:
To be fair, the intent of the automatic save is to provide a backup in case the preview or render crashes or hangs.  It isn't really intended to save your bacon when something else causes you to lose the primary file.  (Consider, for instance, that there is no periodic automatic save and so an ordinary power failure can cause you to lose your edits.)

jordanbrown0, 2026-10-22:
Nor does it change across Save-As.

closed 2026-10-24 by jordanbrown0