openscad/openscad #4769: [Manifold] Produces extra zero thickness face

adrianVmariano, 2023-10-07:
When this code is rendered without manifold it works correctly and produces this:

[image]

But with manifold enabled, it produces a zero thickness face over the end of the model:

[image]

```
include <BOSL2/std.scad>

a=rands(0,1,1,seed=444);

inches = 25.4;
eps = 0.01;
slop = 0.5;

// camera
ch = 40.86 + slop;	// height
cd = 26.05 + slop;	// depth (X)
cw = 84 + slop;		// width (Y)
cc = 40;			// cable
cmo = 6.5;			// mount offset

th = 2;

// snoot
sl = 7 * inches;		// snoot length
sw = 1.8 * inches;		// snoot width
sa = 21;			// snoot angle
so = sw/2 * sin(30) - 0.47;	// snoot offset
ss = .2;//0.2;			// snoot scale - 0.2

gpx = 0.05*sl;
gpy = 0.22*cw;

module OuterSnoot()
	linear_sweep(regular_ngon(n=3, d = sw + 4*th), h=sl, convexity = 10, scale = ss);	

module RoughSnoot() {
	difference() {
		tex = texture("rough", roughness=0.5);
		OuterSnoot();
		translate([0, 0, -1])
			linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2, tex_size=[10,10], style="min_edge", convexity = 10, scale = ss);	
		}
	}	

module Snoot() {
	difference() {
		translate([gpx, gpy, 0])
			rotate([0, 0, sa])
                        translate([0, 0, so + 2.5])
			rotate([0, -90, 0]) 
				RoughSnoot();
		translate([0, -100, -eps])
			cube(1000);	
		}
}

Snoot();
```

pca006132, 2023-10-07:
This is actually a known issue, but it is pretty hard to solve. The problem is that we have floating point inaccuracies, and things that are intersecting may become not intersecting with a tiny error. Here is an example in https://manifoldcad.org/

```js
const result = Manifold.cube([2, 2, 2], true).rotate([45, 0, 0])
  .subtract(Manifold.cube([1, 1, 1]).rotate([45, 0, 0]))
```

[image]

```js
const result = Manifold.cube([2, 2, 2], true)
  .subtract(Manifold.cube([1, 1, 1]))
```
[image]

But yeah I agree this is an annoying thing and we should fix it (after the more annoying triangulator bug). I think to solve this we probably need some sort of thin wall removal function to remove walls with thickness less than epsilon. What do you think @elalish?

However, even though there are zero-thickness surfaces, the result should still be topologically valid. So the CGAL error is a bit weird.

nophead, 2023-10-07:
How can a zero thickness face be topologically valid? I don't think CGAL accepts things that are physically impossible.

pca006132, 2023-10-07:
We don't care about the coordinates when we talk only about the topology, so there is no notion of thickness. Things like self-intersection are topologically valid (manifold) but not geometrically valid.

nophead, 2023-10-07:
Well CGAL wont accept self intersections and zero thickness faces, so it must require geometrically valid as well then.

elalish, 2023-10-09:
Hmm, I always added little tolerances in my OpenSCAD designs for situations like this because even if CGAL didn't have trouble, then certainly the F5 renders would. As for physically impossible, what's the minimum thickness of an object that CGAL accepts? This is a digital representation after all. 

It will always be a bit difficult going back and forth between geometry kernels, since they have different assumptions. CGAL sometimes makes non-manifold geometry (which certainly won't work in Manifold, speaking of physically impossible!) and Manifold will make zero-thickness faces in some situations (we try to minimize it, but that's the nature of inexact arithmetic).

elalish, 2023-10-09:
Also, you should know about other users who explicitly desire zero-thickness surfaces: https://github.com/elalish/manifold/issues/560

pca006132, 2023-10-09:
@elalish what do you think about an explicit function to remove thin wall features?

nophead, 2023-10-09:
I pretty much always render differences in preview because otherwise they are slow and also cause artefacts when negative parts of different objects happen to coincide when you assembly them together. I model large assemblies with hundreds or even thousands of parts.

elalish, 2023-10-09:
I'm not willing to promise anything about thin walls. Our Booleans operate on symbolic perturbation and all vert positions contain uncertainty from floating point error. This means if you try to cut a hole with an object that's coplanar (instead of through), the uncertainty means you might get a hole, you might get a thin membrane, and you might get a partial membrane. If e.g. rotations have been part of the tree, then these problems are nearly unavoidable. Basically, if you set up an operation right on the edge of a topological change, you can't predict which one you'll get. We do a good job in common cases, but it's just not possible in general. It's better to design to not be right on the edge.

adrianVmariano, 2023-10-09:
How does floating point error explain the situation for this example?  Yes, if you subtract an object from another with faces that are supposed to be coincident it's apparent why things are difficult at that coincident face.  But in the example here I have an object, I subtract a cube with an arbitrary relationship to my object and a zero thickness face of the cube is left behind.  The extra face that is left behind does not exist in the model before the cube subtraction.  It's not obvious to me how roundoff errors can play a role in this case.

elalish, 2023-10-09:
Oh, I may have misunderstood this example; I was focusing more on the simple case @pca006132 showed. It could be a triangulation fault, which I'm currently fixing. Can you make the repro any simpler?

adrianVmariano, 2023-10-09:
What would you consider to be a simplification?  The example arose when a user reported it as a bug to the BOSL2 library.  I reduced the example to the above.  If I change to a simpler texture it doesn't happen.  If I change the scale factor of the triangular extrusion (the "ss" variable) it doesn't happen.  So not sure what else I could try.

elalish, 2023-10-09:
Okay, no prob; wasn't sure you'd gone through that process yet. Thanks!

pca006132, 2023-10-10:
I think this particular case is probably due to the triangulator bug. I compiled OpenSCAD with https://github.com/elalish/manifold/pull/564 and the rendering is fine:

[image]

Aside: why is the GUI freezing when I click render? I thought the interpreter should be running on another thread...

t-paul, 2023-11-07:
Should be fixed via https://github.com/openscad/openscad/pull/4728.

closed 2023-11-07 by t-paul