openscad/openscad #4771: bytecode-based interpreter

pca006132, 2023-10-07:
https://github.com/openscad/openscad/pull/4654 contains several PRs that tries to speedup the speed of the interpreter. However, I think the fundamental performance problem with OpenSCAD interpreter is caused by the interpreter interpreting using the AST instead of some simpler structure such as bytecode. With a bytecode based interpreter, a lot of the overhead caused by identifier lookup and frame allocation can be avoided.

I just wrote a toy bytecode interpreter here: https://gist.github.com/pca006132/0786e2f91a24cbe70aeb77766810690d
It is just a simple stack machine implemented in typescript. The test.js is the compiled output that you can immediately run with nodejs. The speed on my machine is roughly 4 times faster (even when not using tailcall).

While one can argue that a complete implementation will be a lot more complicated and can slow things down, I still think that this is something that may bring further performance improvement to OpenSCAD. Translation from AST to bytecode should not be too slow as the translation time only depends on the size of the openscad file. This kind of bytecode format is also easier to perform code optimization and even JIT. Also, as we are not using a real stack, recursions that are not tail recursive can still run safely without having to worry about stack overflow, so the recursion limit can be raised.

jordanbrown0, 2023-10-07:
I believe that it would be possible to statically resolve all symbols, with the current interpreter structure.  (Probably excluding $ symbols.)

pca006132, 2023-10-08:
You can statically resolve every symbol, avoid allocation in most cases by doing something similar to a stack machine, but in the end you will end up having to do code transformation on the AST as well. And AST traversal is fundamentally slower than bytecode traversal due to how the cache works, and it is not harder to change to using bytecode if you are willing to do a big refactor anyway...

gsohler, 2023-10-26:
You write a code, its converted to tokens, these tokens are executed in a huge statemachine with "DISPATCH" instruction in the end of each state which resolves to "goto"... thats Python ...

pca006132, 2023-10-26:
yes, python is also bytecode based