openscad/openscad #4780: Hidden part of a colored object applies it's color to overlapped objects

knyri, 2023-10-12:
Using version 2021.01

When two objects of the same dimension overlap and the latter has a portion hidden with difference(), the hidden portion applies it's color to the overlapped object as if it weren't hidden.

```
color("green")
difference(){
	cube(50);
	translate([25,-1,-1])
		cube([26,52,52]);
}
// color of the unmodified cube overrides the green one
color("red")
difference(){
	cube(50);
	translate([-1,-1,-1])
		cube([26,52,52]);
}
// what it should look like
translate([0,0,51])
union(){
	color("green")
	cube([25,50,50]);
	color("red")
	translate([25,0,0])
	cube([25,50,50]);
}
```
Expected output on top, actual output on bottom.
[image]

t-paul, 2023-10-12:
Sorry, but that behavior is inherent to how the preview works, it actually draws that differenced part which causes the coloring. So while it might not be the behavior one would wish for it's not a bug.

closed 2023-10-12 by t-paul

rcolyer, 2023-10-12:
FYI, to understand it better, click on Design, Display CSG Products, and see how the CSG components are reassembled into a list of components that the rapid preview system slaps together.  If the desire is to avoid/block this efficient reassembly process, you can slap "render()" to the right of the first two color() calls, which takes the child contents of render() out of the preview system and (sometimes with a performance overhead) renders the resulting final geometry before applying the color to it.

thehans, 2023-10-13:
Also see issue #1000

knyri, 2023-10-13:
@rcolyer Awesome! Thanks for the render() tip. I'm new to OpenSCAD and don't know all the functions yet. That one did the trick and the CSG Products will definitely help as I step up my 3D prints!
@thehans Thanks for the reference. Again, new to this thing. Knowing they're called CSG Objects and that this is built on top of OpenCSG will help my future solution searching.
The colors aren't critical, I just needed them to help me visualize how the parts meshed together on this complicated print

jordanbrown0, 2023-10-13:
Three tricks associated with previewer issues:
* Never let two different surfaces be at the same point - positive vs negative, or two different colors.  Always ensure that one is slightly in front of the other.  The primary symptom is "z-fighting", shimmering as you move the camera.
* In perspective mode, don't let the camera be inside the model, and in particular not inside otherwise-invisible objects that you use for differencing and intersections.  The symptom is that parts of the model will appear and disappear.
* For anything operation that creates a new polyhedron, if the polyhedron can be concave, set the "convexity" parameter.  The most common cases for this are linear_extrude, rotate_extrude, and polyhedron, but render, import, and others can be affected.  The symptom is that parts of the model become transparent.  Theoretically the "convexity" value should be the number of times that a line can pass into and out of the model, but usually 5 or 10 is plenty and the performance penalty for setting it too high is very small.